revk/xESP32-Faikout #1: Will it be possible to purchase this board?

msx80, 2022-07-05:
Hi there, i'd really love to have one of these boards. Are you planning on selling some of them?

Thank you!

revk, 2022-07-05:
I have a few prototypes, but getting components is a nightmare. When we can get components we expect to sell, yes. Are you in UK, if so I could post you a one of the prototypes maybe, for some feedback…

Boki3333, 2022-07-16:
> I have a few prototypes, but getting components is a nightmare. When we can get components we expect to sell, yes. Are you in UK, if so I could post you a one of the prototypes maybe, for some feedback…

Hey mate. 
May i ask you , what is your email to contact you ?

revk, 2022-07-17:
I don't usually give out emails, but tweet at me and we can DM.

Boki3333, 2022-07-17:
> I don't usually give out emails, but tweet at me and we can DM.

I do not have tweeter, facebook, nothing .. ( 
I see also here its not possible to sent messeges....
I am modern, but still not as modern to be everywhere. 
I use either mail, or apps for chat. 

Please send me email on : moja.poraba@gmail.com 

Thank you in advance.

Boki3333, 2022-07-17:
> I don't usually give out emails, but tweet at me and we can DM.

First question : Which element is to be choosen for onboard connector from AC to the module ? 
                         I am searching this onboard right-angle male plug, but i am not able to find it. 
                         I am using EasyEda, to do module based on your published schematics.
                         
 Second question : Why Rx signal acts as " reversed " ? Is there any specific reason behind this ? 
                              It has to be reversed constantly or the reason is ESP32 module ? 
                              
 Third question : Since i do not see any benefits in ESP32, and my whole home is automated based on Esp8266, probably 
                           i could do the same module, only with Esp8266 instead of ESP32, correct ? 
                           And those " rgb " lights are actually not needed for me, so i would remove them. 
                           I will check more in detail, why those RGBS are for, but in my case not needed. 
                           
  THANK YOU in ADVANCE for your replies and your help.
[image]

revk, 2022-07-18:
1. The RGB LED is just for show, it is on most of my boards and in this case shows cooling/heating, so not needed.
2. The Daikin units I have already have a lead, and it has a connector that plugs in to this (5 pin connector but only 4 pins used). I believe on most of them the PCB side is called S21, and you need to set the s21 flag in the settings in my code for that as I have not yet made it automatically detect.
3. No reason you could not do this on an ESP8266, it is just serial comms - but the code is all designed for ESP32, so no idea how difficult it would be to make it compile for an ESP8266, sorry. I'd suggest an ESP32 would be easier.
4. The Rx uses a FET because the Daikin is 5V not 3.3V. For Tx, sending 3.3V to the Daikin works and is safe, but for Rx it is not ideal for it to be 5V in to the ESP. I works, but using a FET is a better way to do it, and has the side effect of inverting the Rx signal. My code is designed to allow inverted or non inverted GPIO for Tx or Rx so not a problem if you make a circuit that is not inverted.
5. I do have a load of spare unpopulated PCBs if you want me to post you one.

Boki3333, 2022-07-19:
> 1. The RGB LED is just for show, it is on most of my boards and in this case shows cooling/heating, so not needed.
> 2. The Daikin units I have already have a lead, and it has a connector that plugs [...]

Hi. 

Thanks for all the answers, i appreaciate it, and it makes things clearer to me. 
Please look at my number 2 more accurately, so that i could solve it :)

1 ) Ok, i understand. 
3) Yes maybe can be easier, but Esp8266 has enough of the " needed capacity " for this work. So yes, 
    this code will have to be compiled - rewritten for Esp8266. So this means work work work work......
4) Yes, i know, that 5V tolerancy on pins for esp8266 are ok, but not ok, if you understand what i mean. Its quite 
    an interesting way of " solving " this tollerancy, and leaving it for RX. I was thinking about putting rather 
    a 10k resistor , so that RX pin would be connected on 1 side and then on other side 5V from voltage converter. 
    Maybe that should do the work also, what you think ? But since it works with this FET, and the side effect is " inversion " , 
    probably i should leave it.... You think so ? 
5) You mean by " unpopulated " as not assembled and for ESP32 ? Is this what you mean     

And here is this nr.2 , that i do not have answer for yet : 
S21 - i am not able to find in EasyEda. There is so many different parts, i had to focus on the ones, that i am assuming 
will fit. 
Here are few examples  - much appreciated for your cconfirmation : 
a) https://datasheet.lcsc.com/lcsc/1810222341_Ckmtw-Shenzhen-Cankemeng-W-2510R05P-F000_C169030.pdf
b) https://datasheet.lcsc.com/lcsc/1810221937_BOOMELE-Boom-Precision-Elec-C61410_C61410.pdf
c) https://datasheet.lcsc.com/lcsc/1810221710_BOOMELE-Boom-Precision-Elec-C45012_C45012.pdf ( looks excatly the same 
    as the previous one )

Boki3333, 2022-07-19:
> 1. The RGB LED is just for show, it is on most of my boards and in this case shows cooling/heating, so not needed.
> 2. The Daikin units I have already have a lead, and it has a connector that plugs [...]

Look, here are attached pictures from the Module, and from the cable. 
And those parts do not look like molex to me, neither similat to the one i have choosen, because 
in my " chosen parts " there can not be any " click " sound , because are unshielded from the side. 
What you advise ? 
[image]
[image]

revk, 2022-07-19:
Odd, there were additional comments on here but not showing now. The connector on the board is 4 pins, and could be simple 0.1" molex style pins which will fit in the 5 pin socket on the end of the remote control lead, if that makes sense.

[image]
.

Boki3333, 2022-07-21:
> Odd, there were additional comments on here but not showing now. The connector on the board is 4 pins, and could be simple 0.1" molex style pins which will fit in the 5 pin socket on the end of the remote control lead, if that makes sense.
> 
> [image] .

Hey there.... Been busy with the module. 
Yes, i see that comments are not showing ? What is this ? 
Well, i have found the correct connector. Its from JST, and also i have managed to do the module, based on Esp82, because as i said , esp32 is not needed.

Here is a picture of a module, i created.

What you think ? 
[image]
 
It fits perfectly into Daikin unit :)

Boki3333, 2022-07-25:
> Odd, there were additional comments on here but not showing now. The connector on the board is 4 pins, and could be simple 0.1" molex style pins which will fit in the 5 pin socket on the end of the remote control lead, if that makes sense.
> 
> [image] .

Hi there. 
I hope this reply and request will stay, and will not be gone. 
I had almost no time to get back, i was completely busy with some work. 
May i ask you few more questions about the design ? 

1) You wrote on schematics " TX and RX " pins, when actually you used on ESP32 not TX and RX pins, but GPIO26 and GPIO27 . 
    This is a bit confusing to me, because you marked original TX and RX pins differently. I dont get this. 
    Is this TX or RX used for connection to Daikin or i am missing any point  ? 
2) Colors of the wires are assigned as my assumption is : 
    BLUE - GND
    YELLOW - Power
    ORANGE - TX 
    RED - RX    
3) If my assumption is correct, please help me out witht he correct connections : 
    ORANGE - TX   goes on TX pin on ESP ? 
    RED - RX goees on RX pin on ESP ? 
    Am i right or is it turned around as : 
    ORANGE - TX  goes on RX pin on ESP ? 
    RED - RX goes on TX pin on ESP ?     
4) Why you used regulator Vin / 4V -60 for mains and then Vout 5V ? Daikin Vin is higher then 5V ? 

Thank you very much in advance for your answer .

revk, 2022-07-26:
1. The ESP32 can assign Tx and Rx on any of the three UARTs to any GPIOs. I have assigned those to the communications with the Daikin module. The "default" tx and rx on UART 0 are assigned to the 5 pin pads (J2) for programming.
2. The schematic shows the order of J1: GND, DC Power, Tx (to Daikin),Rx (from Daikin). Looking at your cable, and mine, that is Blue, Yellow, Orange Red, yes.
3. As per the schematic, with arrows on the tabs even, the third pin is Tx *to* Daikin, forth is Rx *from* Daikin.
4. Not sure I understand, the ESP32 needs 3.3V. The power from the Daikin (second pin on J1, labeled DC) goes to the regulator VIN (which allows 4V to 60V). VOUT from the regulator (as per the schematic) is 3.3V to power the ESP32. None is "mains". You can use either the variable LMR16006 with the 100k/30k resistors, or the fixed 3.3V LMR16006 Q3 part with the 0R/NF resistors, either way it is 3.3V regulated output for the ESP32.

Boki3333, 2022-07-26:
> 1. The ESP32 can assign Tx and Rx on any of the three UARTs to any GPIOs. I have assigned those to the communications with the Daikin module. The "default" tx and rx on UART 0 are assigned to the 5 [...]

Hey mate. 
Thank you very much for your quick reply . 
But still i do not understand what you mean " from " and " to " Daikin. I understand this a bit different , or at least i think i understand this ....... :-(

1) Here is how i see it, and also a picture in the attachment , how i understand the connections of TX and RX :

a) Gnd    - Ground / Blue Wire
b) Vin     - this pin connected to regulator Vin and Vout connected on Esp to power it  / Yellow wire
c) Tx       - this pin connected to Tx pin on the Esp / Orange Wire
d) Rx      - this pin connected on FET , and then to Rx on the Esp / Red Wire
e) None - nothing connected

Correct ? 

2) My mistake calling it " mains " and i apologize for it. I was reffering to the " main power " which is called Vin. You are using a regulator with huge range of Voltage - Vin (4V to 60V) . What i dont understand is why such regulator ? Couldnt be a more simpler regulator withouth such range ? 
Does Daikin AC gives out Vin on such high Voltages, that you used such wide range regulator ( 4-60V ) ? Is that the reason ? 
 If a " simple " regulator can be used that is able to regulate from 5V to 3V for esp to be powered, then price for the board reduce sufficiantly. I usually use regulators , that do not have such " extended " Vin range.  This one, that you choose, is awesome, but the range is so wide, i am wondering why... Can you clarify me this please ? 
 
 Thank you very much again for your reply.
[image]
[image]

revk, 2022-07-26:
1. When talking of tx and rx obviously you have to consider "from" and "to". Data is sent from one device to another. On the sending device that is normally considered tx (transmit), and on the receiving device it is considered rx (receive). So labelling always has to involve considering direction. Pin 3 is data FROM the ESP32 and TO the DAIKIN. Pin 4 is date FROM the DAIKIN and TO the ESP32. Simple as that. It is normal to consider tx and rx labels as viewed from a device, and it helps to explain from and to in order to avoid any confusion.
2. I am using a regulator of which I have some in stock and I use on a lot of designs. It is low power, and I have a proven design and layout for it. Simple as that. Yes a simpler LDO could be used instead.

Boki3333, 2022-07-26:
> 1. When talking of tx and rx obviously you have to consider "from" and "to". Data is sent from one device to another. On the sending device that is normally considered tx (transmit), and on the receiving device it is considered rx (receive). So labelling always has to involve considering direction. On my design, pin 3 is data FROM the ESP32 and TO the DAIKIN. Pin 4 is data FROM the DAIKIN and TO the ESP32. Simple as that. It is normal to consider tx and rx labels as viewed from a device, and it helps to explain from and to in order to avoid any confusion.
> 2. I am using a regulator of which I have some in stock and I use on a lot of designs. It is low power, and I have a proven design and layout for it. Simple as that. Yes a simpler LDO could be used instead.

Thank you for quick reply. 
Yes, i agree, and you are apsolutely right. The labeling has to involve direction. But its a bit confusing, this " from " and " to " . 
At least for me, because i understand Tx and Rx pins and what they do. 

This regulator has a wide Voltage range. Its an excellent choice, if such voltage ranges are needed,  because it works in all ranges. But is also true, that such regulator is much more expensive, then the " ordinary " regulator Vin 5V - Vout 3V , and my only reason to asking is, to reduce the price of the board itself. Now its clear to me and understandable.

THANK YOU very very much for all the help with clarification of everything. 
You have done excelent work )

msx80, 2022-07-27:
Hi all, in the end i went ahead and made myself a board, out of off the shelf components. They're a bit bigger but still fit the slot quite nicely. Seems to work well, i've done 3 now and the first is up for more than a week without issues.

For the software, i used Tasmota to handle all the basic configuration/http/mqtt stuff, and i wrote driving code with Berry (tasmota esp32 scripting language).

If anyone is interested, sources and pictures are at https://github.com/msx80/FakeA41

revk thanks again for your kind offer and for the informations you made available on this repo!

PS at 2400 baud the ESP32 is more than fast enougth to drive the serial with software, which i did on my project so you can choose any two pins

boehla, 2022-09-17:
> I have a few prototypes, but getting components is a nightmare. When we can get components we expect to sell, yes. Are you in UK, if so I could post you a one of the prototypes maybe, for some feedback…

Hello revk, are you selling devices now? Would be interested in one device. I am from AT.

revk, 2022-09-17:
Once components are available I plan to have some made up and sold on Amazon, but that is not likely before next year at this rate, sorry.

boehla, 2022-09-17:
> Once components are available I plan to have some made up and sold on Amazon, but that is not likely before next year at this rate, sorry.

Ok thanks for the info :)
With the KiCad folder it should be possible to order an already assembled pcb? 
Otherwise I can wait.

revk, 2022-09-18:
Well yes - making a BOM for your supplier may need a bit of editing but all the details are there.

nightwatcher74, 2022-09-18:
Hi revk,

at first... I'm an absolute ESP32 beginner... :(

What do you think, would it be possible to user an M5Stack Atom Lite Module for the Daikin integration??
https://shop.m5stack.com/products/atom-lite-esp32-development-kit
(The Grove Pins G26/G32 are 5V)

There is also an 6-24VDC => 5V available:
https://shop.m5stack.com/collections/m5-accessories/products/5v-buck-converter-unitme3116am6g

when i understand it correctly, it should be possible to connect in the following scheme:

Daikin     => ESP32
 V+         => PowerConverter +
GND       => PowerConverter -
RX           => G26 (Grove Pin)
TX           => G32 (Grove Pin)

is it possible to run ESP32-Daikin in such an environment??

Thanks 
Frank

revk, 2022-09-19:
It should run on any ESP32. The make file has builds for solo ESP32 even, ESP32 PICO, and ESP32 WROOM. And you can configure the GPIOs to use over MQTT.

mx5gr, 2022-09-24:
@nightwatcher74  I am interested in this as well, i.e. using a M5 Atom Lite (for small size) and a buck converter.

However, the GPIOs of the Atom are at 3.3V. Isn't a level shifter required as well since the Daikin serial I/O operates at 5V (if I am not mistaken)?

revk, 2022-09-24:
You can get away with it, but in my design I added a FET to do the level shift on input. On output the Daikin is happy with 3.3V to its input.

mx5gr, 2022-09-24:
I am thinking of powering the Atom through the USB connector and use a readily-available level shifter, such as https://www.sparkfun.com/products/12009 (High side the Daikin 5V supply and as channels the two respective lines, Low side the 3.3V coming from the Atom and 2 respective GPIO lines).

I am thinking of a bidirectional connection. If Daikin is happy with 3,3V as its input, the ESP32 might not be happy with the 5V coming from the Daikin unit to its respective GPIO input.

In that sense, not even a buck converter is required... that is, if my assumptions are correct :) :) :)

closed 2022-12-30 by revk