revk/xESP32-Faikout #2: Errors on Make

WheresWes, 2022-07-28:
I'm quite excited for this project.  I already have some other panasonic and mitsi heat pumps connected to home assistant through esphome.
I have to admit however that I am in somewhat over my head.  I have spent some hours installing popt/mysql/mosquitto/mariadb on a ubuntu installation to satisfy the dependancies.

However when I try to make I get the following errors.  Hoping you might have some pointers for this poor fool...
pi@raspberrypi:~/ESP32-Daikin $ make
make: components/ESP32-RevK/buildsuffix: Command not found
cc -O -o daikinlog daikinlog.c -lpopt -lmosquitto -ISQLlib SQLlib/sqllib.o -IAJL AJL/ajl.o -L/usr/local/ssl/lib -L/usr/lib/arm-linux-gnueabihf -lmariadb -I/usr/include/mariadb -I/usr/include/mariadb/mysql -I. -I/usr/local/ssl/include -D_GNU_SOURCE -g -Wall -funsigned-char -lm
SQLlib/sqllib.o: In function `main':
sqllib.c:(.text+0x31f8): multiple definition of `main'
/tmp/cc8OoS7i.o:/home/pi/ESP32-Daikin/daikinlog.c:16: first defined here
AJL/ajl.o: In function `main':
ajl.c:(.text+0x9bc8): multiple definition of `main'
/tmp/cc8OoS7i.o:/home/pi/ESP32-Daikin/daikinlog.c:16: first defined here
SQLlib/sqllib.o: In function `sql_real_connect':
sqllib.c:(.text+0x6f4): undefined reference to `strdupa'
AJL/ajl.o: In function `process.10561':
ajl.c:(.text+0x99f4): undefined reference to `j_curl'
ajl.c:(.text+0x9a34): undefined reference to `j_curl'
ajl.c:(.text+0x9a74): undefined reference to `j_curl'
AJL/ajl.o: In function `main':
ajl.c:(.text+0x9e9c): undefined reference to `curl_easy_init'
ajl.c:(.text+0x9ecc): undefined reference to `curl_easy_setopt'
ajl.c:(.text+0xa130): undefined reference to `curl_easy_cleanup'
collect2: error: ld returned 1 exit status
Makefile:77: recipe for target 'daikinlog' failed
make: *** [daikinlog] Error 1

revk, 2022-07-28:
You need to recusive checkout.

A `make pull` make sort it

WheresWes, 2022-07-28:
Hmm that does not appear to have had any effect.
***@***.***:~/ESP32-Daikin $ make pull
make: components/ESP32-RevK/buildsuffix: Command not found
git pull
Already up-to-date.
git submodule update --recursive
***@***.***:~/ESP32-Daikin $ make
make: components/ESP32-RevK/buildsuffix: Command not found
cc -O -o daikinlog daikinlog.c -lpopt -lmosquitto -ISQLlib
SQLlib/sqllib.o -IAJL AJL/ajl.o -L/usr/local/ssl/lib
-L/usr/lib/arm-linux-gnueabihf -lmariadb -I/usr/include/mariadb
-I/usr/include/mariadb/mysql -I. -I/usr/local/ssl/include
-D_GNU_SOURCE -g -Wall -funsigned-char -lm
SQLlib/sqllib.o: In function `main':
sqllib.c:(.text+0x31f8): multiple definition of `main'
/tmp/ccqI22af.o:/home/pi/ESP32-Daikin/daikinlog.c:16: first defined here
AJL/ajl.o: In function `main':
ajl.c:(.text+0x9bc8): multiple definition of `main'
/tmp/ccqI22af.o:/home/pi/ESP32-Daikin/daikinlog.c:16: first defined here
SQLlib/sqllib.o: In function `sql_real_connect':
sqllib.c:(.text+0x6f4): undefined reference to `strdupa'
AJL/ajl.o: In function `process.10561':
ajl.c:(.text+0x99f4): undefined reference to `j_curl'
ajl.c:(.text+0x9a34): undefined reference to `j_curl'
ajl.c:(.text+0x9a74): undefined reference to `j_curl'
AJL/ajl.o: In function `main':
ajl.c:(.text+0x9e9c): undefined reference to `curl_easy_init'
ajl.c:(.text+0x9ecc): undefined reference to `curl_easy_setopt'
ajl.c:(.text+0xa130): undefined reference to `curl_easy_cleanup'
collect2: error: ld returned 1 exit status
Makefile:77: recipe for target 'daikinlog' failed
make: *** [daikinlog] Error 1

revk, 2022-07-28:
Ah chicken and egg, can't run any make until it has the reclusive check out.

`git submodule update --recursive
`

WheresWes, 2022-07-28:
Thanks for helping ..
tried the above but still no change.

***@***.***:~/ESP32-Daikin $ git submodule update --recursive
***@***.***:~/ESP32-Daikin $ make pull
make: components/ESP32-RevK/buildsuffix: Command not found
git pull
Already up-to-date.
git submodule update --recursive
***@***.***:~/ESP32-Daikin $ make
make: components/ESP32-RevK/buildsuffix: Command not found
cc -O -o daikinlog daikinlog.c -lpopt -lmosquitto -ISQLlib
SQLlib/sqllib.o -IAJL AJL/ajl.o -L/usr/local/ssl/lib
-L/usr/lib/arm-linux-gnueabihf -lmariadb -I/usr/include/mariadb
-I/usr/include/mariadb/mysql -I. -I/usr/local/ssl/include
-D_GNU_SOURCE -g -Wall -funsigned-char -lm
SQLlib/sqllib.o: In function `main':
sqllib.c:(.text+0x31f8): multiple definition of `main'
/tmp/ccHKM3Ew.o:/home/pi/ESP32-Daikin/daikinlog.c:16: first defined here
AJL/ajl.o: In function `main':
ajl.c:(.text+0x9bc8): multiple definition of `main'
/tmp/ccHKM3Ew.o:/home/pi/ESP32-Daikin/daikinlog.c:16: first defined here
SQLlib/sqllib.o: In function `sql_real_connect':
sqllib.c:(.text+0x6f4): undefined reference to `strdupa'
AJL/ajl.o: In function `process.10561':
ajl.c:(.text+0x99f4): undefined reference to `j_curl'
ajl.c:(.text+0x9a34): undefined reference to `j_curl'
ajl.c:(.text+0x9a74): undefined reference to `j_curl'
AJL/ajl.o: In function `main':
ajl.c:(.text+0x9e9c): undefined reference to `curl_easy_init'
ajl.c:(.text+0x9ecc): undefined reference to `curl_easy_setopt'
ajl.c:(.text+0xa130): undefined reference to `curl_easy_cleanup'
collect2: error: ld returned 1 exit status
Makefile:77: recipe for target 'daikinlog' failed
make: *** [daikinlog] Error 1

revk, 2022-07-28:
What do you have in components?

Do you have components/ESP32-RevK/buildsuffix

A recursive checkout should have got that. If you did not do a recursive clone in the first place, maybe try again from scratch.

The other errors are a slight concern, but lets try and sort one thing at a time!

WheresWes, 2022-07-28:
Ok,
just ran:
git clone https://github.com/revk/ESP32-Daikin --recursive

followed again by the other commands you suggested.. doesnt appear to
have had any effect.

Yes I have components/esp32-rek/ folder with a buildsuffix file present

revk, 2022-07-28:
So does components/ESP32-RevK/buildsuffix exist?

WheresWes, 2022-07-28:
Yes folder components/ESP32-RevK/ exists and therein is a file buildsuffix

content of buildsuffix ifile s as follows:

#!/bin/csh -f

setenv	SUFFIX	""

setenv SDKCONFIG "sdkconfig"
if("$1" != "") setenv SDKCONFIG "$1"

if(! -e "$SDKCONFIG") then
        echo "Missing $SDKCONFIG"
        exit 0
endif

# processor type
grep -q CONFIG_IDF_TARGET_ESP32=y "$SDKCONFIG"
if(! $status) 	setenv	SUFFIX	"$SUFFIX-S1"
grep -q CONFIG_IDF_TARGET_ESP32S2=y "$SDKCONFIG"
if(! $status) 	setenv	SUFFIX	"$SUFFIX-S2"
grep -q CONFIG_IDF_TARGET_ESP32S3=y "$SDKCONFIG"
if(! $status) 	setenv	SUFFIX	"$SUFFIX-S3"
grep -q CONFIG_IDF_TARGET_ESPC3=y "$SDKCONFIG"
if(! $status) 	setenv	SUFFIX	"$SUFFIX-C3"
grep -q CONFIG_IDF_TARGET_ESPH2=y "$SDKCONFIG"
if(! $status) 	setenv	SUFFIX	"$SUFFIX-H2"

# Single ore
grep -q CONFIG_FREERTOS_UNICORE=y "$SDKCONFIG"
if(! $status) 	setenv	SUFFIX	"$SUFFIX-SOLO"

# Specific processor (ESP32-PICO)
grep -q CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y "$SDKCONFIG"
if(! $status) then
grep -q CONFIG_ESP32_SPIRAM_SUPPORT=y "$SDKCONFIG"
if(! $status) 	setenv	SUFFIX	"$SUFFIX-PICO"
endif

# Min rev build
grep -q CONFIG_ESP32_REV_MIN_3=y "$SDKCONFIG"
if(! $status) 	setenv	SUFFIX	"$SUFFIX-V3"
grep -q CONFIG_ESP32_REV_MIN_2=y "$SDKCONFIG"
if(! $status) 	setenv	SUFFIX	"$SUFFIX-V2"
grep -q CONFIG_ESP32_REV_MIN_1=y "$SDKCONFIG"
if(! $status) 	setenv	SUFFIX	"$SUFFIX-V1"

set bs=(`grep _BUILD_SUFFIX_ sdkconfig | sed
's/.*\(CONFIG_[A-Z0-9_]*_BUILD_SUFFIX_[A-Z0-9]*\).*/\1/'`)
foreach c ($bs)
	grep -q "$c=y" "$SDKCONFIG"
	if(! $status)	setenv	SUFFIX	"$SUFFIX-"`echo -n "$c" | sed
's/CONFIG_.*_BUILD_SUFFIX_\(.*\)/\1/'`
end

grep -q '^CONFIG_REVK_BUILD_SUFFIX="'"$SUFFIX"'"$' "$SDKCONFIG"
if($status) then
	sed -e 's/^CONFIG_REVK_BUILD_SUFFIX=.*$/CONFIG_REVK_BUILD_SUFFIX="'"$SUFFIX"'"/'
"$SDKCONFIG" > "$SDKCONFIG.new"
	mv -f "$SDKCONFIG.new" "$SDKCONFIG"
endif

echo -n "$SUFFIX"

revk, 2022-07-28:
I am guessing you don't have `/bin/csh`

WheresWes, 2022-07-28:
I do not appear to no.
Where should this be located?

WheresWes, 2022-07-28:
I installed csh.

That has resolved the first error.  The rest looks like this :

***@***.***:~/ESP32-Daikin $ make
cc -O -o daikinlog daikinlog.c -lpopt -lmosquitto -ISQLlib
SQLlib/sqllib.o -IAJL AJL/ajl.o -L/usr/local/ssl/lib
-L/usr/lib/arm-linux-gnueabihf -lmariadb -I/usr/include/mariadb
-I/usr/include/mariadb/mysql -I. -I/usr/local/ssl/include
-D_GNU_SOURCE -g -Wall -funsigned-char -lm
SQLlib/sqllib.o: In function `main':
sqllib.c:(.text+0x31f8): multiple definition of `main'
/tmp/cc6fXzNL.o:/home/pi/ESP32-Daikin/daikinlog.c:16: first defined here
AJL/ajl.o: In function `main':
ajl.c:(.text+0x9bc8): multiple definition of `main'
/tmp/cc6fXzNL.o:/home/pi/ESP32-Daikin/daikinlog.c:16: first defined here
SQLlib/sqllib.o: In function `sql_real_connect':
sqllib.c:(.text+0x6f4): undefined reference to `strdupa'
AJL/ajl.o: In function `process.10561':
ajl.c:(.text+0x99f4): undefined reference to `j_curl'
ajl.c:(.text+0x9a34): undefined reference to `j_curl'
ajl.c:(.text+0x9a74): undefined reference to `j_curl'
AJL/ajl.o: In function `main':
ajl.c:(.text+0x9e9c): undefined reference to `curl_easy_init'
ajl.c:(.text+0x9ecc): undefined reference to `curl_easy_setopt'
ajl.c:(.text+0xa130): undefined reference to `curl_easy_cleanup'
collect2: error: ld returned 1 exit status
Makefile:77: recipe for target 'daikinlog' failed
make: *** [daikinlog] Error 1

revk, 2022-07-28:
Ok that is weird, like the libraries have not build properly, which is strange.

Ah, OK, that may be an issue on my part. Let me tinker with the make file a moment.

WheresWes, 2022-07-28:
Appreciate the help. Super impressed that you are able to decipher this
stuff.

I will call it a night for now (I'm in NZ) and pick this up tomorrow

Regards Wes

revk, 2022-07-28:
Ok update, and make clean, and try again. It should make the libraries now.

WheresWes, 2022-07-28:
I ran 'git pull' and 'make clean' then ran make again with the same
result.. was that how I should have run it?

***@***.***:~/ESP32-Daikin $ git pull
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Compressing objects: 100% (1/1), done.
remote: Total 3 (delta 2), reused 3 (delta 2), pack-reused 0
Unpacking objects: 100% (3/3), done.
From https://github.com/revk/ESP32-Daikin
   c5cbf07..3a589db  main       -> origin/main
Updating c5cbf07..3a589db
Fast-forward
 Makefile | 8 ++++++++
 1 file changed, 8 insertions(+)
***@***.***:~/ESP32-Daikin $ make clean
idf.py clean
make: idf.py: Command not found
Makefile:39: recipe for target 'clean' failed
make: *** [clean] Error 127
***@***.***:~/ESP32-Daikin $ make
cc -O -o daikinlog daikinlog.c -lpopt -lmosquitto -ISQLlib
SQLlib/sqllib.o -IAJL AJL/ajl.o -L/usr/local/ssl/lib
-L/usr/lib/arm-linux-gnueabihf -lmariadb -I/usr/include/mariadb
-I/usr/include/mariadb/mysql -I. -I/usr/local/ssl/include
-D_GNU_SOURCE -g -Wall -funsigned-char -lm
SQLlib/sqllib.o: In function `main':
sqllib.c:(.text+0x31f8): multiple definition of `main'
/tmp/ccZLcEs3.o:/home/pi/ESP32-Daikin/daikinlog.c:16: first defined here
AJL/ajl.o: In function `main':
ajl.c:(.text+0x9bc8): multiple definition of `main'
/tmp/ccZLcEs3.o:/home/pi/ESP32-Daikin/daikinlog.c:16: first defined here
SQLlib/sqllib.o: In function `sql_real_connect':
sqllib.c:(.text+0x6f4): undefined reference to `strdupa'
AJL/ajl.o: In function `process.10561':
ajl.c:(.text+0x99f4): undefined reference to `j_curl'
ajl.c:(.text+0x9a34): undefined reference to `j_curl'
ajl.c:(.text+0x9a74): undefined reference to `j_curl'
AJL/ajl.o: In function `main':
ajl.c:(.text+0x9e9c): undefined reference to `curl_easy_init'
ajl.c:(.text+0x9ecc): undefined reference to `curl_easy_setopt'
ajl.c:(.text+0xa130): undefined reference to `curl_easy_cleanup'
collect2: error: ld returned 1 exit status
Makefile:85: recipe for target 'daikinlog' failed
make: *** [daikinlog] Error 1

revk, 2022-07-28:
Ah make clean failed as you have not got the ESP IDF set up.

Have you installed the ESP IDF? See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html

It takes a while to install, and set up, but is needed before you can make any of these ESP modules.

WheresWes, 2022-07-29:
Ok.. Making progress.

I set up a new Debian install on a spare laptop.  The other installation
was pretty old and I think some of the dependencies hadnt been updated
properly.  Python in particular was proving problematic.

Now I am at this point:

Interestingly the buildsuffix error is back - I'm sure it was gone in the
first make after I install csh.

***@***.***:~/ESP32-Daikin$ make
make: components/ESP32-RevK/buildsuffix: No such file or directory
Make: Daikin.bin
Executing action: all (aliases: build)
Running cmake in directory /home/wes/ESP32-Daikin/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1
-DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/wes/ESP32-Daikin"...
-- Found Git: /usr/bin/git (found version "2.30.2")
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler:
/home/wes/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler:
/home/wes/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
- skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler:
/home/wes/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
- skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/wes/ESP32-Daikin/sdkconfig
Loading defaults file /home/wes/ESP32-Daikin/sdkconfig.defaults...
/home/wes/ESP32-Daikin/sdkconfig.defaults:289
CONFIG_ESP32_DEFAULT_CPU_FREQ_160 was replaced with
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_160
/home/wes/ESP32-Daikin/sdkconfig.defaults:291
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ was replaced with
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ
/home/wes/ESP32-Daikin/sdkconfig.defaults:292 CONFIG_ESP32_SPIRAM_SUPPORT
was replaced with CONFIG_SPIRAM
/home/wes/ESP32-Daikin/sdkconfig.defaults:373
CONFIG_ESP32_ULP_COPROC_ENABLED was replaced with CONFIG_ULP_COPROC_ENABLED
/home/wes/ESP32-Daikin/sdkconfig.defaults:374
CONFIG_ESP32_ULP_COPROC_RESERVE_MEM was replaced with
CONFIG_ULP_COPROC_RESERVE_MEM
/home/wes/ESP32-Daikin/sdkconfig.defaults:375 CONFIG_ESP32_DEBUG_OCDAWARE
was replaced with CONFIG_ESP_DEBUG_OCDAWARE
/home/wes/ESP32-Daikin/sdkconfig.defaults:376 CONFIG_ESP32_BROWNOUT_DET was
replaced with CONFIG_ESP_BROWNOUT_DET
/home/wes/ESP32-Daikin/sdkconfig.defaults:377
CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_0 was replaced with
CONFIG_ESP_BROWNOUT_DET_LVL_SEL_0
/home/wes/ESP32-Daikin/sdkconfig.defaults:385 CONFIG_ESP32_BROWNOUT_DET_LVL
was replaced with CONFIG_ESP_BROWNOUT_DET_LVL
/home/wes/ESP32-Daikin/sdkconfig.defaults:386
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1 was replaced with
CONFIG_NEWLIB_TIME_SYSCALL_USE_RTC_HRT
/home/wes/ESP32-Daikin/sdkconfig.defaults:390
CONFIG_ESP32_RTC_CLK_SRC_INT_RC was replaced with CONFIG_RTC_CLK_SRC_INT_RC
/home/wes/ESP32-Daikin/sdkconfig.defaults:394
CONFIG_ESP32_RTC_CLK_CAL_CYCLES was replaced with CONFIG_RTC_CLK_CAL_CYCLES
[... 94 lines cut ...]
/home/wes/esp/esp-idf/components/protobuf-c
/home/wes/esp/esp-idf/components/protocomm
/home/wes/esp/esp-idf/components/pthread
/home/wes/esp/esp-idf/components/sdmmc /home/wes/esp/esp-idf/components/soc
/home/wes/esp/esp-idf/components/spi_flash
/home/wes/esp/esp-idf/components/spiffs
/home/wes/esp/esp-idf/components/tcp_transport
/home/wes/esp/esp-idf/components/tinyusb
/home/wes/esp/esp-idf/components/ulp /home/wes/esp/esp-idf/components/unity
/home/wes/esp/esp-idf/components/usb /home/wes/esp/esp-idf/components/vfs
/home/wes/esp/esp-idf/components/wear_levelling
/home/wes/esp/esp-idf/components/wifi_provisioning
/home/wes/esp/esp-idf/components/wpa_supplicant
/home/wes/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wes/ESP32-Daikin/build
Running ninja in directory /home/wes/ESP32-Daikin/build
Executing "ninja all"...
ninja: error: '../secure_boot_signing_key.pem', needed by
'signature_verification_key.bin', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the
/home/wes/ESP32-Daikin/build/log/idf_py_stderr_output_14865 and
/home/wes/ESP32-Daikin/build/log/idf_py_stdout_output_14865
make: *** [Makefile:12: all] Error 2

revk, 2022-07-29:
That is some good progress.

So, recheck the `components/ESP32-RevK/buildsuffix` does exist and the`/bin/csh` exists. Maybe try running it manually to make sure, etc. That should be something you can fix again.

As for `../secure_boot_signing_key.pem` that is down to a menuconfig setting to use signed code. You can either change the setting in `make menuconfig` not to sign the code, or follow the instructions on the ESP IDF for making your own secure boot `.pem` file.

I have now changed the defaults to not secure sign, but that won't help now you have a config.

WheresWes, 2022-07-29:
Ok winning
I deleted the whole folder and recloned the git as I must have screwed up
something in the menuconfig as after that nothing worked.

Now I get a lot of healthy looking activity with the following exceptions:

Make: DaikinMissing sdkconfig.bin

And then later:

In file included from ../components/ESP32-RevK/revk.c:13:
../components/ESP32-RevK/include/revk.h:12:10: fatal error: esp_wifi.h: No
such file or directory
   12 | #include "esp_wifi.h"
      |          ^~~~~~~~~~~~
compilation terminated.
[832/845] Building C object
esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/jo.c.objninja: build
stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the
/home/wes/ESP32-Daikin/build/log/idf_py_stderr_output_17130 and
/home/wes/ESP32-Daikin/build/log/idf_py_stdout_output_17130
make: *** [Makefile:12: all] Error 2

I can see a esp_wifi.h in the esp/esp-ide/components/include/ folder...
what is the best way of pointing to it?

revk, 2022-07-29:
Now you have got me! Even the esp idf wifi examples simply have `#include "esp_wifi.h"` so I don't understand why that is not working. It is (obviously) working for me.

Maybe I am missing something in the `CMakeLists.txt` `REQUIRES`, but I can't see why it would be working for me if that is needed.

I have added `esp_wifi` to the `REQUIRES` list, do a `make update` to pull in latest ESP32-RevK with this and see if that helps. But it is a total guess.

WheresWes, 2022-07-29:
Well 2 steps forwards...

The sdkconfig.bin error is now gone

And rather than a esp_wifi.h error I get a esp_timer.h error.

808/821 so close!

In file included from ../components/ESP32-RevK/revk.c:13:
../components/ESP32-RevK/include/revk.h:16:10: fatal error: esp_timer.h: No
such file or directory
   16 | #include "esp_timer.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
[808/821] Building C object
esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/jo.c.objninja: build
stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the
/home/wes/ESP32-Daikin/build/log/idf_py_stderr_output_21615 and
/home/wes/ESP32-Daikin/build/log/idf_py_stdout_output_21615
make: *** [Makefile:12: all] Error 2

revk, 2022-07-29:
Wow. This is crazy. Only thing I can thing is I am building with stable not master esp idf.

But anyway, I have added `esp_timer` to the `REQUIRES` now, try another `make update`.

WheresWes, 2022-07-29:
Its really not wanting to look in that include folder

next candidate esp_int_wdt.h

K.dir/revk.c.obj.d -o
esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/revk.c.obj -c
../components/ESP32-RevK/revk.c
../components/ESP32-RevK/revk.c:22:10: fatal error: esp_int_wdt.h: No such
file or directory
   22 | #include "esp_int_wdt.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
[10/19] Building C object
esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/jo.c.objninja: build
stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the
/home/wes/ESP32-Daikin/build/log/idf_py_stderr_output_25607 and
/home/wes/ESP32-Daikin/build/log/idf_py_stdout_output_25607
make: *** [Makefile:12: all] Error 2

WheresWes, 2022-07-29:
One question.  I see it is compiling for a PICO.

I think I saw some other board options while digging around in the code.
Am I correct that I could change (-S1-PICO) Build suffixe at the bottom of
the RevK submenu of menuconfig to (-S1) for the WROOM ?

revk, 2022-07-29:
OK added that. It is a pain that it is trial and error, but as I am unable to reproduce the problem here, we may have to do a few more of these :-)

When you make, you can `make pico`, or `make wroom`, or `make solo`. Or `make set` will make all three. You don't need to manually change the menu config, that is what the script `components/ESP32-RevK/setbuildsuffix` does for you.

So depends what chip you are using which binary you need. Then end up in the main directory as `Daikin-S1.bin`, `Daikin-S1-PICO.bin`, and  `Daikin-S1-SOLO.bin`.

WheresWes, 2022-07-29:
Ahh very handy.. have run with the wroom suffix.

New errors unfortunately:

Have checked and esp_int_wdt is present in the include/esp_private folder

Same error occurs if running with pico suffix

Make: Daikin-S1.bin
Executing action: all (aliases: build)
Running ninja in directory /home/wes/ESP32-Daikin/build
Executing "ninja all"...
[0/1] Re-running CMake...-- Building ESP-IDF components for target esp32
CMake Error at /home/wes/esp/esp-idf/tools/cmake/build.cmake:249 (message):
  Failed to resolve component 'esp_int_wdt'.
Call Stack (most recent call first):
  /home/wes/esp/esp-idf/tools/cmake/build.cmake:285
(__build_resolve_and_add_req)
  /home/wes/esp/esp-idf/tools/cmake/build.cmake:286
(__build_expand_requirements)
  /home/wes/esp/esp-idf/tools/cmake/build.cmake:580
(__build_expand_requirements)
  /home/wes/esp/esp-idf/tools/cmake/project.cmake:434 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/wes/ESP32-Daikin/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja
/usr/bin/cmake --regenerate-during-build -S/home/wes/ESP32-Daikin
-B/home/wes/ESP32-Daikin/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1, output of the command is in the
/home/wes/ESP32-Daikin/build/log/idf_py_stderr_output_26795 and
/home/wes/ESP32-Daikin/build/log/idf_py_stdout_output_26795
make: *** [Makefile:12: all] Error 2

revk, 2022-07-29:
Ah, that I can reproduce - I'll try and find what i should put in there.

revk, 2022-07-29:
Ok it was `esp_system`, added, try that!

WheresWes, 2022-07-29:
Hope you didnt have any plans for today ! :D

__idf_main.dir/Daikin.c.obj -c ../main/Daikin.c
../main/Daikin.c:10:10: fatal error: driver/gpio.h: No such file or
directory
   10 | #include <driver/gpio.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
[813/823] Building C object
esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/jo.c.objninja: build
stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the
/home/wes/ESP32-Daikin/build/log/idf_py_stderr_output_28095 and
/home/wes/ESP32-Daikin/build/log/idf_py_stdout_output_28095
make[1]: *** [Makefile:12: all] Error 2
make[1]: Leaving directory '/home/wes/ESP32-Daikin'
make: *** [Makefile:26: wroom] Error 2

revk, 2022-07-29:
This is so strange, and just not something I had to do on my machine. I have added `driver` now.

WheresWes, 2022-07-29:
I wish I had more to offer in this...

idf_main.dir/Daikin.c.obj -c ../main/Daikin.c
../main/Daikin.c: In function 'wsend':
../main/Daikin.c:925:10: error: unknown type name 'httpd_ws_frame_t'
  925 |          httpd_ws_frame_t ws_pkt;
      |          ^~~~~~~~~~~~~~~~
../main/Daikin.c:926:36: error: 'httpd_ws_frame_t' undeclared (first use in
this function)
  926 |          memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
      |                                    ^~~~~~~~~~~~~~~~
../main/Daikin.c:926:36: note: each undeclared identifier is reported only
once for each function it appears in
../main/Daikin.c:927:16: error: request for member 'payload' in something
not a structure or union
  927 |          ws_pkt.payload = (uint8_t *) js;
      |                ^
../main/Daikin.c:928:16: error: request for member 'len' in something not a
structure or union
  928 |          ws_pkt.len = strlen(js);
      |                ^
../main/Daikin.c:929:16: error: request for member 'type' in something not
a structure or union
  929 |          ws_pkt.type = HTTPD_WS_TYPE_TEXT;
      |                ^
../main/Daikin.c:929:24: error: 'HTTPD_WS_TYPE_TEXT' undeclared (first use
in this function); did you mean 'HTTPD_TYPE_TEXT'?
  929 |          ws_pkt.type = HTTPD_WS_TYPE_TEXT;
      |                        ^~~~~~~~~~~~~~~~~~
      |                        HTTPD_TYPE_TEXT
../main/Daikin.c:930:10: error: implicit declaration of function
'httpd_ws_send_frame_async' [-Werror=implicit-function-declaration]
  930 |          httpd_ws_send_frame_async(req->handle, fd, &ws_pkt);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
../main/Daikin.c: In function 'web_status':
../main/Daikin.c:942:4: error: unknown type name 'httpd_ws_frame_t'
  942 |    httpd_ws_frame_t ws_pkt;
      |    ^~~~~~~~~~~~~~~~
../main/Daikin.c:944:30: error: 'httpd_ws_frame_t' undeclared (first use in
this function)
  944 |    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
      |                              ^~~~~~~~~~~~~~~~
../main/Daikin.c:945:10: error: request for member 'type' in something not
a structure or union
  945 |    ws_pkt.type = HTTPD_WS_TYPE_TEXT;
      |          ^
../main/Daikin.c:945:18: error: 'HTTPD_WS_TYPE_TEXT' undeclared (first use
in this function); did you mean 'HTTPD_TYPE_TEXT'?
  945 |    ws_pkt.type = HTTPD_WS_TYPE_TEXT;
      |                  ^~~~~~~~~~~~~~~~~~
      |                  HTTPD_TYPE_TEXT
../main/Daikin.c:946:20: error: implicit declaration of function
'httpd_ws_recv_frame' [-Werror=implicit-function-declaration]
  946 |    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 0);
      |                    ^~~~~~~~~~~~~~~~~~~
../main/Daikin.c:949:15: error: request for member 'len' in something not a
structure or union
  949 |    if (!ws_pkt.len)
      |               ^
../main/Daikin.c:951:26: error: request for member 'len' in something not a
structure or union
  951 |    buf = calloc(1, ws_pkt.len + 1);
      |                          ^
../main/Daikin.c:954:10: error: request for member 'payload' in something
not a structure or union
  954 |    ws_pkt.payload = buf;
      |          ^
../main/Daikin.c:955:50: error: request for member 'len' in something not a
structure or union
  955 |    ret = httpd_ws_recv_frame(req, &ws_pkt, ws_pkt.len);
[... 46 lines cut ...]
/home/wes/esp/esp-idf/components/bootloader/subproject/components/micro-ecc
/home/wes/esp/esp-idf/components/newlib
/home/wes/esp/esp-idf/components/partition_table
/home/wes/esp/esp-idf/components/soc
/home/wes/esp/esp-idf/components/spi_flash
/home/wes/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wes/ESP32-Daikin/build/bootloader
[1/1] cd /home/wes/ESP32-Daikin/build/bootloader/esp-idf/esptool_...der
0x1000 /home/wes/ESP32-Daikin/build/bootloader/bootloader.binBootloader
binary size 0x45f0 bytes. 0x2a10 bytes (38%) free.
ninja: build stopped: subcommand failed.
HINT: Maybe you forgot to import httpd_ws_send_frame_async,
httpd_ws_recv_frame library(s) in header file or add the necessary REQURIES
component. Try to add missing libraries to your project header file or
check idf_component_register(REQUIRES ...) section in your component
CmakeList.txt file. For more information run 'idf.py docs -sp
api-guides/build-system.html'.
ninja failed with exit code 1, output of the command is in the
/home/wes/ESP32-Daikin/build/log/idf_py_stderr_output_32565 and
/home/wes/ESP32-Daikin/build/log/idf_py_stdout_output_32565
make[1]: *** [Makefile:12: all] Error 2
make[1]: Leaving directory '/home/wes/ESP32-Daikin'
make: *** [Makefile:26: wroom] Error 2

revk, 2022-07-29:
That is special, it means the includes were picked up but not linked in. I'll see what I can track down.

revk, 2022-07-29:
Ah, I think this is easy. The default sdkconfig was missing `CONFIG_HTTPD_WS_SUPPORT`

You can add with `make menuconfig` under the http server section. I'm updating the default as well. I have also added an error to say if not set.

So `make update`, and `make menuconfig` and try that.

revk, 2022-07-29:
Thanks for your help testing this as a new user - allows me to fix lots of niggles I would never know are there.

WheresWes, 2022-07-29:
ok done,,

../components/ESP32-RevK/revk.c:22:10: fatal error: esp_int_wdt.h: No such
file or directory
   22 | #include "esp_int_wdt.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
[810/820] Building C object
esp-idf/main/CMakeFiles/__idf_main.dir/Daikin.c.objninja: build stopped:
subcommand failed.
ninja failed with exit code 1, output of the command is in the
/home/wes/ESP32-Daikin/build/log/idf_py_stderr_output_33356 and
/home/wes/ESP32-Daikin/build/log/idf_py_stdout_output_33356
make: *** [Makefile:12: all] Error 2

WheresWes, 2022-07-29:
Haha.. I certainly have a nose for niggles.

Thanks for helping me through it all.. I think I'm learning something.

Question.. once the bin file is compiled and I flash it to the ESP, does it
generate an AP to set up network and mqtt etc?  Or does that need to be set
up in menuconfig before making?

revk, 2022-07-29:
Is `esp_int_wdt.h` still an issue? Or have you managed to build now?

Yes it makes an AP which should (on an iPhone at least) auto pop up with a page to set the WiFi and MQTT.

WheresWes, 2022-07-29:
yeah its still an issue... possibly it wasnt resolved last time, just that
another error knocked us out earlier.

Ok I have an android phone so that will be interesting

revk, 2022-07-29:
Good news, that looks like an obsolete include file, so removed. Try again.

WheresWes, 2022-07-29:
[1/1] cd
/home/wes/ESP32-Daikin/build/...Daikin/build/bootloader/bootloader.binBootloader
binary size 0x45f0 bytes. 0x2a10 bytes (38%) free.
ninja: build stopped: subcommand failed.
HINT: Maybe you forgot to import spi_flash_get_chip_size, spi_flash_read,
spi_flash_erase_range, spi_flash_write, esp_efuse_mac_get_default,
tcpip_adapter_get_ip_info library(s) in header file or add the necessary
REQURIES component. Try to add missing libraries to your project header
file or check idf_component_register(REQUIRES ...) section in your
component CmakeList.txt file. For more information run 'idf.py docs -sp
api-guides/build-system.html'.
ninja failed with exit code 1, output of the command is in the
/home/wes/ESP32-Daikin/build/log/idf_py_stderr_output_4561 and
/home/wes/ESP32-Daikin/build/log/idf_py_stdout_output_4561
make: *** [Makefile:12: all] Error 2

revk, 2022-07-29:
Definitely getting closer! Added spi_flash, so try that.

WheresWes, 2022-07-29:
A bit going on here so I have included the full output:

Make: Daikin-S1.bin
Executing action: all (aliases: build)
Running ninja in directory /home/wes/ESP32-Daikin/build
Executing "ninja all"...
[0/1] Re-running CMake...-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/wes/ESP32-Daikin/sdkconfig
Loading defaults file /home/wes/ESP32-Daikin/sdkconfig.defaults...
/home/wes/ESP32-Daikin/sdkconfig.defaults:289
CONFIG_ESP32_DEFAULT_CPU_FREQ_160 was replaced with
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_160
/home/wes/ESP32-Daikin/sdkconfig.defaults:291
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ was replaced with
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ
/home/wes/ESP32-Daikin/sdkconfig.defaults:292 CONFIG_ESP32_SPIRAM_SUPPORT
was replaced with CONFIG_SPIRAM
/home/wes/ESP32-Daikin/sdkconfig.defaults:373
CONFIG_ESP32_ULP_COPROC_ENABLED was replaced with CONFIG_ULP_COPROC_ENABLED
/home/wes/ESP32-Daikin/sdkconfig.defaults:374
CONFIG_ESP32_ULP_COPROC_RESERVE_MEM was replaced with
CONFIG_ULP_COPROC_RESERVE_MEM
/home/wes/ESP32-Daikin/sdkconfig.defaults:375 CONFIG_ESP32_DEBUG_OCDAWARE
was replaced with CONFIG_ESP_DEBUG_OCDAWARE
/home/wes/ESP32-Daikin/sdkconfig.defaults:376 CONFIG_ESP32_BROWNOUT_DET was
replaced with CONFIG_ESP_BROWNOUT_DET
/home/wes/ESP32-Daikin/sdkconfig.defaults:377
CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_0 was replaced with
CONFIG_ESP_BROWNOUT_DET_LVL_SEL_0
/home/wes/ESP32-Daikin/sdkconfig.defaults:385 CONFIG_ESP32_BROWNOUT_DET_LVL
was replaced with CONFIG_ESP_BROWNOUT_DET_LVL
/home/wes/ESP32-Daikin/sdkconfig.defaults:386
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1 was replaced with
CONFIG_NEWLIB_TIME_SYSCALL_USE_RTC_HRT
/home/wes/ESP32-Daikin/sdkconfig.defaults:390
CONFIG_ESP32_RTC_CLK_SRC_INT_RC was replaced with CONFIG_RTC_CLK_SRC_INT_RC
/home/wes/ESP32-Daikin/sdkconfig.defaults:394
CONFIG_ESP32_RTC_CLK_CAL_CYCLES was replaced with CONFIG_RTC_CLK_CAL_CYCLES
/home/wes/ESP32-Daikin/sdkconfig.defaults:395
CONFIG_ESP32_DEEP_SLEEP_WAKEUP_DELAY was replaced with
CONFIG_ESP_SLEEP_DEEP_SLEEP_WAKEUP_DELAY
-- App "Daikin" version: 4fba4ab
-- Adding linker script
/home/wes/ESP32-Daikin/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script
/home/wes/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: ESP32-RevK app_trace app_update bootloader
bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc
esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client
esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd
esp_local_ctrl esp_netif esp_phy esp_pm esp_psram esp_ringbuf esp_rom
esp_serial_slave_link esp_system esp_timer esp_wifi espcoredump esptool_py
fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main
mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c
protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tinyusb ulp
unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/wes/ESP32-Daikin/components/ESP32-RevK
/home/wes/esp/esp-idf/components/app_trace
[... 519 lines cut ...]
/home/wes/esp/esp-idf/components/esp_netif/include/esp_netif_ip_addr.h:52:45:
note: in expansion of macro 'esp_ip4_addr4'
   52 | #define esp_ip4_addr4_16(ipaddr) ((uint16_t)esp_ip4_addr4(ipaddr))
      |                                             ^~~~~~~~~~~~~
/home/wes/esp/esp-idf/components/esp_netif/include/esp_netif_ip_addr.h:57:5:
note: in expansion of macro 'esp_ip4_addr4_16'
   57 |     esp_ip4_addr4_16(ipaddr)
      |     ^~~~~~~~~~~~~~~~
../components/ESP32-RevK/revk.c:2203:49: note: in expansion of macro
'IP2STR'
 2203 |                      jo_stringf(i, "gw", IPSTR, IP2STR(&ip.gw));
      |                                                 ^~~~~~
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
HINT: Maybe you forgot to import spi_flash_get_chip_size, spi_flash_read,
spi_flash_erase_range, spi_flash_write, esp_efuse_mac_get_default,
tcpip_adapter_get_ip_info library(s) in header file or add the necessary
REQURIES component. Try to add missing libraries to your project header
file or check idf_component_register(REQUIRES ...) section in your
component CmakeList.txt file. For more information run 'idf.py docs -sp
api-guides/build-system.html'.
ninja failed with exit code 1, output of the command is in the
/home/wes/ESP32-Daikin/build/log/idf_py_stderr_output_4942 and
/home/wes/ESP32-Daikin/build/log/idf_py_stdout_output_4942
make: *** [Makefile:12: all] Error 2

WheresWes, 2022-07-29:
Off to bed now... will have another crack tomorrow

revk, 2022-07-29:
OK can you try with the latest stable ESP IDF instead of the "Master". That should match what I have. This looks like changes for new version, which is probably best looked at when I upgrade, i.e. when it is next stable release.

WheresWes, 2022-07-29:
Ok.. took a bit for me to figure out how to switch to stable and adjust all
the new dependencies.

But that appears to have done the trick.  The .bin file has built!  outputs
as follows:

ake: Daikin-S1.bin
Executing action: all (aliases: build)
Running ninja in directory /home/wes/ESP32-Daikin/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Component directory /home/wes/esp/esp-idf/components/esp_phy does not
contain a CMakeLists.txt file. No component will be added
-- Component directory /home/wes/esp/esp-idf/components/ieee802154 does not
contain a CMakeLists.txt file. No component will be added
-- Component directory /home/wes/esp/esp-idf/components/openthread does not
contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/wes/ESP32-Daikin/sdkconfig
Loading defaults file /home/wes/ESP32-Daikin/sdkconfig.defaults...
-- Found PythonInterp:
/home/wes/.espressif/python_env/idf4.3_py3.9_env/bin/python (found version
"3.9.2")
-- App "Daikin" version: 4fba4ab
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script
/home/wes/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script
/home/wes/ESP32-Daikin/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/wes/esp/esp-idf/components/esp32/ld/
esp32.project.ld.in
-- Adding linker script
/home/wes/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: ESP32-RevK app_trace app_update asio bootloader
bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls
esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid
esp_http_client esp_http_server esp_https_ota esp_https_server
esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom
esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi
espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test
jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash
openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc
spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs
wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/wes/ESP32-Daikin/components/ESP32-RevK
/home/wes/esp/esp-idf/components/app_trace
/home/wes/esp/esp-idf/components/app_update
/home/wes/esp/esp-idf/components/asio
/home/wes/esp/esp-idf/components/bootloader
/home/wes/esp/esp-idf/components/bootloader_support
/home/wes/esp/esp-idf/components/bt /home/wes/esp/esp-idf/components/cbor
/home/wes/esp/esp-idf/components/cmock
/home/wes/esp/esp-idf/components/coap
/home/wes/esp/esp-idf/components/console
/home/wes/esp/esp-idf/components/cxx
/home/wes/esp/esp-idf/components/driver
/home/wes/esp/esp-idf/components/efuse
/home/wes/esp/esp-idf/components/esp-tls
/home/wes/esp/esp-idf/components/esp32
/home/wes/esp/esp-idf/components/esp_adc_cal
/home/wes/esp/esp-idf/components/esp_common
/home/wes/esp/esp-idf/components/esp_eth
/home/wes/esp/esp-idf/components/esp_event
[... 299 lines cut ...]
[80/86] Linking C static library esp-idf/spi_flash/libspi_flash.a
[81/86] Linking C static library esp-idf/micro-ecc/libmicro-ecc.a
[82/86] Linking C static library esp-idf/soc/libsoc.a
[83/86] Linking C static library esp-idf/hal/libhal.a
[84/86] Linking C static library esp-idf/main/libmain.a
[85/86] Linking C executable bootloader.elf
[86/86] Generating binary image from built executable
esptool.py v3.1-dev
Merged 1 ELF section
Generated /home/wes/ESP32-Daikin/build/bootloader/bootloader.bin
[974/974] Generating binary image from built executable
esptool.py v3.1-dev
Merged 2 ELF sections
Generated /home/wes/ESP32-Daikin/build/Daikin.bin

Project build complete. To flash, run this command:
/home/wes/.espressif/python_env/idf4.3_py3.9_env/bin/python
../esp/esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800
--before default_reset --after hard_reset --chip esp32  write_flash
--flash_mode dio --flash_size detect --flash_freq 40m 0x1000
build/bootloader/bootloader.bin 0x8000
build/partition_table/partition-table.bin 0xd000 build/ota_data_initial.bin
0x10000 build/Daikin.bin
or run 'idf.py -p (PORT) flash'
Done: Daikin-S1.bin

WheresWes, 2022-07-30:
Ok, I was able to successfully flash the bin file to the ESP-WROOM-32 with
the following output

***@***.***:/home/wes/ESP32-Daikin#
/home/wes/.espressif/python_env/idf4.3_py3.9_env/bin/python
../esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB1 -b
460800 --before default_reset --after hard_reset --chip esp32  write_flash
--flash_mode dio --flash_size detect --flash_freq 40m 0x1000
build/bootloader/bootloader.bin 0x8000
build/partition_table/partition-table.bin 0xd000 build/ota_data_initial.bin
0x10000 build/Daikin.bin
esptool.py v3.1-dev
Serial port /dev/ttyUSB1
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding
Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized
freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 0c:b8:15:f6:78:28
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Flash will be erased from 0x00010000 to 0x00119fff...
Compressed 17520 bytes to 11783...
Wrote 17520 bytes (11783 compressed) at 0x00001000 in 0.6 seconds
(effective 221.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 155...
Wrote 3072 bytes (155 compressed) at 0x00008000 in 0.1 seconds (effective
319.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.1 seconds (effective
458.7 kbit/s)...
Hash of data verified.
Compressed 1087664 bytes to 689811...
Wrote 1087664 bytes (689811 compressed) at 0x00010000 in 16.2 seconds
(effective 536.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

However when I serial connect to the esp it is constantly rebooting with
the following error:

I (464) psram: This chip is ESP32-D0WD
E (465) psram: PSRAM ID read error: 0xffffffff
E (465) cpu_start: Failed to init external RAM!
Re-enable cpu cache.

abort() was called at PC 0x40081711 on core 0

Backtrace:0x4008d47e:0x3ffe3bb0 0x4008dcad:0x3ffe3bd0 0x40094e3e:0x3ffe3bf0
0x40                                                     081711:0x3ffe3c60
0x40079007:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: 9dabaef1e66ef496

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1380
load:0x40078000,len:12932
load:0x40080400,len:3120
entry 0x400805ec

I am trying to install onto a NodeMCU-ESP32 Devkit v1

WheresWes, 2022-07-30:
Ok so I had some luck with disabling SPI external ram in menuconfig and reflashing to the esp32.

I now have a wifi AP up.  I can connect to it but dont know what IP address to use for a setup page.

probing the serial connection gives the following:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1380
load:0x40078000,len:12480
ho 0 tail 12 room 4
load:0x40080400,len:3068
entry 0x400805ec
I (413) cpu_start: Pro cpu up.
I (413) cpu_start: Starting app cpu, entry point is 0x4008132c
I (0) cpu_start: App cpu up.
I (427) cpu_start: Pro cpu start user code
I (428) cpu_start: cpu freq: 160000000
I (428) cpu_start: Application information:
I (432) cpu_start: Project name:     Daikin
I (437) cpu_start: App version:      4fba4ab
I (442) cpu_start: Compile time:     Jul 30 2022 14:36:06
I (448) cpu_start: ELF file SHA256:  c2663a50a0da6393...
I (454) cpu_start: ESP-IDF:          v4.3.1
I (459) heap_init: Initializing. RAM available for dynamic allocation:
I (466) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (472) heap_init: At 3FFB8ED8 len 00027128 (156 KiB): DRAM
I (478) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (485) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (491) heap_init: At 40097324 len 00008CDC (35 KiB): IRAM
I (498) spi_flash: detected chip: generic
I (502) spi_flash: flash io: dio
I (507) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (524) RevK: Running from ota_0 at 10000 (size 2031616)
I (529) RevK: Partition table at 8000 as expected (components/ESP32-RevK/partitions_4m.csv)
I (538) RevK: nvs_flash_init
I (575) RevK: nvs_flash_init_partition
I (577) RevK: nvs_open_from_partition
E (583) RevK: Setting blink1 failed Bad number
E (594) RevK: Setting apport failed Bad number
E (626) RevK: Setting tx failed Bad number
E (627) RevK: Setting rx failed Bad number
E (627) RevK: Setting b1 failed Bad number
E (630) RevK: Setting b2 failed Bad number
E (635) RevK: Setting b3 failed Bad number
I (641) wifi:wifi driver task: 3ffc329c, prio:23, stack:6656, core=0
I (646) system_api: Base MAC address is not set
I (651) system_api: read default base MAC address from EFUSE
I (660) wifi:wifi firmware version: 88c8747
I (661) wifi:wifi certification version: v7.0
I (665) wifi:config NVS flash: enabled
I (668) wifi:config nano formating: disabled
I (673) wifi:Init data frame dynamic rx buffer num: 32
I (677) wifi:Init management frame dynamic rx buffer num: 32
I (683) wifi:Init management short buffer num: 32
I (688) wifi:Init static tx buffer num: 16
I (691) wifi:Init static rx buffer size: 1600
I (695) wifi:Init static rx buffer num: 10
I (699) wifi:Init dynamic rx buffer num: 32
I (703) wifi_init: rx ba win: 6
I (707) wifi_init: tcpip mbox: 32
I (711) wifi_init: udp mbox: 6
I (714) wifi_init: tcp mbox: 6
I (718) wifi_init: tcp tx win: 5744
I (722) wifi_init: tcp rx win: 5744
I (726) wifi_init: tcp mss: 1440
I (730) wifi_init: WiFi IRAM OP enabled
I (735) wifi_init: WiFi RX IRAM OP enabled
I (740) wifi_init: WiFi SLP IRAM OP enabled
I (745) wifi:Set ps type: 0

I (749) RevK: WiFi STA [IoT]
I (751) RevK: Dynamic IP start
I (755) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (866) wifi:mode : sta (0c:b8:15:f6:78:28)
I (867) wifi:enable tsf
I (870) RevK: STA Start
I (872) RevK: Up 1, Link down 1
W (873) wifi:Haven't to connect to a suitable AP now!
I (2434) RevK: Up 2, Link down 1
W (2434) wifi:Haven't to connect to a suitable AP now!

revk, 2022-07-30:
If you do a `make wroom` it should remove the `CONFIG_ESP32_SPIRAM_SUPPORT` for you. Is that what had to be manually removed? SPI RAM is built in on on the PICO, so included in the PICO build.

I don't know what mechanism android might use for slash pages - it works a treat on iPhone :-).

However, if your wifi settings when connected to the AP can tell you the "router IP"? That is it. Or find your own IP and connect to the same but ending `.1`

Well done.

revk, 2022-07-30:
Ah I wonder if you had to remove `CONFIG_SPIRAM_SUPPORT`, that could simply be a mistake in the default config.

WheresWes, 2022-07-30:
It was:

"Support for external spi connected ram"
Under the component config/esp32 specific menu.

Does the serial output look OK to you?

revk, 2022-07-30:
I just checked, `make wroom` removes that setting, so that should have worked. Maybe try a `make pico` and then `make menuconfig` to confirm it is back, then a `make wroom` and a `make menuconfig` to confirm it is gone again. I just did that and it seems to be working for me no problem.

If you see an AP then that should be the key here. The serial shows it tries to connect to IoT (which I assume you don't have), and it should make an AP for config.

WheresWes, 2022-07-30:
I'd say you are right.. I did run 'make wroom' at first but then forgot
about it and simply ran 'make'.  Does make remember the previous suffix?
The generated file was - s1

revk, 2022-07-30:
The `make pico`, etc, change the `sdkconfig` file, so yes, remember.

When it makes it tries to guess from the config which it is. The guess for is `pico` is not based on SPI so it was fooled and labelled just `-S1`

WheresWes, 2022-07-30:
Yup got it! Used a network scanner app to find the gateway address.

10.120.40.1

And now I can ssid pass and mqtt

:D

revk, 2022-07-30:
Thanks for the feedback - it has helped improve things. Have fun.

closed 2022-07-30 by revk

antonsb, 2022-12-30:
Hello,

maybe you can help me with building as well?
i'm stuck at:

`
[ 90%] Linking C static library libconsole.a
make[4]: Leaving directory '/home/antons/Documents/ESP32-Daikin/build'
[ 90%] Built target __idf_console
make[3]: Leaving directory '/home/antons/Documents/ESP32-Daikin/build'
make[2]: *** [Makefile:136: all] Error 2
make[2]: Leaving directory '/home/antons/Documents/ESP32-Daikin/build'
HINT:esp_mac.h header file is not included by esp_system.h anymore. It shall then be manually included with #include "esp_mac.h"HINT: Please check that the function name is correct. Also it is possible that you've forgot to import spi_flash_get_chip_size library(s) in header file or add the necessary REQURIES component. Try to add missing libraries to your project header file or check idf_component_register(REQUIRES ...) section in your component CmakeList.txt file. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the function has been removed, renamed or replaced by an alternative function - refer to the migration guide for more information.
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
make failed with exit code 2, output of the command is in the /home/antons/Documents/ESP32-Daikin/build/log/idf_py_stderr_output_675642 and /home/antons/Documents/ESP32-Daikin/build/log/idf_py_stdout_output_675642
make[1]: *** [Makefile:33: all] Error 2
make[1]: Leaving directory '/home/antons/Documents/ESP32-Daikin'
make: *** [Makefile:47: wroom] Error 2
`

revk, 2022-12-30:
I have included a binary image anyway. I suspect it needs the slightly older ESP IDF, so I need to update to latest and tidy any changes needed. I'll see what I can do over the weekend maybe.