revk/xESP32-Faikout #3: AP not acting on entered parameters

WheresWes, 2022-07-30:
I'm back.. just when you thought you'd seen the end of me.

Firmware is successfully flashed to ESP board.
ESP board creates AP
I connect to the ap and find the splash screen at 10.120.40.1
This give three fields for SSID. Password and MQTT
I enter in my network ssid and password and the ip address of the MQTT broker in the last field (does this require a port also?)
However when I hit the set button nothing happens.  The set button disappears.
If I refresh the page the entered parameters have reverted back to their original state.  The board does not connect to network.

Have tried with Android - Chrome, and Windows - Chrome & Edge

revk, 2022-07-30:
It should try and connect, and even report the IP to which it connects.

I don't have android to test here sadly, but this is very simple web socket stuff so should work with any browser on any device.

This may need me to find someone with an android to test for me.

It would also be possible to add some debug over serial.

For now, you can use `make menuconfig` to set the defaults for wifi and mqtt under the component RevK settings, and work around it.

WheresWes, 2022-07-30:
Ok I will give that a go... I just dusted off an old iphone to try that out.. it does find the splash by itself but nothing happens when the parameters are entered (there is no set button)

revk, 2022-07-30:
I'll re-test here anyway. Maybe something has broken more recently. This should probably be on the ESP32-RevK project anyway.

WheresWes, 2022-07-30:
I set the parameters in menuconfig and it is happy now.

Sing out if you would like me to test anything

revk, 2022-07-30:
I'll raise an issue to re-test this all on ESP32-RevK project, thanks.

closed 2022-07-30 by revk