revk/xESP32-Faikout #5: Is this board simulating the Daikin local web api? or is it mqtt only?

Amtho, 2022-09-07:
please see above in the titel

revk, 2022-09-07:
No, but no reason that could not be added. It is working over MQTT, with all of the options you might want.

If you feel like creating the old local web API on it, I'm happy to look at a pull request. It looks pretty simple, and my code has all the hooks to provide the data and take actions.

closed 2022-09-07 by revk

BenJamesAndo, 2024-01-18:
I did some digging and found some local web API commands to do such things as turning it on and off. I think most of it was added after this issue was opened. I only got my Faikin yesterday so still new to this.

I documented it here: https://github.com/revk/ESP32-Faikin/wiki/Example-HTTP-API
@revk hopefully that's alright me adding it in there.

I'm personally using it over MQTT but I did want the flexibility to change values with HTTP. I was curious to see if I could add the Faikin into the official Daikin phone app but it didn't work. I did however add it to the official Daikin Home Assistant integration. Meaning you could use the Faikin without using MQTT inside of Home Assistant.