revk/xESP32-Faikout #6: Ubuntu ninja subcommand compile error

boehla, 2022-09-24:
Hello RevK,

I tried to build with "make pico", but I always get an error and I don't know how to find out, what the problem is.
Do you have an idea? 

Trying with Ubuntu 22.04 LTS.
Installed ESP IDF: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html
> mkdir -p ~/esp
> cd ~/esp
> git clone --recursive https://github.com/espressif/esp-idf.git
> cd ~/esp/esp-idf
> ./install.sh esp32
> . $HOME/esp/esp-idf/export.sh

Compiling example IDF Project works:
> simma@simma:~/esp/esp-idf/examples/get-started/hello_world$ idf.py build

Cloning:
> git clone --recursive https://github.com/revk/ESP32-Daikin.git

Installed all the required modules while compiling (mqtt, mysql, ...)

When ESP32-Daikin compiling, there are a lot of warning & errors:
> simma@simma:~/tmp1/ESP32-Daikin$ make pico
components/ESP32-RevK/setbuildsuffix -S1-PICO
make[1]: Entering directory '/home/simma/tmp1/ESP32-Daikin'
Make: Daikin-S1-PICO.bin
Executing action: all (aliases: build)
Running ninja in directory /home/simma/tmp1/ESP32-Daikin/build
Executing "ninja all"...
[1/1] cd /home/simma/tmp1/ESP32-Daikin/build/bootloader/esp-idf/esptool_py && /home/simma/.espressif/python_env/idf5...._table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/simma/tmp1/ESP32-Daikin/build/bootloader/bootloader.binBootloader binary size 0x45e0 bytes. 0x2a20 bytes (38%) free.
[3/13] Building C object esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/lwmqtt.c.objFAILED: esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/lwmqtt.c.obj
/home/simma/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DBUILD_ESP32_USING_CMAKE -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -I/home/simma/tmp1/ESP32-Daikin/build/config -I/home/simma/tmp1/ESP32-Daikin/components/ESP32-RevK/include -I/home/simma/esp/esp-idf/components/newlib/platform_include -I/home/simma/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/simma/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/simma/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/simma/esp/esp-idf/components/freertos/esp_additions/include -I/home/simma/esp/esp-idf/components/esp_hw_support/include -I/home/simma/esp/esp-idf/components/esp_hw_support/include/soc -I/home/simma/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/simma/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/simma/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/home/simma/esp/esp-idf/components/heap/include -I/home/simma/esp/esp-idf/components/log/include -I/home/simma/esp/esp-idf/components/soc/include -I/home/simma/esp/esp-idf/components/soc/esp32/. -I/home/simma/esp/esp-idf/components/soc/esp32/include -I/home/simma/esp/esp-idf/components/hal/esp32/include -I/home/simma/esp/esp-idf/components/hal/include -I/home/simma/esp/esp-idf/components/hal/platform_port/include -I/home/simma/esp/esp-idf/components/esp_rom/include -I/home/simma/esp/esp-idf/components/esp_rom/include/esp32 -I/home/simma/esp/esp-idf/components/esp_rom/esp32 -I/home/simma/esp/esp-idf/components/esp_common/include -I/home/simma/esp/esp-idf/components/esp_system/include -I/home/simma/esp/esp-idf/components/esp_system/port/soc -I/home/simma/esp/esp-idf/components/esp_system/port/include/private -I/home/simma/esp/esp-idf/components/xtensa/include -I/home/simma/esp/esp-idf/components/xtensa/esp32/include -I/home/simma/esp/esp-idf/components/lwip/include -I/home/simma/esp/esp-idf/components/lwip/include/apps -I/home/simma/esp/esp-idf/components/lwip/include/apps/sntp -I/home/simma/esp/esp-idf/components/lwip/lwip/src/include -I/home/simma/esp/esp-idf/components/lwip/port/esp32/include -I/home/simma/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/simma/esp/esp-idf/components/nvs_flash/include -I/home/simma/esp/esp-idf/components/spi_flash/include -I/home/simma/esp/esp-idf/components/app_update/include -I/home/simma/esp/esp-idf/components/bootloader_support/include -I/home/simma/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/simma/esp/esp-idf/components/esp_app_format/include -I/home/simma/esp/esp-idf/components/esp_http_client/include -I/home/simma/esp/esp-idf/components/esp-tls -I/home/simma/esp/esp-idf/components/esp-tls/esp-tls-crypto -I/home/simma/esp/esp-idf/components/mbedtls/port/include -I/home/simma/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/simma/esp/esp-idf/components/mbedtls/mbedtls/library -I/home/simma/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/simma/esp/esp-idf/components/esp_http_server/include -I/home/simma/esp/esp-idf/components/http_parser -I/home/simma/esp/esp-idf/components/esp_wifi/include -I/home/simma/esp/esp-idf/components/esp_event/include -I/home/simma/esp/esp-idf/components/esp_phy/include -I/home/simma/esp/esp-idf/components/esp_phy/esp32/include -I/home/simma/esp/esp-idf/components/esp_netif/include -I/home/simma/esp/esp-idf/components/esp_timer/include -I/home/simma/esp/esp-idf/components/driver/include -I/home/simma/esp/esp-idf/components/driver/deprecated -I/home/simma/esp/esp-idf/components/driver/esp32/include -I/home/simma/esp/esp-idf/components/esp_pm/include -I/home/simma/esp/esp-idf/components/esp_ringbuf/include -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Og -fmacro-prefix-map=/home/simma/tmp1/ESP32-Daikin=. -fmacro-prefix-map=/home/simma/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.1-dev-862-g09f7589ef2\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/lwmqtt.c.obj -MF esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/lwmqtt.c.obj.d -o esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/lwmqtt.c.obj -c /home/simma/tmp1/ESP32-Daikin/components/ESP32-RevK/lwmqtt.c
In file included from /home/simma/tmp1/ESP32-Daikin/components/ESP32-RevK/lwmqtt.c:21:
/home/simma/tmp1/ESP32-Daikin/components/ESP32-RevK/lwmqtt.c: In function 'client_task':
/home/simma/esp/esp-idf/components/log/include/esp_log.h:265:27: error: format '%d' expects argument of type 'int', but argument 7 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  265 | #define LOG_COLOR(COLOR)  "\033[0;" COLOR "m"
      |                           ^~~~~~~~~
/home/simma/esp/esp-idf/components/log/include/esp_log.h:268:27: note: in expansion of macro 'LOG_COLOR'
  268 | #define LOG_COLOR_E       LOG_COLOR(LOG_COLOR_RED)
      |                           ^~~~~~~~~
/home/simma/esp/esp-idf/components/log/include/esp_log.h:282:37: note: in expansion of macro 'LOG_COLOR_E'
  282 | #define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
      |                                     ^~~~~~~~~~
/home/simma/esp/esp-idf/components/log/include/esp_log.h:410:86: note: in expansion of macro 'LOG_FORMAT'
  410 |         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                                                      ^~~~~~~~~~
/home/simma/esp/esp-idf/components/log/include/esp_log.h:432:41: note: in expansion of macro 'ESP_LOG_LEVEL'
  432 |         if ( LOG_LOCAL_LEVEL >= level ) ESP_LOG_LEVEL(level, tag, format, ##__VA_ARGS__); \
      |                                         ^~~~~~~~~~~~~
/home/simma/esp/esp-idf/components/log/include/esp_log.h:342:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  342 | #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
/home/simma/tmp1/ESP32-Daikin/components/ESP32-RevK/lwmqtt.c:736:7: note: in expansion of macro 'ESP_LOGI'
  736 |       ESP_LOGI(TAG, "Waiting %d (mem:%d)", handle->backoff, esp_get_free_heap_size());
      |       ^~~~~~~~
[... cut ...]
HINT: esp_mac.h header file is not included by esp_system.h anymore. It shall then be manually included with #include "esp_mac.h"
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
ninja failed with exit code 1, output of the command is in the /home/simma/tmp1/ESP32-Daikin/build/log/idf_py_stderr_output_92881 and /home/simma/tmp1/ESP32-Daikin/build/log/idf_py_stdout_output_92881
make[1]: *** [Makefile:12: all] Error 2
make[1]: Leaving directory '/home/simma/tmp1/ESP32-Daikin'
make: *** [Makefile:22: pico] Error 2

boehla, 2022-11-25:
Also tried it with ubuntu 20 and this idf setup: https://gist.github.com/abobija/2721427f18d78e439fa9875cac960ffa

Looks like the same issue again.. :(
What could I do wrong? :S

Do you (@revk) want to release a finished bin file for flashing ? :)

revk, 2022-11-26:
It really should just work. I'll have a look, and maybe add some binaries.

revk, 2022-11-26:
OK I have added binaries anyway.

closed 2022-11-26 by revk