revk/xESP32-Faikout #8: esp32 wroom stuck and bootloop

antonsb, 2023-01-08:
Hi,

now code builds, but wroom is stuck in boot-loop after flashing.
`22:06:40.056 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:06:40.056 -> configsip: 0, SPIWP:0xee
22:06:40.056 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:06:40.089 -> mode:DIO, clock div:2
22:06:40.089 -> load:0x3fff0030,len:1236
22:06:40.089 -> load:0x40078000,len:13620
22:06:40.089 -> load:0x40080400,len:2916
22:06:40.089 -> entry 0x40080590
22:06:40.553 -> [0;32mI (475) quad_psram: This chip is ESP32-D0WD[0m
22:06:40.553 -> [0;31mE (478) quad_psram: PSRAM ID read error: 0xffffffff[0m
22:06:40.553 -> [0;31mE (478) cpu_start: Failed to init external RAM![0m
22:06:40.553 -> 
22:06:40.553 -> abort() was called at PC 0x400818c4 on core 0
22:06:40.553 -> 
22:06:40.553 -> 
22:06:40.553 -> Backtrace: 0x40082469:0x3ffe3b90 0x4008f449:0x3ffe3bb0 0x40096445:0x3ffe3bd0 0x400818c4:0x3ffe3c40 0x4007935a:0x3ffe3c90 |<-CORRUPTED
22:06:40.553 -> 
`

revk, 2023-01-08:
The wroom build should not be trying PSRAM. Are you sure you are using the wroom binary or doing make wroom?

If you are I'll take a look.

antonsb, 2023-01-08:
I'm using make wroom.

step by step:
* Recursive clone
* make menuconfig -> disable signed security
* make wroom
* idf.py -p /dev/ttyUSB0 flash

revk, 2023-01-08:
The idf used to not care if the PSRAM was not present and seems it now does. But the script to make wroom was changed to turn it off. So must be something silly. I have some wroom here so can check in the morning.

closed 2023-01-09 by revk

antonsb, 2023-01-10:
sorry for bombarding You with questions, but now it resets in different stage:
`
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1256
load:0x40078000,len:13620
load:0x40080400,len:3032
entry 0x40080598
[0;32mI (467) cpu_start: Pro cpu up.[0m
[0;32mI (467) cpu_start: Starting app cpu, entry point is 0x40081414[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (481) cpu_start: Pro cpu start user code[0m
[0;32mI (481) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (481) cpu_start: Application information:[0m
[0;32mI (486) cpu_start: Project name:     Daikin[0m
[0;32mI (491) cpu_start: App version:      a96fb90-dirty[0m
[0;32mI (496) cpu_start: Compile time:     Jan 10 2023 20:12:18[0m
[0;32mI (502) cpu_start: ELF file SHA256:  3080fdf8fb66f65a...[0m
[0;32mI (508) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND[0m
[0;32mI (514) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (521) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (527) heap_init: At 3FFB8AC8 len 00027538 (157 KiB): DRAM[0m
[0;32mI (534) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (540) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (546) heap_init: At 40097C30 len 000083D0 (32 KiB): IRAM[0m
[0;32mI (554) spi_flash: detected chip: generic[0m
[0;32mI (557) spi_flash: flash io: dio[0m
[0;32mI (563) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (579) RevK: Running from ota_0 at 10000 (size 2031616)[0m
[0;32mI (585) RevK: Partition table at 8000 as expected (components/ESP32-RevK/partitions_4m.csv)[0m
[0;32mI (593) RevK: nvs_flash_init[0m
[0;32mI (626) RevK: nvs_flash_init_partition[0m
[0;32mI (651) RevK: nvs_open_from_partition[0m
[0;31mE (936) task_wdt: esp_task_wdt_init(594): TWDT already initialized[0m
[0;32mI (937) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
ets Jun  8 2016 00:22:57
`

i'm i missing something?

revk, 2023-01-10:
OK then what? That looks like it got as far as setting up some GPIO. It should have gone on and done things.

antonsb, 2023-01-10:
this pasted part repeats (loops over)

revk, 2023-01-10:
Hmm, my last test on an WROOM worked, but just tested now, and I see the same. Not very helpful...

Ah, OK, looks like default GPIO for Daiking board class with SPI on WROOM. I'll add some checks and test.

revk, 2023-01-10:
OK I have changed the RevK library to avoid such GPIO for blink, that should then avoid a clash and allow to run, works here.

revk, 2023-01-10:
[image]

revk, 2023-01-10:
I have also fixed the fact the config tx and rx GPIO are not picked up as well.

revk, 2023-01-10:
Always good to have someone testing this. Thanks.

Hopefully that is it now.

The good news is I think I can get some boards made now though so we may be selling some soon.

commit 2318c58e (2023-01-09): Update library so wroom build works, fix #8