revk/xESP32-Faikout #10: Automatically work out if S21

revk, 2023-01-10:
Try initial comms as S21 or not and work it out, rather than a config.

antonsb, 2023-01-11:
Does this means that S21 socket interconnection may be supported as well?
Currently my BRP069C4X has very small connector that i have problem to find online as well as connect pins to it without cutting the wire - thus voiding warranty on the unit. But there is a free S21 slot

revk, 2023-01-11:
There is an MQTT based setting for S21 mode, but I really should make it auto detect some time. Both types of unit I have had a WiFi connect lead and the Daikin WiFi so was easy.

antonsb, 2023-01-11:
lead goes to mine as well, but it's small JST SHJ 1mm pitch type connector (on both sides).On PCB it does not connect to S21,but to some other port module looks like this:
[image]

antonsb, 2023-01-11:
If it worked from S21,it would be much easier (at least for my case) to make custom wire harness, then to hunt down that small connector.

revk, 2023-01-11:
I can look in to making it auto detect soon. But I can probably sort a make menuconfig for S21 in the mean time. Probably in the morning.

antonsb, 2023-01-11:
Damn that's fast. I just ordered JST EH for S21 from Ali, not sure when they will arrive and i will be able to test this.

revk, 2023-01-11:
Well I hope it works. I'll let you know when I have tested.

revk, 2023-01-12:
I have added the S21 setting to make menuconfig now. Like the other settings this is just the default and can be overridden using MQTT, but it should allow you to build an S21 based build to try.

antonsb, 2023-01-12:
build seems to be working for me with S21 option, i would check functionality next week as i will get some connectors for cabling

revk, 2023-01-20:
Ok not working very well...

closed 2023-01-20 by revk

revk, 2023-01-20:
It seems to be working well now, and clean from power up as well.

Only concern is the non S21 model could, with a lot of poking crap at it, get in to a state of just reporting errors. This may mean we need a null header or some such. Something to consider if the problem is seen in the wild.

commit aeee5d61 (2023-01-20): Fix #10