revk/xESP32-Faikout #12: ESP32 wroom ignores menucofig wireless settings

antonsb, 2023-01-12:
I'm not that familiar with ESP chipsets, but it seems that wireless settings are not updated from menuconfig. On first build they are saved into some ESP32 memory that is not erased after reflash and ESP tries to connect on AP that was saved previously.  I even tried writing different bin files and new git clones.

revk, 2023-01-12:
The menuconfig settings are the initial/default. If you have set them then yes it will connect to the saved previously settings, obviously. If it cannot connect it should got to AP mode and allow new settings via a web interface.

revk, 2023-01-12:
You can idf.py erase-flash to clear them.

revk, 2023-01-12:
The menu config even makes it clear:
                This defines the factory reset default WiFi SSID
                This defines the factory reset default WiFi passphrase

closed 2023-01-12 by revk

antonsb, 2023-01-12:
sorry missed that part.
Thanks for clarification.