revk/xESP32-Faikout #16: Help BRP069A81

gabyhaos, 2023-02-17:
Good morning,
A month ago they installed Daikin through ducts (ADEA-A / ARXM-R), I have a home assistant and my idea was to integrate it with my home automation but the model they sell is the brp069c81 only works with the cloud. The brp069a81 model that is discontinued was compatible but after a month of calling all the distributors in my country I have not been able to find one. Today looking at the HA forum I see your github and I think I see the light at the end of the tunnel. I am not very clear on how to install it since I am not a programmer, but I do have some knowledge. I have some questions.
Is any esp32 board worth it? (I have lylygo t7 v1.5)
I assume it is compatible with my motherboard? (I can send photo)
What is the wiring diagram?
I think so but is it possible to integrate it with HA?
Thank you very much for your work and I hope I can achieve it.
I'm sorry if you don't understand something, English is not my native language.
Regards Gabi

revk, 2023-02-17:
This may be useful, https://www.amazon.co.uk/dp/B0BTZ4F5FH assuming amazon can ship to you. It just plugs in, see the picture.

closed 2023-02-17 by revk

gabyhaos, 2023-02-17:
Nice product, but it cant be shipped to Spain, where i live. Any other website to buy it?

revk, 2023-02-17:
I don't know, and would actually be really interested to know if it can be. I think Amazon can sort that, but see what happens. The main idea of putting on Amazon was to sort shipping.

gabyhaos, 2023-02-17:
It says "Currently unavailable.
We don't know when or if this item will be back in stock. "

revk, 2023-02-17:
Oh, shows as in stock for me, but only started showing today, may be worth checking in a day or two.

gabyhaos, 2023-02-17:
ok, i ll check. Thanks, and implementation with ha with mqqt broker would be too hard?

revk, 2023-02-17:
I don't know as I don't have them, but let me know. I have tried to support the same URLs the old Daikin wifi modules used, so they may just work. Failing that there is probably a way to make them work via MQTT.

JeanOlivier, 2023-03-22:
@gabyhaos I have a brp069b41 if you're interested. It turns out not to be compatible with my AC unit.

revk, 2023-03-22:
By the way, I have done the Home Assistant MQTT stuff now on my module.