revk/xESP32-Faikout #18: Not fully working on S21

urbozz, 2023-03-05:
Just bought a faikin from Amazon, nice! Plugged into two different units via S21 port, one 10 years old (FDXS50CVMB), the other a few years (FVXG25K2V1B). Same symptoms on both.

Unit powers up, configured, MQTT messages receiving. 

Some values are reporting, such as Power, Mode, Fan, Set Temp. However Temp and Coil are undefined, and setting any values does not work (so unit will not turn on, mode will not set, fan selector does nothing)

MQTT reporting:

Message 1166 received on error/bedroom-daikin/comms at 21:03:
{
    "s21": true,
    "nak": true
}

Any tips for the next steps?

revk, 2023-03-06:
Sorry to hear that.

Can you confirm what works via the web interface and what does not?

And what exactly you are sending in MQTT to control it?

Thanks.

urbozz, 2023-03-06:
I am using the web interface to control, not yet done anything with MQTT other than only reading the status messages.

I will get some debug output later today.

urbozz, 2023-03-06:
The only two json payloads I see when running debug

{"s21":true,"cmd":"G5","payload":"30303080","text":"000\x80"}

{"s21":true,"cmd":"G1","payload":"30344241","text":"04BA"}

Changing any settings via web or MQTT (for example command/bedroom-daikin/heat command/bedroom-daikin/on) makes no difference to the payload.

urbozz, 2023-03-06:
Also just noticed

error/bedroom-daikin/failed-set
{"power":true}

Shortly after turning on

revk, 2023-03-06:
Ok that means we are sending what we think is the command to turn on the power but not seeing the response saying the power is on.

From my notes, the command (in debug) we are looking for to turn on power is D1, can you see if that is logged at all when you try?

The response codes for this are G1, and I see you have 04BA which I think means power odd, heat, 19.0C, Auto fan.

I think the D1 should therefore be 14BA. See if you can capture more of the process of trying to turn on.

Also, turn on with remote and see if the G1 changes.

urbozz, 2023-03-06:
>  04BA which I think means power odd, heat, 19.0C, Auto fan.
Correct - this is what the unit is set as

G2 changes to {"s21":true,"cmd":"G1","payload":"31344241","text":"14BA"} when you turn it on from the remote.

I don't see any D1 being sent.

revk, 2023-03-06:
OK not seeing D1 is interesting, I'll try and reproduce it here.

urbozz, 2023-03-06:
Ok, get the following on rx on the latest code:

info/bedroom-daikin/tx
{"s21":true,"error":"Cannot send as not talking yet","cmd":"D1","payload":"31344641","text":"14FA"}

And now the errors that are sent every second

error/bedroom-daikin/comms
{"s21":true,"error":"Cannot send as not talking yet","nak":true}

revk, 2023-03-06:
Ok that is what I thought, I need a larger log, showing what the `"nak"` is responding to so I can make a change to the code, probably why no temp/coil temp even.

revk, 2023-03-06:
I have made new code so the "nak" logs what command was being tried. That should help. Try an upgrade. We are getting closer...

urbozz, 2023-03-06:
Web interface now not working, but a few messages seen on MQTT

info/bedroom-daikin/cannot-tx
{"s21-try":false,"cmd":"CB","payload":"0000"}
{"s21-try":false,"cmd":"CA","payload":"0000000000000000000000000000000000"}

info/bedroom-daikin/tx
{"s21-try":false,"cmd":"AA","payload":"01"}

error/bedroom-daikin/comms
{"s21-try":true,"cmd":"F1","payload":"","text":null,"noack":true}
{"s21-try":false,"timeout":true}

urbozz, 2023-03-06:
Led is blue - one long one short blink

revk, 2023-03-06:
OK that looks like it is not talking at all now. It may start talking, or we may have upset the aircon (maybe power cycle that). It started trying s21 alternating with another protocol. Normally works it out. If it is not, we may need a more complex back off. Maybe unplug it for a minute and plug back in.

It needs `"s21":true `showing meaning it has started talking. Then we can see the clean `"nak"` error.

revk, 2023-03-06:
I've made code that does not get upset at a `"nak"` for now, so you can try that. I'll be AFK for a bit now.

urbozz, 2023-03-06:
Since the last two OTA updates, the unit is not contactable after a few seconds of power-up. The only way to get access (inc ping) is to power up without data pins attached.

I see the messages as my prior message before the unit is unreachable, and blue led flashes blue long followed by short.

Using the BRP069B41 works still, so the aircon unit is not that upset it seems.

revk, 2023-03-06:
That is rather worrying. If you then attach pins does it stop working?

urbozz, 2023-03-06:
Latest update works! (can turn on and coil and temp are now showing). Will run through all over settings shortly.

Did notice MQTT messages
error/bedroom-daikin/comms
{"s21":true,"cmd":"F7","nak":true}
{"s21":true,"cmd":"F6","nak":true}

revk, 2023-03-06:
OK good news, and explains the problems, my Daikin handles F6 and F7.

- F6 appears to be a "powerful" setting
- F7 appears to be an "econo" setting

I am guessing you don't have these. It caused the system to think S21 was not working at all, and giving up over and over again.

I'll look in to an improved logic so it is not constantly trying these.

closed 2023-03-06 by revk

urbozz, 2023-03-06:
Yep, I do have econo and powerful on this unit. Can open a separate issue on this another time. Thank you!

urbozz, 2023-03-06:
Errors stop after several attempts. Nice!

The FVXG25K2V1B has Econo, Powerful, Quiet mode (to make the external unit quiet) and also uniquely two radiant modes.

revk, 2023-03-06:
OK, latest has a setting morepoll, send setting/bedroom-daikin/morepoll 1 to turn on, it does a load more.

Then try changing some of those settings, and see if any of the things being possible give different results when you change settings, usually a 0 to a 1 for off to on.

Let me know, and I can add more settings.

urbozz, 2023-03-06:
I presume with debug mode on too?

I see some changes - but not confident it is what we need (as some 'SL' text duplicate) reading info/bedroom-daikin/rx

Should I be looking anywhere else?

Powerful:
Off
{"s21":true,"cmd":"SL","payload":"393131","text":"911"}
{"s21":true,"cmd":"SX","payload":"3031322B","text":"012+"}
On
{"s21":true,"cmd":"SL","payload":"373431","text":"741"}
{"s21":true,"cmd":"SX","payload":"3032332B","text":"023+"}

Econ
Not seeing any changes

Night (G1 not updating, flag remains as A)
Off
{"s21":true,"cmd":"SL","payload":"393131","text":"911"}
On
{"s21":true,"cmd":"SL","payload":"343830","text":"480"}

Quiet:
Off
{"s21":true,"cmd":"SL","payload":"393131","text":"911"}
On
{"s21":true,"cmd":"SL","payload":"383131","text":"811"}

Radiant 1
Off
{"s21":true,"cmd":"SL","payload":"393131","text":"911”}
On
{"s21":true,"cmd":"SL","payload":"343830","text":"480"} 
Radient 2
{"s21":true,"cmd":"SL","payload":"373730","text":"770"}

revk, 2023-03-06:
Look at the `/rx` is great, yes - really a question of how duplicatable. Doing off, on, off, on, off, etc...

It may be several things in one field, so SL for a number of different things. That said, what you show there is rather confused.

That said SL could be some other thing, like fan speed or something, so could be a side effect of what you are doing.

urbozz, 2023-03-06:
I have a feeling SL is a side effect. Not seeing any other changes, maybe the other features are not supported on the S21 interface, albeit the Night Fan speed is a mystery why that is not part of G1/D1 on this unit...

I'll take another look later in case I missed anything.

revk, 2023-03-06:
Night fan mode should be there. Is that not showing on Web interface under fan? Should be 1 to 5, auto and night.

urbozz, 2023-03-06:
It shows in the web interface, but the unit doesn't accept the N flag, so when setting the unit to Night, the flag in D1 shows as A

error/bedroom-daikin/comms
{"s21":true,"cmd":"D1","payload":"30344651","text":"04FQ","nak":true}

urbozz, 2023-03-06:
When on night mode:
{"s21":true,"cmd":"G1","payload":"31344841","text":"14HA"}

This explains maybe why using the BRP069B41 it's possible to set the mode, but never reads the mode if you later check on the setting. But I have been able to set night mode using the BRP069B41.

revk, 2023-03-06:
OK so that may be a weird one, I have changed to B, it will "fail" as not returning B. I wonder if it manages to set it. Can you "tell" is it actually night mode?

revk, 2023-03-06:
Ok see if the new code does that - sending B is consistent with the old http stuff I think. Not sure why I sent Q, as I thought I dumped this from an S21 interface. But indeed, I do not see it report back as B or Q and a nak on the send suggests it is not accepting it.

Only alternative is we dump what the old wifi module sent. That is not impossible to set up, but fiddly.

urbozz, 2023-03-06:
That works, night mode sets, but then reports back auto, even though it is running as night mod.

revk, 2023-03-06:
Ok so same as Daikin WiFi module in that respect, interesting, ands stupid, but best we can do I expect. Thanks.