revk/xESP32-Faikout #20: List of supported AC models or equivalent WiFi adapters?

JeanOlivier, 2023-03-20:
I have a Daikin CTXS09HVJU unit that does not seem to work with a BRP069B41 WiFi adapter, but should apparently work with a BRP072A43.
However I've been completely unable to find the latter anywhere online, so I digged online about the S21 protocol and found this very neat project.

Would this project work with my unit? If not clear, how can I help you test this?

revk, 2023-03-20:
Basically I have no way to know. I have tried on three different modules, and a customer has tried on one, but I don't actually have any list of model numbers - perhaps we should start such a list.

If it talks the S21 protocol, I think it should work.

I have the boards on amazon, but a simple ESP32 dev / test board you can get will probably work well enough if you can power it. The Daikin seems to feed something like 12V, and the tx/rx are 5V, so not ideal.

I have my boards on Amazon (UK).

closed 2023-03-20 by revk

JeanOlivier, 2023-03-20:
@revk I can't seem to find it on Amazon UK, can you provide a link or a search term that allow me to find it?

revk, 2023-03-20:
[image]

Like this link in the README.md on this project?

revk, 2023-03-20:
[image]

It is only on Amazon UK, sorry - we have not figured out quite how we can export stuff yet.

revk, 2023-03-20:
I would live to, but it seems not so simple. Most requests I have had have been from outside UK.

JeanOlivier, 2023-03-20:
I'll try to order through an English friend.
If I want to debug as well in case my unit isn't directly compatible, do you sell the required board as well?

I'd of course contribute any findings here.