revk/xESP32-Faikout #22: Home Assistant - Consistency to official Daikin Integration

urbozz, 2023-03-21:
The Faikin now just appears in HA with MQTT. Nice!

Few Suggestions that would be nice to have to keep consistency with regards to icons and descriptions with other HA Climate & Daikin integration (some screenshots of official Daikin attached). 

Also would be nice to select Sienet as Fan mode.

Climate Entity: mdiThermostat
Temperature Entities: mdiThermometer

[image]
[image]
[image]
[image]
[image]

revk, 2023-04-14:
The fan mode is a problem - the quite/silent/night mode appears to be possible to set but not report - if you find anything in S21 messages that covers this, do let me know.

revk, 2023-04-16:
If you know the HA JSON config in use in these other cases, can you paste it, that will help work out what changes can be made to be consistent.

revk, 2023-05-28:
I'll close for now - if you can find the config/JSON for HA for this, I am happy to change.

closed 2023-05-28 by revk