revk/xESP32-Faikout #23: P1P2 Protocol Supported?

erenoglu, 2023-03-29:
Hi, I have a ducted unit which works with P1P2 two wire communications. It's a Ducted split unit but using same P1P2 as Daikin VRV systems.

There are 3rd party Wifi units working with this (ie Airzone) which need other thermostat to be operational as well. 

Is your ESP32-Daikin support P1P2 or only S21?

erenoglu, 2023-03-29:
My ducted has P1P2 ports and it's connected to a wall thermostat (Daikin Madoka). I'm currently using it with bluetooth integration but I like to have more native wifi based connection.
So how can we confirm if the ducted P1P2 would work? What is your ducted AC model? I can check, or maybe you can check behind your wall thermostat or indoor unit panel if the cables are connected to P1P2?

If it's P1P2 for data, I guess power need to fed separately? Based on this I'll order 1 or 3 units. I have one wall unit with S21 and 2 ducted with P1P2.

revk, 2023-03-29:
Yes, I have a Modoka, and seems still needed. But the wifi connects to some different port on the device.

Best I can suggest is I take some pictures - bear with me.

erenoglu, 2023-03-29:
That's great, the pics are not very clear, can you identify the ports on the indoor units, is the power X35A and data goes to X50H? I can't read properly.  I will check if I have the same on my units. 

Also, I see power are 2 cables, but then to X50H(?) you have 3 cables going in, so total 5,  while the lead has 4 cables?

revk, 2023-03-29:
Those are right by the look of it.

Yes, I assume the X50H has a GND that is not used. It is a 5 pin plug with 5 wires. That seems to be the same for S21 as well.

erenoglu, 2023-03-29:
Thanks. It seems I have the X35A but not the X50H  :(  Do you mind taking a higher resolution picture or more complete view of the board? That X50H port you have seem to have 8 connectors.

I can try to find alternative port on the board which does same function. Which protocol is  your ESP32 talking to the board? I mean, for example to set temperature setpoint, change fan speed etc need to give some serial command right?

revk, 2023-03-29:
Very hard to get picture as wood beam right by it. The X50 and S21 protocols are different but my module does both and works out which is being used.

erenoglu, 2023-03-29:
OK, it seems I dont have X50 port :( I have the attached. X30A seems related port, which also has the P1P2 ports.

[image]

revk, 2023-03-29:
This needs to be Tx and Rx (i.e. single wire each way serial data), plus GND and power.

urbozz, 2023-03-30:
Some work others has done on the P1P2 protocol (very common on Ducted systems or VRV systems)

https://github.com/Arnold-n/P1P2Serial
https://github.com/budulinek/Daikin-P1P2---UDP-Gateway

urbozz, 2023-03-30:
Also some useful info on the X50 harness https://github.com/unixko/DaikinCon

revk, 2023-04-14:
We handle X50.

closed 2023-04-14 by revk