revk/xESP32-Faikout #31: Is an external temperature source supported?

matt-nz, 2023-04-28:
Firstly thanks for the great project, since I couldn't order the PCB not being in UK I built a very simple version using just an ESP32 and 32pin dev board which handles the 12v power conversion. I learnt that ESP (and experienced in previous projects) can handle 5v on input pins so didn't even use a level shifter (Just removed the use of invert on TX). I then used dupont wires to connect to the S21 port ran those out of the metal shielded areas and taped internally on the side. All working great so far and discovered in HA.

Anyway my question was I was looking at the docs and code and I see there is support for an external remote [Environmental monitor]. Can I just a sensor for my home assistant and get HA to send a temp update when it changes? I tried using MQTT on topic `command/heatpump/control` with a payload of 
`{
  "env":"22"
}`
but didn't seem to work as expected, instead it looked like auto control was activated and a scale of builtin / external being applied. 

I just want to change the temperature source to be external rather than the heatpumps inbuilt is that possible with the current build?

revk, 2023-04-28:
Basically it handles some specific MQTT messages from the other sensor. I may even have documented them now.

But also can now handle a BlueCoinT BLE temperature sensor directly, which seems to work very well.

revk, 2023-05-28:
The external temp logic only works by the Faikin controlling the settings hight/low, it is not possible to tell the Aircon the external temperature itself sadly (that would be nice), so sending external control temperature tells it to take control, setting heat/cool/fan accordingly. This means you need to send target too as the Faikin can no longer rely on the target from the Aircon itself as that would need to be set high/low to control it.

An example of the external control is:

`command/GuestAC/control {"env":22.00,"target":[16.000,32.000]}`

closed 2023-05-28 by revk