revk/xESP32-Faikout #32: Can you add the Humidity Value from the AC?

Amtho, 2023-05-05:
Is it possible to get the humidity value over s21 and at it to mqtt?

revk, 2023-05-05:
If you set the `dump` setting, and the `morepoll` setting true you will see all the S21 traffic we know about.

If you can work out which is humidity, I am happy to add it!

Amtho, 2023-05-05:
I have looked into this also everything what I can found so far regarding s21 protocol nowhere is the humidity mentioned. Beside that my emura is showing this when I do a get_sensor_info call: ret=OK,htemp=26.0,hhum=45,otemp=17.0,err=0,cmpfreq=0
so there must be something like a humidity sensor... Will look further into this.

revk, 2023-05-28:
I'll close for now, if anyone finds the S21 protocol for this, I will be happy to add it.

closed 2023-05-28 by revk

revk, 2023-05-30:
The latest code has a much improved S21 debug. Change the setting `debug` to 1, e.g. `setting/GuestAC/debug 1` and it will log to MQTT for the S21 responses, e.g. 

```
info/GuestAC/s21 {"G1":"13@A","G2":"=;\u0000?","G3":"0??\u0000","G4":"0??0","G5":"000?","G6":"0008","G7":"00B0","G8":"0200","G9":"???0","GA":"4970","GB":"0331","GC":"57D0","GG":"0300","GK":"q{51","GM":"53A3","GN":"0000","GP":"5A00","GQ":"1D00","GS":"0000","GT":"5000","SH":"532+","SI":"500+","Sa":"551+","SL":"060","SN":"100+","SX":"091+","SD":"000"}
```

I am not sure I have humidity, but if you can see your logs you may find one of the fields that is. Note that it is likely to be backwards, e.g. if your humidity was 25, then it is likely to be `52` or `052` or even `0520`, or some such. If you see one that is humidity, let me know and I'll add to the code.