revk/xESP32-Faikout #34: Feature Ventialtiemode silence

kroon040, 2023-05-27:
Hi,

I have beside de ventialation modus auto,1,2,3,4,5 also de mode silence. Is this also to implement?

revk, 2023-05-28:
"Night" is "silence" I believe.

But "Night" does not show as status if you change from remote, as the S21 protocol seems to be unable to report Night mode.

[image]

kroon040, 2023-05-28:
Hi,

The option is not selectable in the homeassistant integration. Only Auto and 1,2,3,4,5 are selectable.

revk, 2023-05-28:
I will look in to that.

revk, 2023-05-28:
I have issued new code, please test as I done have HA set up here.

kroon040, 2023-05-28:
Here my findings:

- Setting first the airco on cooling in the webinterface and when running clicking on (night). Nothing happens on the airco, and refreshing the webinterface the fan status is back on auto.

- Did thet same action with mqtt in homeassistant, Set it on cooling, and than set it on silence. Clicking that the result is it jumps to  fan only, and not cooling.

I have another airco where I have on the S21 interface andwifi interface BRP069B42 with an EKRS21 cable. There it works as expected, aircostay on cooling and fan goes in silence mode. 

I hope later to have another esp32 from amazon and putting this on also on the EKRS21 cable.

revk, 2023-05-29:
Sounds like we need to sort some debug. The debugging in quite verbose, on MQTT. I might see if any way to make it neater.

revk, 2023-05-29:
If you are able to get a serial dump of what works on the BRP069B42, that would help.

Also, I'll try and update the debugging anyway in the code to make it easier to use.

But changing to fan only is odd as that is a separate setting. So I am not sure if this is a control issue or an HA issue, or what.

revk, 2023-05-29:
Do a `setting` of `debug` to `1` and monitor MQTT for debug logs.

E.g. 

```
info/GuestAC/s21 {"G1":"03@","G5":"000","G6":"000","G7":"00B","SH":"532","SI":"032","Sa":"041","G2":"=;\u0000","G3":"000","G4":"0??","G8":"020","G9":"???","GB":"033","GG":"000","GK":"q{5","GM":"33A","GN":"000","GP":"5A0","GQ":"1D0","GS":"000","GT":"500","SN":"500","SX":"091"}
```

revk, 2023-05-30:
I have made more improvements to the HA config in latest release as well.

kroon040, 2023-05-30:
Hi,

I will upgrade later this day. And check it. I have on my infrared remote also the option to set it in quiet mode , so I can than debug it with your software? 

I don;t now it I can set wifi module from daikin on S21 together with the esp32 parallel, I think that won't work. I can maybe check if the new wifi module that is also available on my daikin (cloud only) if there is a quiet mode. I have to enable this one again en config this.

revk, 2023-05-30:
I doubt they can be connected in parallel, sorry. You may be able to use an ESP32 to tap the serial line and dump traffic, which may be useful.

The quiet mode is a pain - yes, I too can set this on my IR remote. But it appears in the S21 messages exactly the same as "auto" and I see nothing to indicate it is in "quiet" mode. I think we manage to set quite mode though.

If you get a serial dump of traffic either way I'm happy to look at it. If you find anything in debug to help, I can add to the code.

kroon040, 2023-05-30:
I have it over the quiet mode. I use this a lot when it's hot and my airco (external unit) is on the flat roof of my bedroom..

revk, 2023-05-30:
OK, I do not know if that can be set via S21 or reported via S21. If the official WiFI allows that setting then getting a serial dump of what it sends is what I would need.

revk, 2023-05-30:
Yes, "B" seems to be the fan rate for night mode and we set that.

But the underlying S21 message reports back as "A" (auto).

kroon040, 2023-05-30:
I notice , that when cooling is active , and than change the fan speed for the cooling in HA, it changes to fan instead of keeping cooling with another fan speed/ In the web interface it works as expected.

revk, 2023-05-30:
Ok hanging mode to fan does not make sense. That we need to work out.

revk, 2023-05-30:
Oh, I have a thought why that may be. Will check in morning.

revk, 2023-05-30:
OK, yes, command "fan" sets mode to "fan"

Command "fan" and number sets fan speed, but because it is command "fan" also sets mode to "fan". It should not set mode to "fan" unless no value.

Fixed, try latest code.

commit a72a747b (2023-05-28): Fix #34
closed 2023-05-30 by kroon040