revk/xESP32-Faikout #36: Disable Led blinking when connected to wifi

kroon040, 2023-05-30:
Hi,

Is it possible when the device is setup and has connection to the home wifi that the led blinking stops, preferred off. Now I see the led blinking even when it's builtin.

 It should be nice to have an option in the webinferface to choose what the do with the led.

patrickli, 2023-05-30:
I noticed this too. The LED is very bright even it is installed under the aircon shell. I ended up hiding it inside the PCB box which is black.

A related question, @revk can you explain what does the LED color/blink mean? I know steady green is normal operations but I've seen it blink blue. An explanation maybe in README would be helpful. Thanks.

revk, 2023-05-31:
Putting in the manual is a good idea. I have done that.

It is possible to disable totally with a setting, e.g. `setting/GuestAC {"blink":[0,0,0]}`

Having a mode for LED only when "something wrong" is an idea. I could do that if people want.

patrickli, 2023-05-31:
@revk Thanks for putting things in the manual but what does green mean? mine blinks green when the aircon is off.

revk, 2023-05-31:
Err, it should not do green! It might mean the LED does not have a "red", i.e. faulty board.

patrickli, 2023-05-31:
hm... doesn't make sense right? It should do yellow when off and green is not part of yellow. Let me test it tomorrow with a few different operation modes.

revk, 2023-05-31:
Err green is very much part of yellow! Yellow is green+red. So a faulty red would make the attempt at yellow show green.

patrickli, 2023-05-31:
Yeah you are right. I'm being silly. Anyway I will do some test to prove it.

revk, 2023-05-31:
Let me know if you want a replacement.

patrickli, 2023-05-31:
Tested a bunch of modes and the LED doesn't blink anything that needs red. So you might be right. As for replacement etc, is there anyway we can discuss privately? I don't want to hijack OP's ticket.

JeanOlivier, 2023-05-31:
> Putting in the manual is a good idea. I have done that.
> 
> It is possible to disable totally with a setting, e.g. `setting/GuestAC {"blink":[0,0,0]}`
> 
> Having a mode for LED only when "something wrong" is an idea. I could do that if people want.

That'd be nice. Better than always off IMO.

revk, 2023-06-01:
Simplest is probably a DM on https://toot.me.uk/@revk

revk, 2023-06-01:
OK a mode to have LED off for "normal working"... I'll look in to it.

kroon040, 2023-06-01:
Agree that should be a nice option

revk, 2023-06-01:
I think I'll expand the LED colours to better show state anyway - but a dark mode makes a lot of sense.

closed 2023-06-01 by revk

revk, 2023-06-01:
I have updated the LEDs, added a `dark` settings, and also added a test mode - connecting tx and rx detects loopback and causes red/green/blue LED cycle, so as to allow simple testing of tx, rx, and all LEDs to avoid any issues with faults in future.

revk, 2023-06-02:
You don't need to be on the same server to send me a message. I don't see a message from you yet. @revk@toot.me.Uk

commit 0443737a (2023-06-01): Fix #36