revk/xESP32-Faikout #37: Missing GPIO pins in documentation

kroon040, 2023-05-30:
Hi,

It took me some time, but after a while I found howto flash the premade esp32 that I bought on amazon. On the print it's not readable so it should be nice to find it in the documentation. I couldn't find it in the documentation

revk, 2023-05-30:
The documentation explains you can assign any pins you like in the settings

kroon040, 2023-05-30:
for flashing you need the tx , rx , 3.3v , gpio0 and gnd. The pins on the board I bought are available , but couldn't read which one is

revk, 2023-05-30:
[image]

kroon040, 2023-05-30:
Ok, but 1 and 3 are the tx and rx.