revk/xESP32-Faikout #38: IPv6 Support with SLAAC

kroon040, 2023-06-01:
Hi,

Is it possible to add ipv6 support with SLAAC. I know it's already in the ESP-IDF frameworks. It should be nice to have. Also connecting then with ipv6 to the mqtt broker.

revk, 2023-06-02:
I thought that was already turned on. I'll check.

revk, 2023-06-02:
Apologies for that, for some reason it was not set on the config for the code issue build. I have enabled and issued new code.

closed 2023-06-02 by revk

kroon040, 2023-06-02:
HI,

I see get get a Link local ipv6. I must check if I get an IPv6 global with SLAAC. 

2 questions:

-Ccan I put in the mqtt broker also my ipv6 global address and it should work? It should ben nice to have in te interface a status that mqtt is connected in  the settings.
- Is it possible to have the netwerk info also in the settings, so I see the ipv4 adress, dns and ipv6  link-local and ipv6 global.

revk, 2023-06-02:
Not a bad idea, I'll look in to that.

kroon040, 2023-06-02:
I check and filled in the mqtt broker ipv6, and that worked. And by that I see also de IPV6 Global address (mqtt broker is in another subnet). In the logging of the broker the ipv6 global address appeared

revk, 2023-06-02:
Showing DNS is proving more complex, but showing IPv4/IPv6 is simple, so doing that in settings page.

kroon040, 2023-06-02:
ok , thats enough. And than state of mqtt also in this page (connected/disconnected)

revk, 2023-06-02:
Sorry, yes, it shows if MQTT is connected.

kroon040, 2023-06-02:
Yes I see it works. Maybe in MQTT connected more loig to place it between MQTT Pass and set button, so you see in that section when connected

kroon040, 2023-06-03:
I got today my second esp32 faikin, and when configuring I noticed whatever I fill in in the MQTT settings , it always say connected, while mqtt wasn't connected.

revk, 2023-06-03:
Interesting - might be something of mistake then.

revk, 2023-06-04:
OK it was a simple simplistic, saying connected even if the client was trying to connect. I have updated and made a tick next to the MQTT host field now.

kroon040, 2023-06-04:
Tested on smartphone and laptop, I only get strange characters. Also in incognito browser

[image]

revk, 2023-06-04:
Odd, something is not realising the page is utf-8. I'll see if I can work out how you do that. Works on Safari, but maybe it assumes UTF-8.

revk, 2023-06-04:
Yes, utf-8 was missing on the settings page. I'm surprised it is not default to be honest. Try now.

kroon040, 2023-06-04:
Now it works as expected