revk/xESP32-Faikout #40: Power limiting

damige, 2023-06-05:
The onecta app has the ability to limit the power of the system from 100% down to 40% 
This feature comes in handy as lowering the peak power performance helps my unit to have longer runs.

I could not find a reference that this project includes this feature. Did is miss it? And if not could it be added?

revk, 2023-06-05:
If you do a setting of debug, e.g. MQTT `setting/GuestAC/debug 1` you will get logs on MQTT showing the response to the S21 messages.

If you do that, and then enable the feature, see if any of the responses change.

That will give some clues so we could perhaps add this.

Set debug back to `0` afterwards.

The issue is there is no proper documentation of the S21 protocol so it is reverse engineered.

damige, 2023-06-05:
Thanks for the quick response. This setting is only available through the app, and not through the remote. Does the app still work with this project?

revk, 2023-06-05:
Depends how the app works - might not be able to connect at the same time as the Faikin which may make it difficult. If you can, then at least we may see the "status" side of it and perhaps can guess the command side.

damige, 2023-06-05:
Alright, i will try to get 3 of these devices to the Netherlands for this experimentation.

revk, 2023-06-11:
More stock hopefully end of the week, reopen this issue if you find any more.

closed 2023-06-11 by revk

apos-t, 2023-07-29:
Hi,
I'm also interested in power limiting and made the following observation on my FTXM20R2V1B (FTXM-R) Prefera unit.

It looks to me, that the Powerlimit is controlled by the field Econo there is for example the value 0.52 in the Econo field when I activate the Econo mode. In this case I assume it means 52% powerlimit. I also saw already values like 0.32 or 0.40 in this field.

MQTT Logs attached:

{"protocol":"S21¬Tx","ts":"2023-07-29T10:05:00Z","online":true,"home":21.50,"heat":false,"fanrpm":[340,349,350],"comp":140,"outside":[22.00,22.17,22.50],"liquid":10.00,"power":true,"mode":"D","temp":22.00,"fan":"A","swingh":false,"swingv":false,"econo":0.53,"powerful":false}

in the next log it switches to true:

{"protocol":"S21¬Tx","ts":"2023-07-29T10:06:00Z","online":true,"home":21.50,"heat":false,"fanrpm":[340,349,350],"comp":140,"outside":[22.50,22.68,23.00],"liquid":10.00,"power":true,"mode":"D","temp":22.00,"fan":"A","swingh":false,"swingv":false,"econo":true,"powerful":false}

revk, 2023-07-29:
The reporting for boolean is indeed the proportion of the time (usually last whole minute) that was `true`. So sounds like `econo` is indeed being changed. But the S21 will be just on/off.

Were you changing `econo` during the minute?

apos-t, 2023-07-29:
I've changed it in the faikin gui. Ok then it has nothing to do with the power limit, I will try to debug the change of the power limit. Any idea which data I should monitor?

revk, 2023-07-29:
The S21 code is all reverse engineered. You can enabled `debug` more which reports what we get from a load of queries, and see if they change when you change settings on a remote control.

revk, 2023-07-29:
Nothing obvious in there is there... It could be possible that it is a code we do not know/try. If ever either of us find the code, it would be easy to add. But for now, sorry.

apos-t, 2023-07-29:
ok thank you, would be great as it helps to deal with the on / off intervals when heating with low demand and a multi split.

apos-t, 2023-07-29:
My UseCase looks like this:

I have a multi split with 4 indoor units and 1 outdoor unit. I try to stretch the on / off intervals of the outdoor unit with the following logic.

1. demand control - always start with the lowest max. demand 40% and wait if it's enough to get the target temp. If not increase the max. demand by 10%. When the target temp is reached -  set demand to the lowest value and s.o

2. If the lowest demand is still too much and the outdoor unit starts with the on/off intervals (monitored by a power meter for the outdoor unit) I set the target temp. of all indoor units to a low temp. (17 degrees) for 30 minutes. The outside unit stays off for this period. After the 30 min I set the indoor units back to 21 degrees and s.o.

Maybe there is a better way for my UseCase

revk, 2023-07-29:
Sounds complicated, sorry. I don't really have a multiple indoor unit management solution. I do have 4 indoor units on a single outside unit, but don't have any issues with demand, it is a nice large outside unit.

But you can use MQTT with the Faikin to control the target temperatures and mode of all of the indoor units using some code/scripts you write yourself I expect.

apos-t, 2023-07-29:
I'm using home assistant for the multiple indoor management, just missing the power limit control feature in faikin. But I can use the wifi Controller and Daikin cloud to control the power limit (In my post I meant power control not demand, sorry) in the moment. But my goal is to  completely shutdown the wifi Controller and the Daikin cloud...