revk/xESP32-Faikout #41: A few issues/questions after a few days use

patrickli, 2023-06-05:
* I can't seem to control the AC through HA/MQTT. The local web control works perfectly. Any changes gets updated in Home Assistant immediately but I can't use the climate entity/services in HA to control the AC. I'm using HA 2023.5.4.
* The lower limit of the temperature slider seems to be 18°C. I usually set mine to 17 in winter so that's a number I can't set with the web control. The remote can set to far lower than that.
* The AC remote have timed on/off function. When the AC is off and a on timer is set, Faikin reports the AC as powered on. But it is actually not true. Is there a way to identify this status and show power as off? Not sure if you can get the timer information but at least don't how it as being on while it is not.

Thanks for your hard work.

patrickli, 2023-06-05:
Did some debugging for the first issue. When setting things from HA, it is publishing to the wrong topic:

[image]

Instead of `command//temp` it should be `command/UNIQUE_ID/temp`. So in your `homeassistant/climate/UNIQUE_ID/config` payload you should set `~` to `command/UNIQUE_ID`.

revk, 2023-06-06:
In the web config when setting up, did you give it a name? That is what would appear where Unique_ID would be.

Also make sure you have upgraded the software (in the settings link under web controls).

patrickli, 2023-06-06:
No I didn't. I thought if I don't then it would use the default? I'll try put the default name in.

patrickli, 2023-06-06:
Alright that worked. Wasn't expecting to have to set it explicitly.

revk, 2023-06-06:
Indeed, it should use the default, so I'll look in to that anyway.

revk, 2023-06-06:
I have fixed default handling - it was default in a lot of places, but not all.

patrickli, 2023-06-06:
Thanks for fixing this. What about the other 2 questions I mentioned?

revk, 2023-06-06:
Ah, OK...

revk, 2023-06-06:
I don't know where it is getting the 18C, I'll have to look in to that. The timer one is weird. I have an AC with a timer on the remote, so I should be able to investigate that.

Leave with me.

revk, 2023-06-06:
More investigation, looks like S21 has 18C as the base. Is this an S21 cable. If so I'll need some debug for when you have it set to 17 so I can see what it does.

revk, 2023-06-06:
I mean it is a S21 port and protocol, which I expect. If you are able to turn on debug, and set 17 using the remote, and post the debug, that would help. It may be that I need a new setting for this. How low can your a/c go?

And if doing that, debug when the timer is set would also help.

Debug enabled by setting `debug`, e.g. MQTT `setting/GuestAC/debug 1`. Set back to `0` when you have finished. The debug comes out on the MQTT, and shows all the S21 responses.

patrickli, 2023-06-06:
Which topic should I listen to for the debug messages?

revk, 2023-06-06:
I have issued some code with configurable min and max, defaulting to 16C min.

Let me know if that works.

revk, 2023-06-06:
OK, good, the change I have made for temperature should work then. You can do `setting/Name/tmin 14` to set the min allowed to 14C. Impressive it goes that low.

The traces show G1 first character 0 meaning not "on". Could be the lack of name being set that caused the issue in the first place.

G3 is the only clue that there is a timer set. I would be interested to see that with a few more times, including changes to end time, to see if that is indeed timer settings.

patrickli, 2023-06-06:
I'll do it tomorrow as the only Faikin I have is in the kids bedroom and they are in bed.

patrickli, 2023-06-07:
Some more testing:

18C, start at 6:00
```
b'{"G1":"04@A","G2":"5:\\u0000\x80","G3":"1\x8d\xfe0","G4":"0\\u0000\x80\\u0000","G5":"000\x80","G8":"0\\u0000\\u0000\\u0000","SH":"091+","SI":"502+","Sa":"002+","SL":"000","SN":"711+","SX":"502+","SD":"290"}'
```

18C, start 6:00, stop 23:30
```
b'{"G1":"04@A","G2":"5:\\u0000\x80","G3":"3\x8df0","G4":"0\\u0000\x80\\u0000","G5":"000\x80","G8":"0\\u0000\\u0000\\u0000","SH":"091+","SI":"502+","Sa":"591+","SL":"000","SN":"711+","SX":"502+","SD":"290"}'
```

18C, stop 23:30
```
b'{"G1":"04@A","G2":"5:\\u0000\x80","G3":"2\xfef0","G4":"0\\u0000\x80\\u0000","G5":"000\x80","G8":"0\\u0000\\u0000\\u0000","SH":"091+","SI":"502+","Sa":"591+","SL":"000","SN":"711+","SX":"502+","SD":"290"}'
```

18C, start 7:00
```
b'{"G1":"04@A","G2":"5:\\u0000\x80","G3":"1\x92\xfe0","G4":"0\\u0000\x80\\u0000","G5":"000\x80","G8":"0\\u0000\\u0000\\u0000","SH":"091+","SI":"502+","Sa":"591+","SL":"000","SN":"711+","SX":"502+","SD":"890"}'
```

18C, stop 23:00
```
b'{"G1":"04@A","G2":"5:\\u0000\x80","G3":"2\xfeb0","G4":"0\\u0000\x80\\u0000","G5":"000\x80","G8":"0\\u0000\\u0000\\u0000","SH":"091+","SI":"012+","Sa":"591+","SL":"000","SN":"711+","SX":"502+","SD":"790"}'
```

revk, 2023-06-07:
Yep, looks like G3 has a pattern that matches - interesting.

I wonder how the aircon knows the time?

I'll write that up anyway.

It may be that I change the start/stop logic I have now to "telling the aircon" the start stop, so it matches the remote.

Thanks for doing those test.

revk, 2023-06-07:
Indeed, but I was pondering more on if there is a message to set it on S21 :-)

closed 2023-06-11 by revk

pfak, 2023-07-27:
On my FTXS09LVJU, if I set the setpoint to 16C it will bump it up to 18C after a few seconds.

revk, 2023-07-27:
There is a min temp setting in the code which can be changed, but also the aircon may have a min. Mine does the same, the aircon itself won't set below a certain temp.

pfak, 2023-07-27:
```{"protocol":"X50¬Tx","timeout":true}``` when set below 18C.

revk, 2023-07-27:
Sounds like it does not like the setting? It does go to lower temps does it? Ie via remote control?

pfak, 2023-07-27:
Sadly, no. I purposely bought the Faikin adapter to try to get it below 18C. I need to force cooling on it because it reads my temperatures wildly inaccurate. Huge bummer!

Still neat project :)

revk, 2023-07-27:
Ah, sorry.

pfak, 2023-07-27:
Blame Daikin!