revk/xESP32-Faikout #44: Changing temperature when on doesn't trigger MQTT status update

kroon040, 2023-06-10:
When I change the temperature in home assistant or in the web interface. I see that the state/airco/status that setting the temp, I don;'t get a new update. Is this possible to add>

revk, 2023-06-10:
Hmm, it probably should.

It does report via a home assistant specific message (prefixed with MAC), and a regular summary message starting `Faikin/...` but the `state/...` is not for a temperature change.

IIRC the idea was that you see live changes in temperature readings all the time and reporting every 0.1C change is OTT. There is a setting, `live` which can be enabled to override that.

It may make sense to force such an update on any control being changes, so I'll look at that too.

kroon040, 2023-06-10:
I checked almost every value, and that reports status back to the mqtt topic, only the temp not

revk, 2023-06-10:
OK new code issued, see if that does what you need.

But if using for live temperature, then sending `setting/airco/live 1` will get you a lot more `state/...` messages.

closed 2023-06-10 by revk

kroon040, 2023-06-10:
Great, that solves it. Thnx