revk/xESP32-Faikout #46: How does it work with recent units that have wifi built-in?

ThiefMaster, 2023-06-11:
Recent interior units (I think 2022+) have wifi included by default, you no longer need a separate wifi module.

Does anyone know if they simply included the wifi module but still have the connectors necessary for this module, or whether soldering (and thus probably loss of warranty) would be necessary?

joshbenner, 2023-06-12:
I think you'll have to be more specific -- I just bought brand new units from Daikin that did not have any included Wifi but do have S21 port. What models are you hoping to integrate with?

revk, 2023-06-12:
The real question is whether they have S21 port, and if that works in parallel with their built in wifi module. (I have no idea)

closed 2023-06-13 by revk

ThiefMaster, 2023-06-13:
I asked in a forum (sorry, only in German) and some guy with a Daikin model that has built-in wifi confirmed that there's still an S21 port: https://www.kaelte-treffpunkt.de/viewtopic.php?f=1&t=11702

According to his post he actually ordered a bunch of Faikin controllers and will also report whether they'll work fine in parallel w/ the original wifi module once he connected them.

njh, 2023-07-11:
I have a FTXM-R (Perfera) that I bought in 2022.

I have it connected up, so I can currently use both the the official Wifi Module and the Faikin Module at the same time. The built-in module also has a 5-pin header but it is much smaller and may have a completely different pinout / protocol.

[image]

But I will probably disconnect the built-in Wifi module, once I have the Faikin working the way I want it to.

revk, 2023-07-12:
I'd love to know if the 5 pin connectors is the same as S21. Can you check voltages on it?