revk/xESP32-Faikout #51: Compatibility list

Sonic-Amiga, 2023-06-19:
This information is missing from docs, would be useful to add. What AC models is this tested with? What models use S21 and what models use "new" ?

revk, 2023-06-19:
That's what the wiki is for, which lists a number of models, and is linked from the readme. So what am I missing?

revk, 2023-06-19:
[image]

Sonic-Amiga, 2023-06-19:
Probably i am missing the link. Perhaps because i am basing on an older version. Thank you for the response anyways.

closed 2023-06-19 by Sonic-Amiga

MassiPi, 2023-07-10:
i don't know if you want to add this, since i'm not using your hardware or your software, but Daikin FTXSxxG are working fine on S21 port, 3.3V compatible and first pin not connected.
Do not let set powerful or eco modes (Even if supported with remote controller), but power, mode, fan swing are working and temp inside, outside, coil and compressor idle state are reported
they also answer to a long list of commands but with no sense :)

revk, 2023-07-11:
You should find the list on the Wiki is editable if you want to add with notes. And thank you.