revk/xESP32-Faikout #52: New compatible model: ADEA71A2VEB

jamofer, 2023-06-19:
**Sky Air ADEA71A2VEB**
* Official WiFi module: BRP069C81.
* Ports: X35A (power supply) and X50A (communication).

I've already used it for a week replacing my old BRP069C81 that was failing a lot with Onecta APP.  Faikin Works perfectly!

Here it is my home-assistant screenshot:

[image]

If it is okay, we can update the compatible list Wiki adding this new model.

revk, 2023-06-19:
It should be directly editable now? Do try.

jamofer, 2023-06-19:
Done.

Yes it is, but I don't know if I had to discuss it here first before adding it. Thanks!

revk, 2023-06-19:
Fair enough. As long as people are not being dicks I'm happy for people to update the wiki as needed. So far so good.

thanhdatneu, 2023-08-20:
Hello, can u show me how to connect with port X35 and x50A ? Thanks

revk, 2023-08-20:
My air-con came with a lead - not sure if anyone else has pictures.

thanhdatneu, 2023-08-20:
[image]

[image]

This is my x35A and x50A and i don't know how to connect. Tks

thanhdatneu, 2023-08-20:
[image]

[image]

This is my x35A and x50A and dont know how to connect, Pls check, Many thanks

revk, 2023-08-20:
The X35A looks like power, the yellow and blue wires here.
The X50A is data, the orange and red wires. The brown is not used.

[image]
[image]
[image]

thanhdatneu, 2023-08-20:
Oh, your port is X50H, It is different from my port, my unit is x50A 😅

revk, 2023-08-20:
Where do you get X50H from, it's X50A

[image]

thanhdatneu, 2023-08-21:
Oh sorry, the image is not clear so i thinh it was "H". I will check my port. Thanks

DatutaX, 2023-09-09:
Hey, i connect module to my X50A, but it does not work
Wifi ok but system is offline
On X50A the orange pin is 26 or 27 on your module ?

revk, 2023-09-09:
I'm just about to get on a plane, but see the pictures in this thread.

DatutaX, 2023-09-12:
@jamofer hello, can u show me your X50A port, thanks

revk, 2023-09-12:
Literally look at the posts in this discussion, you see the X50 and wiring to it.