revk/xESP32-Faikout #54: FTXJ20A2V1BW advice

RJ, 2023-06-22:
Hi, just received the chip from amazon.co.uk. awaiting delivery of an S21 cable.

I have a fairly new FTXJ20A2V1BW (Daikin Emura 3) wall mounted.

This model unfortunately came with the cloud-only built-in wifi module BRP069C4x, unlike my older Emura models.

Per the circuit board diagram and label on the pcb itself, i can see a currently empty S21 socket:

[image]

So the plan is to power off the unit, plug into the empty S21 port, and power back on then do setup over wifi.. No need to try and unplug the existing wifi module right, they can coexist? (in a different socket anyway)

Thanks

revk, 2023-06-22:
No idea, let us know.

Is the existing module the same plug as the Faikin? If so you may be able to just swap it.

RJ, 2023-06-22:
nope the existing wifi is a different plug with more pins. I will see if they are happy to coexist.

was curious if home assistant state would be out of sync if I mostly used HA but sometimes turned off with the remote.

will report back once my s21 lead arrives 👍

revk, 2023-06-22:
Experience with using a remote is that it all stays in step

Sonic-Amiga, 2023-06-23:
@RJ Perhaps they can coexist. Some small time ago, in order to get familiar with S21, i tried googling, and found some tech educational videos (in russian). They explain some AC of the expensive line (probably yours), and say that "it has built-in wifi support but that's not fully functional; so we've designed a cloud adapter for you". They also explain that "not fully functional" means you don't have a fancy smartphone app, which you can just install and go anywhere. I. e. it's not cloud-enabled. I have a suggestion that they have support for their original protocol built-in. You may try probing for that:
1. Power on the AC.
2. Set up built-in module, connect to your wifi. I expect the user manual to explain how.
3. Find out its IP address.
4. Go to http://ip.of.the.ac/aircon/get_control_info

If everything is OK, you'll see a plaintext response, looking like this one:

ret=OK,pow=1,mode=3,adv=,stemp=23.0,shum=0,dt1=25.0,dt2=M,dt3=23.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=AUTO,dh2=50,dh3=0,dh4=0,dh5=0,dh7=AUTO,dhh=50,b_mode=3,b_stemp=23.0,b_shum=0,alert=255,f_rate=A,f_dir=0,b_f_rate=B,b_f_dir=0,dfr1=5,dfr2=5,dfr3=B,dfr4=5,dfr5=5,dfr6=5,dfr7=5,dfrh=5,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0

If so, then your AC supports their old good web protocol out of the box; toss your cloud module out of the window 😄

Sonic-Amiga, 2023-06-23:
Or, don't actually toss it, but if you're techie enough, open it and look what's inside. Changes are it's something widespread, like ESP family. In this case you can turn it into a useful thing of any kind; it cost you money after all.

Just FYI: i did the same with my cloud-only liberated, denazified and import-substituted module that came with my new Sensira. I mean not tossing out of the window; but porting Faikin to it: https://github.com/Sonic-Amiga/ESP8266-Faikin

Sonic-Amiga, 2023-06-23:
Doh, really sorry, i am so impatient having read things a bit backwards. You have a cloud-only built-in; and you've ordered Faikin.

BTW, it's not in the release yet; but i've fixed and completed implementation of the protocol, compatible with older Daikin online controllers (i have one on my first AC). So if HA supports Daikin out of the box, it will work with the upcoming Faikin, It already runs great in my apartment with OpenHAB. Has some bugs, i am working on them, but generally very nice.

Sonic-Amiga, 2023-06-23:
And i also confirm that the AC always reports correct state back, even if disturbed by remote. This is the advantage of using wifi module over el cheapo "control-it-all" IR transmitters.

njh, 2023-06-23:
I have just managed to get Faikin [working with my FTXM-R](https://github.com/revk/ESP32-Faikin/discussions/49). It also has a built-in Cloud Wifi controller connected to a different port (and empty S21 port).

With Faikin installed, I am able to use both the official ONECTA app and the Faikin at the same time.
Although the ONECTA app is a bit slow to notice changes.

Once I am happy with my Faikin setup, I will probably disconnect the built-in Wifi controller, as one less thing to cause a security incident on my home network!

revk, 2023-06-24:
Do add working models to the wiki please. Well done.

closed 2023-06-24 by revk

RJ, 2023-07-04:
bought an EKRS21 cable, and had to connect socket to socket but all working nicely now. wiki updated.

Sonic-Amiga, 2023-07-04:
BTW, maybe add cable pinout infos to the wiki?

migueldc73, 2024-03-22:
I just bought a Faikin module, but the JST-EH connector as indicated in the main thread does not fit 

> The S21 plug needed is [JST, EH Female Connector Housing, 2.5mm Pitch, 5 Way, 1 Row](https://uk.rs-online.com/web/p/wire-housings-plugs/3116237), which needs pins and crimps as well.

The connector on my Emura 3 board is clearly smaller with probably 1.5 or 2mm pitch, still 5 way though.

Does anyone have any idea which connector I need to order ?

Delivery times for the EKRS21 cable are 4-5 weeks. Can't wait that long since the new HA cloud integration fails after 12 hours.

migueldc73, 2024-03-26:
on my Emura 3, there was no voltage on the Pin 4 (supposedly 12-15V), so used the 5V from Pin 1 to feed into Pin 4 of the Faikin and it all works.
Bit scary though when I didn't find voltage on pin4, thought I blew up my Emura board...

yuvalfar, 2024-07-17:
Hey @RJ 
I was just planning to test the voltage on pin 4 and eventually connect my faikin, but I’m stuck at the beginning of removing the small screw covers (little plastics behind the flap). Have you got any tips? The suggestion on the manual „ruler wrapped in cloth“ doesn’t help me much as even a thin one gets no grip. Thanks!

RJ, 2024-07-17:
@yuvalfar do you mean the secret little plastic covers that hide the screws so you can get the case off? the one under the front vents? if so I think I just popped them off with a small screwdriver. they just snapped back on afterwards iirc. not home ATM so can't check

yuvalfar, 2024-07-17:
[image]

@RJ exactly. they seem quite stubborn on my unit. did you put the screwdriver below them and pushed them up and out? thanks

RJ, 2024-07-17:
> [image]
> 
> @RJ exactly. they seem quite stubborn on my unit. did you put the screwdriver below them and pushed them up and out? thanks

hmm, can't remember tbh. don't recall it being difficult to remove once I finally identified them. maybe I used needle nosed pliers to grab them and pull, maybe by hand, maybe levered with a screwdriver 🤷‍♂️ sorry I can't be more help

RJ, 2024-07-17:
they do lift up and out

yuvalfar, 2024-07-17:
Tried all three without success. not sure if upwards is the direction as otherwise pulling horizontally out which would be my instinct doesn’t work. Too little material to grip with the pliers. maybe a combination of these. I’ll wait here for information as there is no particular rush and I’d rather avoid potential damage. If anyone reads and has tips please drop them here :)

yuvalfar, 2024-07-17:
Here is an illustration from the manual of the old emura. but I’m not sure it’s the same as even a very thin piece (used the cutout excess of the daikin board) doesn’t fit). the new manual has no illustration only mentioning the ruler covered in cloth

[image]

yuvalfar, 2024-07-17:
Update for posterity: just had great success with the Lego prying tool pushing these covers up basically

Take-Off90, 2024-08-18:
> on my Emura 3, there was no voltage on the Pin 4 (supposedly 12-15V), so used the 5V from Pin 1 to feed into Pin 4 of the Faikin and it all works. Bit scary though when I didn't find voltage on pin4, thought I blew up my Emura board...

Same issue here.
Did you bridge PIN1 and PIN4 or do you connect PIN1 to PIN4?

revk, 2024-08-18:
Pin 1 on the a/c is usually 5V. If it can supply enough current then the Faikin can be powered by it - connected to pin 4 on the Faikin side.

migueldc73, 2024-08-18:
> Same issue here. Did you bridge PIN1 and PIN4 or do you connect PIN1 to PIN4?

Bridging could be potentially dangerous, as Daikin support even claimed that I blew up the board (for not having 12V on Pin 4).
Now that I see others have the same issue, I guess Daikin doesn't have a clue that there is something different on this board and just blame anything on the users. 

So, to answer your question, I just took the 5V from Pin on the A/C side and connected to Pin 4 Faikin side.

Take-Off90, 2024-08-18:
> > >  
> So, to answer your question, I just took the 5V from Pin on the A/C side and connected to Pin 4 Faikin side.

So let me get this right. You have not connected PIN1 on the Faikin?

revk, 2024-08-18:
Indeed, pin 1 on Faikin is only for tx pull up, which the Daikin does not need.

dahadd, 2024-12-05:
> on my Emura 3, there was no voltage on the Pin 4 (supposedly 12-15V), so used the 5V from Pin 1 to feed into Pin 4 of the Faikin and it all works. Bit scary though when I didn't find voltage on pin4, thought I blew up my Emura board...

@migueldc73 

Is the 5V from pin 1 sufficient to supply the Faikin with voltage? Can you please send me or explain a connection diagram. thank you

revk, 2024-12-05:
Cannot say, usually the 5V supply is not enough, and usually you need the 5V, but some devices have enough on 5V

dahadd, 2024-12-05:
I installed S21 Pin1 on Fainkin Pin4 and it works.
Pin4 has no Power