revk/xESP32-Faikout #57: "liquid" temperature sensor

Sonic-Amiga, 2023-06-23:
In your code one of sensor variables is called "liquid" Why so and what does it mean ? I see in S21 protocol it's populated by 'SI' response; and in my opinion 'I' should stand for 'inlet', which you also have.

I have a guess that on some ACs there's an extra temperature sensor in the remote control, so that it feels more of the same temperature as a human would feel in the room. A traditional sensor in the indoor unit itself becomes "inlet".

My original BRP controller which i have also reports "hhum=-", which most likely stands for "home humidity". There's also a corresponding control on get_control_info, which reports "shum=0". I guess that's "set humidity". Looks like support for built-in humidifier in some top-grade models. Does this have to do with "liquid" ?

njh, 2023-06-23:
I assume the "liquid" temperature is the temperature of the refrigerant liquid in the coil.

revk, 2023-06-24:
I think it is.

revk, 2023-06-24:
It I recall some models report “liquid” and “gas” temps. These are refrigerant. Inlet is the inlet air as I understand it.

closed 2023-06-24 by revk

revk, 2023-06-24:
[image]

Sonic-Amiga, 2023-06-24:
Thank you for explaining, IMHO worth mentioning  in docs somewhere (or wiki ?)