revk/xESP32-Faikout #61: Documentation?

calebemelo, 2023-06-27:
I'm new here, sorry but I couldn't find the documentation on how to link the components.

revk, 2023-06-27:
How do you mean?

It says clone recursive and make. That's it. If you have EDP IDF set up.

What problem are you having?

calebemelo, 2023-06-27:
I saw that some users built their own modules through your project, I didn't find how to build the module, where to connect tx, rx, count controller.. sorry I'm newbie.

revk, 2023-06-27:
I assumed you meant software linking, sorry.

The S21 has a tx and rx pin, you need a module that can tolerate 5V. You can connect to any GPIO pins on your module and configure them to be the pins to use for Tx and Rx.

calebemelo, 2023-06-27:
Thank you very much, I will try to find someone with knowledge of electronics to do it for me.

revk, 2023-06-28:
The GitHub has `PCB` (which contains various versions) with `production` directory that includes the JLCPCB specific `gerber`, `bom`, and `position` files needed. The latest PCB `Faikin3` needs to be ordered with *Designs:2*.

`Faikin3` is being updated over the next few days, and will then replace the other `PCB` designs once prototypes from JLCPCB are confirmed, next week.

So may be worth waiting a few days.

closed 2023-06-28 by revk

njh, 2023-06-28:
What is the best way of people identifying if they a Faikin 1 / Faikin 2 / Faikin 3 board?

I fairly sure I have a Faikin 2, which has a date of 2023-06-01 on the PCB.

revk, 2023-06-29:
They all work the same.

2 and 3 are just trying out some new parts. I don't think any have been sold yet.