revk/xESP32-Faikout #62: FTXP25M5V1B9 no connection

Di-rk, 2023-06-29:
Hi, I just bought some of your faikin boards, they look very promising. Unfortunately I can't get any connection when connected to the S21 port on my FTXP25M5V1B9.  I don't have any other wifi modules to test on these airco units. 

[image]

I measured the following voltages on the S21 port (measured to pin 5 which I assume is GND):
Pin 1: 5V
Pin 2: 5V
Pin 3: 5V
Pin 4: 14V
Pin 5: 0V

I connected the faikin board dupont cables. The faikin works and connects to my WiFi network without problems, so I assume my wiring is correct, however I can not control the AC with it. It also says on the web interface that the system is off line.

[image]

Do you have any ideas on how to get it working or how I can debug this?

Thanks

revk, 2023-06-29:
If you connect the pins on the Faikin module, pins 3 and 4 together, and power it on, it should blink red, green, blue (if latest firmware loaded), that shows it has detected a loop back and confirms the Faikin board is working correctly, which would be a good start.

revk, 2023-06-29:
Looks like it is not seeing any responses at all. Short of an oscilloscope on Tx/Rx, or trying one of the official wifi modules to confirm the port works, I am not sure what else to suggest. It would be a shame if we found an aircon that did not work with it.

Di-rk, 2023-06-29:
I can connect it to one of my other ACs (same model, but one is more powerful) and see if that works, to rule out if the port on this one is broken. I can test with a scope later this week probably. Otherwise I'll have to buy the official Wifi modules.

revk, 2023-06-29:
Well let me know. I think the S21 port is very standard in Daikin, but I have not found a specification to follow sadly, but a lot of models do work. So I hope we can work it out.

njh, 2023-06-29:
@revk do you know which direction are the pins are officially numbered from?
(Is GND Pin 1 or Pin 5?)

revk, 2023-06-29:
The triangle is pin 1, which is the one that is not used on my board (NC), and is normally 5V.
Pin 5 is GND.

revk, 2023-06-29:
Someone has confirmed FTXF35D5V1B worked, if that helps. This is very strange.

Di-rk, 2023-06-30:
I think I found it. I copied some of your code to send S21 packets with the F1 command from my computer using an FTDI adapter and the AC responds to that correctly. It does not respond to the Faikin sending the same command. On the scope I verified that they sent the exact same data.

Here is a screenshot using the FTDI adapter.
[image]

Then I noticed that my FTDI adapter was running on 5V data and the Faikin on 3.3V. I then switched my FTDI adapter to 3.3V.

[image]

And I got no response.

It seems that my specific Daikin model is incompatible with 3.3V signals.

revk, 2023-06-30:
That is a bugger, but does explain it. And well done diagnosing that.

It is possible to make a design that uses pin 1 (5V) to drive the output. But usually 3.3V is enough to work a logic level like that.

What is the exact model, I'll make a note on the compatibility list, and see if I can work out any way of tackling this. One issue is some leads only have 4 connections not 5, i.e. don't have the 5V connection, so they would not work with a new design.

One possible think to try might be a resistor from pin 1 to the Tx pin to pull it up a bit. It may not need much. Have you got a picture of your Faikin module? Trying to work out if the one you has includes a 100 ohm on the Tx, which would make that reasonably safe to do. Maybe 1k resistor?

Di-rk, 2023-06-30:
Models that I have are FTXP25M5V1B9 and FTXP35M5V1B9, I'm guessing all FTXP-M models (Daikin Comfora) are affected. 

[image]

I don't think my board has the 100 ohm resistors.

revk, 2023-06-30:
The fact you measured 5V on that pin means there may be a pull up. It may be an open collector output would do the trick. If I also did a resistor to pin 1 for when pin 1 is in use, that may work in either case. I'll have to try that out here. It may make for a design that will work to 5V properly.

revk, 2023-06-30:
OK, I have checked mine, and it has a pull up internally to 5V, so an open collector output should work. I'll order some samples, it will take a week. Where in the world are you? I could send you a sample of the new design...

Di-rk, 2023-06-30:
I tried the resistor but that did not help unfortunately. I'm currently trying to set the TX pin to open drain output mode to see if that helps. (Can I push my own binary via OTA?)
If that doesn't work, I'll gladly test some samples for you. I am in The Netherlands.

revk, 2023-06-30:
The pin itself will have a parasitic diode to 3.3V I bet, but worth a try.

If you have a p-channel FET you may be able to fit in in place of 100 hon resistor and cap, and set port to `-26` instead of `26` to invert in code.

You can, via MQTT, tell it the cost to get binary.

But I can make the change here and let you update if you want to wait a moment.

revk, 2023-06-30:
Ok I tried setting OUTPUT_OD and then re-setting the iomux - but the documentation for that is crap and it did not work.

If you find a way, let me know.

In meant time, I'll make new boards - but that is at least a week.

revk, 2023-06-30:
I have ordered new PCBs using open drain.

revk, 2023-07-01:
In the mean time I have prototyped it here with a simple N-channel FET to make an open drain.

It works on my unit, which is encouraging, as I don't have the 5V on pin 1 so this is internal pull up.

[image]

revk, 2023-07-02:
DM me on https://toot.me.uk/@RevK and I can post that one to you anyway. New boards will be at least a week.

Di-rk, 2023-07-02:
DM sent, glad to hear it already works on your unit!

haissio, 2023-07-03:
> Someone has confirmed FTXF35D5V1B worked, if that helps. This is very strange.

Yup! That was me.
Just to add some hopefully useful information:

My modules are self-made, I'm not using the Faikin boards, and I tried to wire directly the ESP32 module to the S21 port of the A/C unit.
I have several units to test (FTXF35D5V1B/FTXF42D5V1B) and I observed the 5v pin was able to power the ESP32 on some units but not on others, which seems to indicate that the logic board of the A/C units is not very stable in terms of voltage.
I was having the same 'unit is offline' problem with all of the tests, until I added a DC-DC regulator to get the 5v from S21 pin 4 instead of pin 1, and despite that it did not work until I also added a 3.3v-5v voltage shifter for the TxD/RxD signals.

With that in place it works like a charm, so maybe some recent units are more voltage picky than older ones? The A/C units were all installed this quarter.

Thanks for such a great project!

revk, 2023-07-03:
OK that does sort of confirm these units want 5V signalling, so my change should fix that. I'll post out a board for you.

danijelt, 2023-07-04:
From what I gathered, some units need 5V _**and**_ their TX is not inverted? I have the same problem with FTXC-D unit which officially uses BRP069B45 module.

Granted, I use [esphome-daikin-s21](https://github.com/joshbenner/esphome-daikin-s21) rewrite of this code for ESPHome on my DIY build, but the symptoms are the same. In serial log it says that the ESP didn't receive ACK even with level shifter. And if I don't connect UART at all, it logs timeouts, not ACK errors, so I guess that would indicate correct wiring but that it received unexpected data - possibly inverted?

I'll try it and report back. If it works on my unit too, that would indicate that all units that need *045 wifi module will need this modification. [See this brochure, page 122](https://www.daikin.co.uk/content/dam/internet-denv/catalogues_brochures/commercial/100%20-%20Sky%20Air%20product%20catalogue%20for%20professionals.pdf). My FTXC-D and @haissio's FTXF-D are listed under BRP069B045.

revk, 2023-07-04:
It is easy to make the code try tx inverted and not. That would also solve the problem of my newer modules having an inverting output.

mjbnz, 2023-07-05:
I have an FTXV71UVMA, and after installing it today, I'm wondering if I'm hit with this too.

Can you suggest any quick method for verifying?

Of course, I'd love a replacement board - but I'm in NZ, so might be a bit annoying for shipping!

revk, 2023-07-05:
Tx invert is easy to try via a setting, but I expect to have code later today that tries inverting Tx, and that may be easier.

To try 5V you probably need to budge a FET on to the board.

revk, 2023-07-05:
OK latest code will try S21/X50 and S21/X50 with inverted Tx, until it finds a valid reply.

It stores this so next restart will try the right one first.

mjbnz, 2023-07-05:
> OK latest code will try S21/X50 and S21/X50 with inverted Tx, until it finds a valid reply.
> 
> It stores this so next restart will try the right one first.

no joy on my end, I'm happy to give a bodge a go, can you recommend a part number?

revk, 2023-07-05:
Ok the image above shows the bodge, any small n-channel FET should do. 

I used https://uk.farnell.com/nexperia/pmf250xnex/mosfet-n-ch-30v-0-9a-sc-70/dp/3440055?st=pmf250xnex

danijelt, 2023-07-05:
That's it, adding level shifter and disabling inverted RX fixed the communication on my FTXC-D.

However, the unit sends NAK for RH, RI, Ra and RL queries. Do you have any idea why? Here's the log of sending the turn off command, which works, but I'm getting NAK for these 4 commands all the time.
```
[17:02:26][D][daikin_s21:415]: Sending basic climate CMD (D1): 00\x11A
[17:02:26][D][uart_debug:114]: >>> 06:02:44:31:30:30:11:41:27:03
[17:02:26][D][uart_debug:114]: <<< 06
[17:02:26][D][uart_debug:114]: >>> 02:46:31:77:03
[17:02:26][D][uart_debug:114]: <<< 06:02:47:31:30:33:5C:41:78:03
[17:02:26][D][uart_debug:114]: >>> 06:02:46:35:7B:03
[17:02:26][D][uart_debug:114]: <<< 06:02:47:35:30:30:30:30:3C:03
[17:02:26][D][uart_debug:114]: >>> 06:02:52:48:9A:03
[17:02:26][D][daikin_s21:261]: NAK from S21 for RH query
[17:02:27][D][uart_debug:114]: <<< 15
[17:02:27][D][uart_debug:114]: >>> 02:52:49:9B:03
[17:02:27][D][daikin_s21:261]: NAK from S21 for RI query
[17:02:27][D][uart_debug:114]: <<< 15
[17:02:27][D][uart_debug:114]: >>> 02:52:61:B3:03
[17:02:27][D][daikin_s21:261]: NAK from S21 for Ra query
[17:02:27][D][uart_debug:114]: <<< 15
[17:02:27][D][uart_debug:114]: >>> 02:52:4C:9E:03
[17:02:27][D][daikin_s21:261]: NAK from S21 for RL query
[17:02:27][D][uart_debug:114]: <<< 15
[17:02:27][D][uart_debug:114]: >>> 02:52:64:B6:03
[17:02:27][D][uart_debug:114]: <<< 06:02:53:64:39:39:39:62:03
```

revk, 2023-07-05:
No idea, I assume some queries only work on some units. I made my code give up after a number of NAKs for any specific type of query.

danijelt, 2023-07-05:
Thanks. The issues that I'm seeing now due to NAKs are specific to ESPHome port, so I'll move there.

If you plan to document this, you can note that FTXC-D is also affected by this problem, and my guess is that any unit that uses *B045 adapter is, too - since all units reported here are listed under BRP069B045 adapter in the catalogue that I linked above.

imdos, 2023-07-06:
Hi @Di-rk I live in the Netherlands and couldn't find your contact information. I have a FTXF35C; Could we arrange something to get me a board. If you have a spare one? I don't know how to contact you directly. If you would like to contact me drop me a line  imdos @ hotmail.com

revk, 2023-07-06:
I have a prototype new board coming soon (early next week) so could post one.

danijelt, 2023-07-06:
I bought the BRP069B45. It works fine with FTXC-D. I sniffed the serial bus and here's what I observed:
- Rd response is 999 (39:39:39) when the unit is off, not 000 as noted in your code
- inside and outside temperatures, humidity and error codes are retrieved with single F9 query. Example output:
  - `02:47:39:B2:B4:FF:30:17:03`
    - B2 - inside temperature (25 °C)
    - B4 - outside temperature (26 °C)
    - FF - probably humidity - BRP returns "hhum=-" on `get_sensor_info` endpoint
    - 30 - probably some error code (don't know if AC or adapter) - BRP returns `err=0` on `get_sensor_info` endpoint
  - To get the temperature in °C from the bytes: `bytes / 2 - 64`. Example: B2 = 178 / 2 = 89 - 64 = 25. My unit doesn't send 0.5 °C steps, although the format makes it possible in theory.
  - Since this is also present on models that do respond to RH and Ra queries (seen in the spreadsheet found [here](https://community.openenergymonitor.org/t/hack-my-heat-pump-and-publish-data-onto-emoncms/2551/128)), maybe RH and Ra queries could be replaced with F9 to make it universal? Though, I see in your comments that you received FF in place of the outside temperature, so maybe some additional handling would be required if that wasn't just a temporary bug?

revk, 2023-07-07:
That is excellent.

revk, 2023-07-07:
I have added some F9 handling now, not issued this for automated OTA yet, need to do a bit more testing here. I get FF for outside, yes.

revk, 2023-07-07:
Can you clarify the inverted signal(s)?

Is it just from Faikin to Daikin that needs inverting, and driving for 5V?

Or is signal from Daikin to Faikin also an issue?

danijelt, 2023-07-07:
Speaking for my unit (FTXC-D):
- TX line from the AC unit to the ESP's RX is not inverted. In case of your PCB, it would mean bypassing the FET on the ESP's RX line or inverting it in the software.
- TX line from the ESP to the AC unit's RX is not inverted (as-is on current PCB) but it should be 5V, not 3.3V, or the unit will never respond to any queries.

Now, this doesn't rule out other possible combinations which we maybe just haven't encountered yet, such as inverted Daikin's RX or even 3.3V signals. So perhaps the best solution would be to cover them in advance (with jumpers maybe?), default to 3.3V and instruct users to experiment if the communication doesn't work?

revk, 2023-07-07:
Ok the 5V/3.3V I think I have covered. Ideally need someone to try a sample when they arrive.

I can make s/w handle all invert cases, just in case.

revk, 2023-07-07:
New version issued, including Tx/Rx invert checks, and F9 handling (if no RH/Ra).

Let me know if that works OK for you?

danijelt, 2023-07-07:
It should work with my DIY board (just an ESP32 devkitc board with level shifter wired to GPIO), right? I can test it later today.

revk, 2023-07-07:
It tries inverted and non inverted, so yea.

danijelt, 2023-07-07:
Are your bin files not complete images (flashable to 0x0 with esptool.py)? When I flash with `esptool.py --chip esp32 -p /dev/ttyUSB1 write_flash 0x0 ./release/Faikin-S1.bin` I get a bootloop:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xad14defb
invalid header: 0xad14defb
invalid header: 0xad14defb
invalid header: 0xad14defb
invalid header: 0xad14defb
invalid header: 0xad14defb
invalid header: 0xad14defb
invalid header: 0xad14defb
ets Jul 29 2019 12:21:46
```

revk, 2023-07-07:
They are what esp idf makes which is, I think, separate to boot loaded.

But if have updated the ones are ota.RevK.Uk now so you can do an OTA upgrade.

danijelt, 2023-07-07:
I setup the esp-idf environment and got it working, but unfortunately I can't test it right now. Without getting in the details, I'm deadlocked with the wiring right now and my ESP must be powered from the USB, which results in the AC not communicating over S21 because it expects some power draw from VCC line to enable the bus.

Perhaps the better test would be if someone else in this thread can try it on an actual Faikin board (@Di-rk @mjbnz). I can test it next week when I get all the connectors and wires and finish my HW.

In the meantime, I sniffed the bus again when powered on. RH and RI still don't respond, and Rd always responds with 999.

revk, 2023-07-07:
"power draw from VCC line to enable the bus" worries me as my module does not do that. Is that a thing?

I expect to have new Faikin with open drain on Tx to allow for 5V, and pull up to VCC (will that be "power draw"?).

I only have a few coming, but can post out. I'll let you know when they arrive and we can sort addresses.

danijelt, 2023-07-07:
Sorry, I meant 12V line, not 5V. When anything (official adapter, Faikin, etc.) starts drawing power, the AC beeps once. It's independent of whether the RX and TX lines are connected. In my case, there's currently (temporarily) no power draw from 12V, the unit doesn't beep and there's nothing on the data bus.

revk, 2023-07-07:
That is weird, IMHO, but good, we do that, phew. You had me worried for a moment :-)

Amtho, 2023-07-11:
Im powering my own board with 12V from the s21 and then use an ADuM1201 for TX/RX shifting, this one is powered on one site from the 5V from the S21 on the other Site from 12V(s21) --> to 3V for the ESP32. This works fine without an inverted signal on the RX or TX. Also you should not power your ESP32 from the s21 5V line I think possible supply current on this one is maybe not high enough for the ESP32 during max consumption like transmitting or so.

I cannot confirm that you need to pull substancial power over the 5V line to activate the interface. (its only powering one site of the Adum which only need maybe 1-2ma from the 5V.

revk, 2023-07-11:
Indeed, I am using the nominal 12V for power, and the 5V as a reference if present.

danijelt, 2023-07-11:
> I cannot confirm that you need to pull substancial power over the 5V line to activate the interface. (its only powering one site of the Adum which only need maybe 1-2ma from the 5V.

As I already clarified, power draw on 12V line is needed to activate the S21 bus, not 5V. I use 5V only to power the level shifter, too.

closed 2023-07-11 by revk

Di-rk, 2023-07-14:
I just tested the board with the mosfet on it and it works perfectly! Thank you!

danijelt, 2023-07-21:
@revk I know this is closed, but I'd just let you know that, for some reason, Faikin reports "system if offline" and there seems to be no connection, both on OTA firmware and latest grabbed from the repository. When I switch to ESPHome communication works.

As I previously stated, this is on a custom PCB with ESP32 devkitc, UART wired to pins 26 and 27 and a level shifter powered from 5V on S21 connector and 3.3V on ESP32 board (the board doesn't invert any signals). So you don't have to waste time on my custom contraption, but I'm curious because it's obviously not some wiring issue and Faikin uses same pins, 26 and 27.

revk, 2023-07-22:
- [ ] Does setting debug show anything received at all?

metarutaiga, 2023-07-25:
Alira X
06.02.47.39.C0.FF.76.30.E5.03

revk, 2023-07-25:
A tad more would help, does that say tx or rx, etc. if rx then that is a big clue.

danijelt, 2023-07-26:
I found some time to setup everything (I have successfully avoided MQTT until now), here's some info:
- shorting pins either directly on the ESP, or through the level shifter, correctly detects loopback
- log with debug and dump set to true - it just loops trying all protocols and RX/TX combinations but it never receives anything from the AC:
```
Message 437 received on error/C049EFE4DEC8/comms at 22:47:
{"protocol":"X50","timeout":true}

Message 436 received on info/C049EFE4DEC8/tx at 22:47:
{"protocol":"X50","dump":"06AA0701000146"}

Message 435 received on info/C049EFE4DEC8/tx at 22:47:
{"protocol":"X50","cmd":"AA","payload":"01"}

Message 434 received on info/C049EFE4DEC8/s21 at 22:47:
{}

Message 433 received on error/C049EFE4DEC8/comms at 22:47:
{"protocol":"S21¬Rx¬Tx","cmd":"F1","noack":true}

Message 432 received on info/C049EFE4DEC8/tx at 22:47:
{"protocol":"S21¬Rx¬Tx","dump":"0246317703"}

Message 431 received on info/C049EFE4DEC8/cannot-tx at 22:47:
{"protocol":"X50¬Rx¬Tx","cmd":"CB","payload":"0000"}

Message 430 received on info/C049EFE4DEC8/cannot-tx at 22:47:
{"protocol":"X50¬Rx¬Tx","cmd":"CA","payload":"0000000000000000000000000000000000"}

Message 429 received on error/C049EFE4DEC8/comms at 22:47:
{"protocol":"X50¬Rx¬Tx","timeout":true}

Message 428 received on info/C049EFE4DEC8/tx at 22:47:
{"protocol":"X50¬Rx¬Tx","dump":"06AA0701000146"}

Message 427 received on info/C049EFE4DEC8/tx at 22:47:
{"protocol":"X50¬Rx¬Tx","cmd":"AA","payload":"01"}

Message 426 received on info/C049EFE4DEC8/s21 at 22:47:
{}

Message 425 received on error/C049EFE4DEC8/comms at 22:47:
{"protocol":"S21¬Rx","cmd":"F1","noack":true}

Message 424 received on info/C049EFE4DEC8/tx at 22:47:
{"protocol":"S21¬Rx","dump":"0246317703"}

Message 423 received on info/C049EFE4DEC8/cannot-tx at 22:47:
{"protocol":"X50¬Rx","cmd":"CB","payload":"0000"}

Message 422 received on info/C049EFE4DEC8/cannot-tx at 22:47:
{"protocol":"X50¬Rx","cmd":"CA","payload":"0000000000000000000000000000000000"}

Message 421 received on error/C049EFE4DEC8/comms at 22:47:
{"protocol":"X50¬Rx","timeout":true}

Message 420 received on info/C049EFE4DEC8/tx at 22:47:
{"protocol":"X50¬Rx","dump":"06AA0701000146"}

Message 419 received on info/C049EFE4DEC8/tx at 22:47:
{"protocol":"X50¬Rx","cmd":"AA","payload":"01"}

Message 418 received on info/C049EFE4DEC8/s21 at 22:47:
{}

Message 417 received on error/C049EFE4DEC8/comms at 22:47:
{"protocol":"S21¬Tx","cmd":"F1","noack":true}

Message 416 received on info/C049EFE4DEC8/tx at 22:47:
{"protocol":"S21¬Tx","dump":"0246317703"}

Message 415 received on info/C049EFE4DEC8/cannot-tx at 22:47:
{"protocol":"X50¬Tx","cmd":"CB","payload":"0000"}

Message 414 received on info/C049EFE4DEC8/cannot-tx at 22:47:
{"protocol":"X50¬Tx","cmd":"CA","payload":"0000000000000000000000000000000000"}

Message 413 received on error/C049EFE4DEC8/comms at 22:47:
{"protocol":"X50¬Tx","timeout":true}

Message 412 received on info/C049EFE4DEC8/tx at 22:46:
{"protocol":"X50¬Tx","dump":"06AA0701000146"}

Message 411 received on info/C049EFE4DEC8/tx at 22:46:
{"protocol":"X50¬Tx","cmd":"AA","payload":"01"}

Message 410 received on info/C049EFE4DEC8/s21 at 22:46:
{}

Message 409 received on error/C049EFE4DEC8/comms at 22:46:
{"protocol":"S21","cmd":"F1","noack":true}

Message 408 received on info/C049EFE4DEC8/tx at 22:46:
{"protocol":"S21","dump":"0246317703"}
```

Messages 408 and 409 at the bottom are what my unit should respond to, but it doesn't appear to. This same board works when I flash ESPHome with Daikin S21 component.

revk, 2023-07-26:
The next step would be oscilloscope and comparing between the working and non working s/w. I am somewhat at a loss on this one.

ChielBoots, 2024-04-11:
> I just tested the board with the mosfet on it and it works perfectly! Thank you!

Would you mind sharing the wiring scheme? I have a FTXP35M5V1B like yourself, and a Faikin underway. Not too many people with this combination AFAIK.

ChielBoots, 2024-04-17:
> I just tested the board with the mosfet on it and it works perfectly! Thank you!

@Di-rk : I have the identical split unit (FTXP35) and just got my Faikin PCB in. Would you mind sharing the wiring scheme for this unit to the Faikin board? It's a bit confusing reading through all the threads. Also, can you confirm the connector on the FTXP35 S21 port? I know the Faikin uses the JST EH 2,54 pitch but not sure about connector on the AC board?

[7 more replies]