revk/xESP32-Faikout #64: Help With Flashing

iancoop, 2023-06-30:
Hello, Very excited to find this project, something I've wanted for a while, having used an IR Controller for some time https://github.com/oznu/homebridge-daikin-esp8266 but wanting hardwire connection, so thanks very much for your efforts.

I would love to purchase the prebuilt boards but it's been a challenge to get Amazon to ship to Australia, and I have some ESP-32 Wroom boards and the ability to make a similar (but crude in comparison) interface, so I've been trying to flash my modules with your pre built binaries. I have also tried to build the code and flash, but no joy.  I've tried  

iancooper@m101b ESP32-Faikin % esptool.py --port /dev/cu.usbserial-0001 write_flash --flash_mode dio --flash_size 4MB 0x0 Faikin-S1.bin

esptool.py v4.6.2
Serial port /dev/cu.usbserial-0001
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:c9:a3:c6:6d:ec
Uploading stub...
Running stub...
Stub running...
Configuring flash size...

A fatal error occurred: Faikin-S1.bin requires chip revision in range [v3.0 - v3.99] (this chip is revision v1.0). Use --force to flash anyway.

Can you tell me if this ESP-32 should be suitable, or give me some pointers to the way forward ? Any ideas would be greatly appreciated, thanks.

revk, 2023-06-30:
Amazon are being weird, I agree. They have shipped to NZ for example!

I have not seen that error before. I suspect you need to do some `make menuconfig` to try different ESP32 targets, and `make flash` to find a combination that works.

iancoop, 2023-06-30:
Ah yes.. here is where I am out of my depth. I have followed the Standard Toolchain Setup https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html and can compile and flash the hello_world project, but when I try to make yours I hit this error :

iancooper@m101b ESP32-Daikin % make flash     
Makefile:27: mariadb/mysql not installed, needed for tools
idf.py flash

as the first error. 

I've installed mariadb with `brew install mariadb`, but beyond this I don't know the way forward in order to get to the `make menuconfig` which is why I was trying to flash your pre-built binary. I have not been able to find any info relating to getting mariadb/mysql installed for use with the esp-idf.  Maybe the path of least resistance would be to get the same module as yours, I think this would work ? ESP32-PICO-KIT https://au.mouser.com/ProductDetail/Espressif-Systems/ESP32-PICO-KIT?qs=MLItCLRbWsyoLrlknFRqcQ%3D%3D

Thanks for your help,

revk, 2023-06-30:
`Makefile:27: mariadb/mysql not installed, needed for tools` is a warning, ignore it - you are not making tools

iancoop, 2023-06-30:
OK, getting a little closer perhaps. Changed the `make menuconfig` to not have signed code to get past a .pem error, changed the > Component config > Hardware Settings > Chip revision to Rev 1.0, and Make wroom.  I am now getting these errors : 

/Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c: In function 'ip_event_handler':
/Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c:1258:13: error: implicit declaration of function 'sntp_stop'; did you mean 'ap_stop'? [-Werror=implicit-function-declaration]
 1258 |             sntp_stop ();
      |             ^~~~~~~~~
      |             ap_stop
/Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c:1259:13: warning: 'sntp_init' is deprecated: use esp_sntp_init() instead [-Wdeprecated-declarations]
 1259 |             sntp_init ();
      |             ^~~~~~~~~
In file included from /Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c:26:
/Users/iancooper/esp/esp-idf/components/lwip/include/apps/esp_sntp.h:255:6: note: declared here
  255 | void sntp_init(void)
      |      ^~~~~~~~~
/Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c: In function 'task':
/Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c:1562:69: error: the comparison will always evaluate as 'true' for the address of 'addr' will never be NULL [-Werror=address]
 1562 |                      if (!esp_netif_get_ip6_global (sta_netif, &ip) && ip.addr)
      |                                                                     ^~
In file included from /Users/iancooper/esp/esp-idf/components/esp_netif/include/esp_netif.h:12,
                 from /Users/iancooper/esp/esp-idf/components/esp_wifi/include/esp_wifi_default.h:10,
                 from /Users/iancooper/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:58,
                 from /Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/include/revk.h:12,
                 from /Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c:12:
/Users/iancooper/esp/esp-idf/components/esp_netif/include/esp_netif_ip_addr.h:103:14: note: 'addr' declared here
  103 |     uint32_t addr[4]; /*!< IPv6 address */
      |              ^~~~
/Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c: In function 'revk_boot':
/Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c:1840:4: warning: 'sntp_setoperatingmode' is deprecated: use esp_sntp_setoperatingmode() instead [-Wdeprecated-declarations]
 1840 |    sntp_setoperatingmode (SNTP_OPMODE_POLL);
      |    ^~~~~~~~~~~~~~~~~~~~~
/Users/iancooper/esp/esp-idf/components/lwip/include/apps/esp_sntp.h:235:6: note: declared here
  235 | void sntp_setoperatingmode(u8_t operating_mode)
      |      ^~~~~~~~~~~~~~~~~~~~~
/Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c:1843:4: warning: 'sntp_setservername' is deprecated: use esp_sntp_setservername() instead [-Wdeprecated-declarations]
 1843 |    sntp_setservername (0, ntphost);
      |    ^~~~~~~~~~~~~~~~~~
/Users/iancooper/esp/esp-idf/components/lwip/include/apps/esp_sntp.h:249:6: note: declared here
  249 | void sntp_setservername(u8_t idx, const char *server)
      |      ^~~~~~~~~~~~~~~~~~
/Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c: In function 'revk_web_config':
/Users/iancooper/esp/ESP32-Faikin/components/ESP32-RevK/revk.c:2371:57: error: the comparison will always evaluate as 'true' for the address of 'addr' will never be NULL [-Werror=address]
 2371 |          if (!esp_netif_get_ip6_global (sta_netif, &ip) && ip.addr)
      |                                                         ^~
/Users/iancooper/esp/esp-idf/components/esp_netif/include/esp_netif_ip_addr.h:103:14: note: 'addr' declared here
  103 |     uint32_t addr[4]; /*!< IPv6 address */
      |              ^~~~
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.

Any thoughts ? Sorry, esp-idf newbie, obviously :-)

Thanks,

revk, 2023-06-30:
Which version of ESP IDF?
I'm building on 5.0.1

iancoop, 2023-06-30:
Yes 5.1 I think, I pulled the latest a few days ago. I'll give it a go with 5.0.1

Thanks,

revk, 2023-07-01:
I'll assume that works for now - I have to sort ESP IDF 5.1 soon anyway.

closed 2023-07-01 by revk

iancoop, 2023-07-02:
Yes, I have just managed to make it work with v5.0.1, and changed the minimum hardware to V1.0 to suit my ESP-32 vroom module. 

Thanks for your help! 

One thing I did notice is that once it came up, the set button was missing in the captive portal window when accessed from my MBP M1 running OS 13.4.1. It was fine from my phone.

revk, 2023-07-02:
Any way to see any error logs. It uses web sockets which usually work.

It's an area I may be able to improve anyway. I'm planning to tidy that up. I have a non web socket build. I may be able to do that and just used the web sockets for the WiFi list.

revk, 2023-07-02:
I may be able to add a V1 to my standard builds for you so it can OTA update for my changes. Can you confirm exactly what you changed?

iancoop, 2023-07-02:
> Can you confirm exactly what you changed?

I think so, it would in the file sdkconfig ? Should I zip and post here ? OTA update would be great, thanks.

> Any way to see any error logs.

From running `idf.py -p <PORT> monitor` ? or is there another place it is logged ?

revk, 2023-07-02:
diff `sdkconfig` to `sdkconfig.defaults` should show.

Logs: I meant from browser - is it upset over web sockets maybe?

iancoop, 2023-07-02:
I'll have a hunt around for logs for the Captive portal. 

Not sure I've done this correctly : 
```

iancooper@m101b ESP32-Faikin % diff sdkconfig sdkconfig.defaults
8c8
< CONFIG_SOC_CAPS_ECO_VER_MAX=301
---
> CONFIG_SOC_CAPS_ECO_VER_MAX=3
239a240,242
> CONFIG_SECURE_SIGNED_ON_UPDATE=y
> CONFIG_SECURE_SIGNED_APPS=y
> CONFIG_SECURE_BOOT_V2_RSA_SUPPORTED=y
241c244,249
< # CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT is not set
---
> CONFIG_SECURE_BOOT_V2_PREFERRED=y
> CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=y
> CONFIG_SECURE_SIGNED_APPS_ECDSA_SCHEME=y
> # CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME is not set
> # CONFIG_SECURE_SIGNED_ON_BOOT_NO_SECURE_BOOT is not set
> CONFIG_SECURE_SIGNED_ON_UPDATE_NO_SECURE_BOOT=y
242a251,252
> CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
> CONFIG_SECURE_BOOT_SIGNING_KEY="secure_boot_signing_key.pem"
243a254,259
> 
> #
> # Potentially insecure options
> #
> # CONFIG_SECURE_BOOT_ALLOW_SHORT_APP_PARTITION is not set
> # end of Potentially insecure options
279,280c295,296
< CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
< # CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
---
> # CONFIG_ESPTOOLPY_FLASHSIZE_4MB is not set
> CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y
285c301
< CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
---
> CONFIG_ESPTOOLPY_FLASHSIZE="8MB"
303,304c319,320
< CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="components/ESP32-RevK/partitions_4m.csv"
< CONFIG_PARTITION_TABLE_FILENAME="components/ESP32-RevK/partitions_4m.csv"
---
> CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="components/ESP32-RevK/partitions_8m.csv"
> CONFIG_PARTITION_TABLE_FILENAME="components/ESP32-RevK/partitions_8m.csv"
352c368,477
< # CONFIG_BT_ENABLED is not set
---
> CONFIG_BT_ENABLED=y
> # CONFIG_BT_BLUEDROID_ENABLED is not set
> CONFIG_BT_NIMBLE_ENABLED=y
> # CONFIG_BT_CONTROLLER_ONLY is not set
> CONFIG_BT_CONTROLLER_ENABLED=y
> # CONFIG_BT_CONTROLLER_DISABLED is not set
> 
> #
> # NimBLE Options
> #
> CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_INTERNAL=y
> # CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL is not set
> # CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_DEFAULT is not set
> # CONFIG_BT_NIMBLE_LOG_LEVEL_NONE is not set
> # CONFIG_BT_NIMBLE_LOG_LEVEL_ERROR is not set
> CONFIG_BT_NIMBLE_LOG_LEVEL_WARNING=y
> # CONFIG_BT_NIMBLE_LOG_LEVEL_INFO is not set
> # CONFIG_BT_NIMBLE_LOG_LEVEL_DEBUG is not set
[... 272 lines cut ...]
> # CONFIG_BTDM_CONTROLLER_FULL_SCAN_SUPPORTED is not set
> CONFIG_BLE_ADV_REPORT_FLOW_CONTROL_SUPPORTED=y
> CONFIG_BLE_ADV_REPORT_FLOW_CONTROL_NUM=100
> CONFIG_BLE_ADV_REPORT_DISCARD_THRSHOLD=20
1629d1854
< # CONFIG_ESP_SYSTEM_PD_FLASH is not set
1650,1651c1875,1877
< # CONFIG_SPIRAM_SUPPORT is not set
< # CONFIG_ESP32_SPIRAM_SUPPORT is not set
---
> CONFIG_SPIRAM_SUPPORT=y
> CONFIG_ESP32_SPIRAM_SUPPORT=y
> # CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST is not set
1653,1655c1879,1881
< # CONFIG_ESP32_DEFAULT_CPU_FREQ_160 is not set
< CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
< CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240
---
> CONFIG_ESP32_DEFAULT_CPU_FREQ_160=y
> # CONFIG_ESP32_DEFAULT_CPU_FREQ_240 is not set
> CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=160
1703a1930
> CONFIG_SW_COEXIST_ENABLE=y
iancooper@m101b ESP32-Faikin % 
```

revk, 2023-07-02:
OK I think `CONFIG_ESP32_REV_MIN=1`

Does the binary it makes have `-V1` in it? That will be it.

iancoop, 2023-07-02:
Yes I am fairly sure the only thing I changed was > Component config > Hardware Settings > Chip revision to Rev 1.0.

Sorry, where would I see the `-V1` ? The binary is just called Faikin.bin

revk, 2023-07-02:
OK `make` usually creates is as `Faikin`[something]`.bin`, e.g. `Faikin-S1.bin`, etc.

You should have `Faikin-S1-V1.bin`

iancoop, 2023-07-02:
OK I am not sure how I managed.. I used just `make` and at the end I got

```
Successfully created esp32 image.
Generated /Users/iancooper/esp/ESP32-Faikin/build/Faikin.bin
[856/856] cd /Users/iancooper/esp/ESP3...oper/esp/ESP32-Faikin/build/Faikin.binFaikin.bin binary size 0x117f70 bytes. Smallest app partition is 0x1f0000 bytes. 0xd8090 bytes (44%) free.

Project build complete. To flash, run this command:
/Users/iancooper/.espressif/python_env/idf5.0_py3.11_env/bin/python ../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0xd000 build/ota_data_initial.bin 0x10000 build/Faikin.bin
or run 'idf.py -p (PORT) flash'
cp: illegal option -- -
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-aclpsvXx] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-aclpsvXx] source_file ... target_directory
make: *** [all] Error 64
```

Not sure about `make: *** [all] Error 64`

I used the long form to flash rather than `'idf.py -p (PORT) flash`

revk, 2023-07-02:
Are you running `make`?

Last line of `all:` target is

` @echo Done: $(PROJECT_NAME)$(SUFFIX).bin`

iancoop, 2023-07-02:
Yes I just used `Make`

revk, 2023-07-02:
Err `make` not `Make`... 

try `make all` just to be sure

iancoop, 2023-07-02:
ah yes, `make` 

So yes, you are correct, `make all` says 

`iancooper@m101b ESP32-Faikin % make all
Makefile:27: mariadb/mysql not installed, needed for tools
Make: Faikin-S1-V1.bin
/Users/iancooper/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
Running ninja in directory /Users/iancooper/esp/ESP32-Faikin/build
Executing "ninja all"...
[4/5] Generating binary image from built executableesptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /Users/iancooper/esp/ESP32-Faikin/build/bootloader/bootloader.bin
[5/5] cd /Users/iancooper/esp/ESP32-Fa...Faikin/build/bootloader/bootloader.binBootloader binary size 0x4650 bytes. 0x29b0 bytes (37%) free.
[6/7] Generating binary image from built executableesptool.py v4.5.1
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated /Users/iancooper/esp/ESP32-Faikin/build/Faikin.bin
[7/7] cd /Users/iancooper/esp/ESP32-Fa...oper/esp/ESP32-Faikin/build/Faikin.binFaikin.bin binary size 0x117f70 bytes. Smallest app partition is 0x1f0000 bytes. 0xd8090 bytes (44%) free.

Project build complete. To flash, run this command:
/Users/iancooper/.espressif/python_env/idf5.0_py3.11_env/bin/python ../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0xd000 build/ota_data_initial.bin 0x10000 build/Faikin.bin
or run 'idf.py -p (PORT) flash'
cp: illegal option -- -
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-aclpsvXx] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-aclpsvXx] source_file ... target_directory
make: *** [all] Error 64
iancooper@m101b ESP32-Faikin `

but the Error 64 at the end.

Apologies, hope this isn't wasting your time.

revk, 2023-07-02:
Ah the error at the end depends on your shell. I will fix that for you now.

I have issued `Faikin-S1-V1.bin` on `ota.revk.uk` so you can upgrade. Give that a try.

iancoop, 2023-07-02:
Amazing, seems to be working a treat. I didn't have the BLE and the automated local controls before, Thanks very much :-)

revk, 2023-07-02:
Well done.