revk/xESP32-Faikout #68: Add support for /aircon/get_model_info API endpoint

njh, 2023-07-07:
I have just been trying out [homebridge-daikin-local](https://github.com/cbrandlehner/homebridge-daikin-local) with Faikin but it fails to initialise with:

```
[Bedroom 2 Faikin] _doSendGetRequest: ERROR: API request to http://192.168.1.120/aircon/get_model_info returned error Error: Not Found
```

Searching for information about what that endpoint should return, finds this curl command/response:
```
curl --insecure -H "X-Daikin-uuid: 26d62729763b41189928d7dafc354fa8" "https://192.168.0.149/aircon/get_model_info"
ret=OK,model=1358,type=C,pv=2,cpv=2,cpv_minor=00,mid=NA,humd=0,s_humd=0,acled=0,land=0,elec=0,temp=1,temp_rng=0,m_dtct=1,ac_dst=--,disp_dry=0,dmnd=0,en_scdltmr=1,en_frate=1,en_fdir=1,s_fdir=3,en_rtemp_a=0,en_spmode=0,en_ipw_sep=0,en_mompow=0,en_patrol=0,en_fdir2=0,en_filter_sign=0%
```

I am not sure how essential this API call is - it appears to just be checking that remote device is up and logs the model. The [fields it is interested in](https://github.com/cbrandlehner/homebridge-daikin-local/blob/a261b4ab5244b5b6b077b5fd1d2d0f5fe6f34bb9/index.js#L1093) are:
* `ret`
* `model`

The other [endpoints it is trying to use](https://github.com/cbrandlehner/homebridge-daikin-local/blob/a261b4ab5244b5b6b077b5fd1d2d0f5fe6f34bb9/index.js#L200) are:
* `/aircon/get_sensor_info`
* `/aircon/get_control_info`
* `/aircon/get_model_info`
* `/aircon/set_control_info`
* `/common/basic_info`

revk, 2023-07-07:
So you have any idea where the model comes from.

If you have MQTT can you set debug. If I can find what S21 message has the model we can send that.

njh, 2023-07-08:
Yeah, I have no idea where the model number is supposed to come from. I have tried debugging the S21 protocol over MQTT, but I don't know enough about what each of the fields refer to to try and pick out something that might be a model number.

Given that it seems to be being used to check that there is a Daikin aircon system alive and well, how do you feel about returning this for now?
```
ret=OK,model=0000
```

And then it could be improved, if someone works out where the data could come from?

revk, 2023-07-08:
Could do. Bear in mind the S21 stuff is often backwards, so a model of `1358` may be `8531` in a message.

revk, 2023-07-08:
OK, I have added that. If we find the actual model somehow, that would be great, and happy to update.

closed 2023-07-08 by revk

njh, 2023-07-08:
Amazingly fast turn-around!
My local Faikin is now returning `ret=OK,model=0000` already 😀

njh, 2023-07-08:
And now confirmed that the [homebridge-daikin-local](https://github.com/cbrandlehner/homebridge-daikin-local) plugin loads and basic functions work 🎉 

[image]

Will do some more testing to see other functions work correctly.