revk/xESP32-Faikout #69: Switching to ESPHome firmware

XMaarten, 2023-07-08:
Would it be possible to instead of upgrading to the latest firmware version, to upgrade to ESPHome instead?

I would like to use https://github.com/joshbenner/esphome-daikin-s21 instead of the default firmware with MQTT.

revk, 2023-07-08:
MQTT `command/Faikin/upgrade https://someurl...` should allow you to fetch any binary.

Or use a lead to reflash directly.

closed 2023-07-08 by revk

XMaarten, 2023-07-08:
Great, thanks! Maybe add this to the documentation, I couldn't find it myself.

revk, 2023-07-08:
[image]

It is in the manual for the library it uses. https://github.com/revk/ESP32-RevK/blob/master/revk-user.md

XMaarten, 2023-08-11:
I am getting a ESP_ERR_OTA_VALIDATE_FAILED error, probably because the image isn't signed? Is there a way to continue without a signed image?

revk, 2023-08-11:
You would need to flash your own boot loader and image to start with. You can then decide if you want signed code or not, and set up your own signing.

I have not set "secure boot" fuses on the modules we sell, so the boot loader requires signing with *my* key, so flashing from scratch with your own code should be possible.

revk, 2023-08-11:
You will need to do a serial flash for that.

But I am working on HomeKit/Matter stuff anyway, so hopefully won't be long before we have that.

Surely ESPHome binary would not talk to the Daikin though - would it?

XMaarten, 2023-08-11:
When I have it flashed to a default ESPHome binary, I can flash it again using EPSHome itself to use https://github.com/joshbenner/esphome-daikin-s21 .  

Serial flash requires me to open the units, and I am always having problems with flashing serial.

revk, 2023-08-11:
I suggest using https://www.amazon.co.uk/dp/B0C15K9ZC4

XMaarten, 2023-08-11:
I wish I found that sooner :). 
Thanks, I just ordered one.

revk, 2023-08-11:
Thanks, I am planning to put way more PCBs on Amazon soon.