revk/xESP32-Faikout #70: Value of "power" attribute in MQTT status message?

njh, 2023-07-09:
I was expecting the value of the `power` attribute to either be `true` or `false` and it usually is, but I was confused to see what it is sometimes a decimal value.

```
Faikin/faikin-bedroom2 {"protocol":"S21","ts":"2023-07-09T10:46:00Z","online":true,"home":22.50,"heat":false,"fanrpm":0,"outside":19.00,"liquid":[18.50,18.84,19.00],"power":0.88,"mode":"C","temp":[22.00,22.24,22.50],"fan":"Q","swingh":false,"swingv":false,"econo":false,"powerful":false}
```

Is this deliberate?

revk, 2023-07-09:
That message is a summary for last minute, and so has min/ave/max for temperatures, and a proportion of time that is true for booleans. I should put that in the manual.

closed 2023-07-11 by revk

njh, 2023-07-11:
Sorry, I have been looking for the changeset where this was documented, but I can't find it?

Is this specific to the ESP32-Faikin codebase, rather than ESP32-RevK?

revk, 2023-07-12:
`Advanced.md` already has

*The reporting data for logging contains the controls and status, but has arrays of min/ave/max for temperatures, min/max for target temp, and value between 0 and 1 for booleans when not all the same for the period. For enumerated codes (fan, mode) the data is a snapshot.*

njh, 2023-07-12:
I see you have just updated [Advanced.md](https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Advanced.md#) to add information about the `Faikin/` topic prefix - that covers what I wasn't understanding, thank you!