revk/xESP32-Faikout #71: SET button hidden since last update

andreground, 2023-07-09:
Hi,

Since last update, the "SET" button is disappeared on all the units.
Even forcing visibility through CSS, the values aren't updated after submit.

Please see attached screenshot.

[image]

revk, 2023-07-09:
Can you see if you have any errors in browser for web sockets?
The set vanishes if the websocket closes as the fields are sent via websocket.

If I can get an idea of the error I should be able to fix.

andreground, 2023-07-09:
Here we are!

[image]

revk, 2023-07-09:
Well, err, I don't suppose we know why? Working fine here... so a bit odd.

andreground, 2023-07-09:
Hope this may help you fixing the issue.
It's really a mess because everything was working until the last update. I need to change the MQTT server, is there any workaround to apply this change meanwhile you fix the issue?

Thanks

[image]

iancoop, 2023-07-09:
I also found the set button had disappeared, but only from a web browser on my Mac. From the phone it was ok.

revk, 2023-07-10:
Thanks for the feedback. I will investigate more this morning to sort this for you.

revk, 2023-07-10:
OK, it did not have enough URLs... it should work now.

The upgrade button may work. If not, then a power cycle should force an auto upgrade within a couple of hours.

revk, 2023-07-10:
I M going to restructure it, make the settings page the first URL and make it work even if no web sockets. Old school.

revk, 2023-07-10:
I have done another issue, with a change to a simple form post now, should be more robust to future changes.

closed 2023-07-10 by revk