revk/xESP32-Faikout #73: Stylish CTX/FTX compatibility?

davea, 2023-07-11:
Does the module work with Daikin Stylish indoor units such as CTXA15/FTXA25 etc? I see the S21 wire harness listed as a compatible accessory on the Stylish brochure (ref EKRS21).

revk, 2023-07-11:
If it has S21 lead, then probably. Someone else may be able to confirm. The wiki has a list of known compatible devices.

closed 2023-07-18 by revk

shikousa, 2023-11-26:
Is there already anybody who is using the module for the CTXA/FTXA units? At the moment they are not listed as known working models in the Wiki.

hargathor, 2024-01-09:
Interested since the last update from Daikin bricked two of my 3 devices which are CTXA15AW

hargathor, 2024-03-30:
I can confirm that faikin is compatible with CTXA15AW
You just need a PAP-05V-S connector such as this one https://a.aliexpress.com/_Ev970JX

savage7, 2024-06-23:
@hargathor Can you give me a tip how you opened the unit? I have only one screw at the front. 

[image]
[image]

hargathor, 2024-07-02:
hello @savage7 

you need to remove the overall split as described in the manual. Remove the front cover using the blue thingy underneath (to open it manually you just need to pull gently, otherwise I thing there is a process using the remote control). Then you will be able to remove the main cover. There is three "pins" a the top and three at the bottom. Then you have two screws hidden by the cover where the air is blown. You will also need to unplug the connector on top of your picture.

Then you will be able to remove the cover of your split by gently pulling it. After that you will see a metal cover on the right side of the split. Remove the screws and you will have access to the motherboard and the S21 connector.

Don't forget to switch of the main power first

savage7, 2024-07-02:
@hargathor Thanks taking the time two help me!