revk/xESP32-Faikout #74: What causes state/faikin-1/status to update?

njh, 2023-07-11:
I have been looking at using the `fanrpm` parameter from status messages, to determine if the air-conditioner is _idle_ or not (assuming fanrpm=0 means it isn't actually doing anything when power=true).

But I am not quite clear what causes messages to be published to the `state/faikin-1/status` topic. As far as I can tell changing settings causes a message to be sent - but the changing `fanrpm` doesn't seem to result in messages being sent.

This is one of the reasons why I was getting confused by the `Faikin/faikin-1` topic in #70 - because it updates frequently.

revk, 2023-07-12:
I believe I have a setting to make it send those more often. I'll have to double check.

revk, 2023-07-12:
Looking at the code, a status change is set and so should trigger a report, if a value changes by 0.1 (for floats) or by 10 (for ints like rpm). It should also report if the mode changed, or periodically, but not until any pending changes are complete.

Using debug, can you confirm the `fanrpm` is changing in the S21, but we are not reporting?

joshbenner, 2023-07-12:
While testing my unit, I learned that the fan will continue to blow at all times in AC mode. A possibly better indicator for idle was the compressor sensor (`Sd` response to `Rd` query) at 0 when idle.

revk, 2023-07-12:
Do we pick those up - `liquid` or `gas` temps maybe?

If not, post some examples of the S21 and what they mean and I can add.

joshbenner, 2023-07-12:
Here is where I parse in esphome-daikin-s21: https://github.com/joshbenner/esphome-daikin-s21/blob/main/components/daikin_s21/s21.cpp#L326-L328

Looks like it's just Sd followed by typical 3-char number. IIRC, it matches the value reported as "compfreq" by the Daikin wifi controller.

revk, 2023-07-12:
OK, I'll look a bit later, and see if we can add compressor frequency

revk, 2023-07-12:
I have adde `comp` for that, see what you think...

njh, 2023-07-12:
> Looking at the code, a status change is set and so should trigger a report, if a value changes by 0.1 (for floats) or by 10 (for ints like rpm). It should also report if the mode changed, or [...]

When sending commands, I receive `/status` messages when changing mode or target temperature. But I don't seem to receive `/status` messages as things like the `liquid` temperate changes or the `fanrpm`.

I have verified this by checking the web console and making WebSocket requests:

```
wscat --connect ws://faikin-1.local/status
```

When I turn the air conditioner on, I can see the fan speed ramping up, as I repeatedly request /status over a WebSocket. But I don't see any messages to `state/faikin-1/status` until something else changes.

revk, 2023-07-13:
I don't remember it being an exception in the code, but I'll take another look. It is a bit of a trade off against sending messages all the time, but I could have a setting to control it maybe.

revk, 2023-07-13:
Ok re-reading code, can you do a `setting/whavetername/livestatus 1`

That should force a report on any change.

closed 2023-07-13 by revk

njh, 2023-07-14:
Sadly I have been unable to get this to work.

I have tried:
```
mosquitto_pub -h 'mqtt.local' -t 'setting/faikin-1/livestatus' -m '1'
```

And:
```
mosquitto_pub -h 'mqtt.local' -t 'setting/faikin-1' -m '{"livestatus": true}'
```

But no additional messages seem to get published to `state/faikin-1/status` 😞

revk, 2023-07-15:
If you send setting/faikin-1 with no payload do you see a response with all the settings?

njh, 2023-07-15:
Nothing initially and then:

```
2023-07-15T11:59:27+0100 [retain=0]: state/faikin-bedroom2 {"up":false}
```

revk, 2023-07-15:
Is it `faikin-1` or `faikin-bedroom2` ?

njh, 2023-07-15:
Sorry, yes, tested with two devices.

I have upgraded to f4e439c but behavior is the same.

revk, 2023-07-15:
OK I'll see what I can find.

revk, 2023-07-16:
I don't think the release yesterday had the increased stack size. I'll release now. The setting command with no payload should not at least confirm settings, and may possible mean the problem is fixed.

njh, 2023-07-16:
Just upgraded to b6d8955 and I now receive a message back, when sending an empty message to `setting/faikin-1`, listing the current settings. Very useful 👍 

Sadly it hasn't solved the problem of getting update messages, when the fan speed changes.

revk, 2023-07-16:
OK, well, progress anyway, so `livestatus` is `true`.

I'll check the code again.