revk/xESP32-Faikout #76: Which firmware does the board on Amazon use?

njh, 2023-07-13:
After getting in a bit of a mess with the firmware version with [no "Set" button](https://github.com/revk/ESP32-Faikin/issues/71), the OTA button wasn't working either and so I ran:

```
mosquitto_pub -h 'mqtt.local.' -t 'command/faikin-1/upgrade' -m 'http://ota.revk.uk/Faikin-S1.bin'\n
```

Which I think was a mistake. It seems to work fine apart from the LED being stuck on solid Green!

The following binaries are available but I am not sure which hardware uses each of them:
* [Faikin-S1-PICO.bin](http://ota.revk.uk/Faikin-S1-PICO.bin)
* [Faikin-S1-SOLO.bin](http://ota.revk.uk/Faikin-S1-SOLO.bin)
* [Faikin-S1-V1.bin](http://ota.revk.uk/Faikin-S1-V1.bin)	 
* [Faikin-S1.bin](http://ota.revk.uk/Faikin-S1.bin)

I think the Faikin board available on Amazon should actually be `Faikin-S1-PICO.bin`, because the [BOM](https://github.com/revk/ESP32-Faikin/blob/main/PCB/Faikin/production/bom.csv) lists a ESP32-PICO-MINI-02?

revk, 2023-07-14:
Yes it is the PICO one.

njh, 2023-07-14:
Thanks. I have replaced the firmware with the correct one and the LED is behaving normally again :)

closed 2023-07-14 by njh