revk/xESP32-Faikout #77: Faikin reports system is offline

HA-Ben, 2023-07-14:
Hi,

Just installed the Faikin module. Connects well to WiFi and can navigate to webpage.
Also MQTT is up and running to my HA instance.

When navigating to the Faikin homepage to control the airco I see under the under the controls:
System is off line
[image]

Am I doing something wrong or is my air conditioner not compatible? 
I have connected the Faikin module to the S21 bus on the control board.
Model: FTX25K2V1B (2016)

Any ideas on how to fix this? Thank you for the support.

Bernard

HA-Ben, 2023-07-14:
Just noticed another thread with same issue. Topic "FTXP25M5V1B9 no connection"
A few minutes ago it was confirmed Di-rk that the modified board with mosfet works :)
How can I get one? 
Thanks for the support

revk, 2023-07-15:
I have some samples here, send me your address, I'm on mastodon at https://toot.me.uk/@RevK

revk, 2023-07-15:
In the post

closed 2023-07-15 by revk

HA-Ben, 2023-07-21:
Received the updated version and it works! Thank you for the quick support.

Bernard