revk/xESP32-Faikout #78: Wifi hotspot not closed after connect to wifi network

Aldoszx, 2023-07-20:
I would like to spot this. 
After connecting to my wifi network, the Faikin AP is still visible on wifi networks list.
I think it should not be like that.
Also, in the serial console appears this error: 
(13100) httpd: httpd_server_init: error in listen (112)

Thank you !

revk, 2023-07-20:
That is odd, it hangs around after initial config for around a minute, or should do. It definitely should not stay beyond that. It should not start the AP mode at all if there is a working wifi setting.

The error in listen is odd, have you updated to latest s/w?

Aldoszx, 2023-07-20:
Yes, I have the latest version, installed today.
Also, I can't see the Powerful option and the icons are messed up mostly (see attached picture).
I have to mention that I am not connected to any AC. I'm only test the installation on ESP32-WROOM-32U board.
[image]

Aldoszx, 2023-07-20:
> That is odd, it hangs around after initial config for around a minute, or should do. It definitely should not stay beyond that. It should not start the AP mode at all if there is a working wifi setting.
> 
> The error in listen is odd, have you updated to latest s/w?

After few tests, it seems that the AP is started after a reset but disappears after a couple of minutes.

revk, 2023-07-20:
The AP should only start if the WiFi settings don't work. But I can test here to confirm.

The icons being wrong is very odd as that was fixed ages ago, it was just a character set setting in the http responses. Again, I can retest here to make sure it is right.

Aldoszx, 2023-07-20:
Thank you for your answer !
Looking forward for updates !
I will came back later with feedback after I have all set and connected to my AC unit.
Best regards !

revk, 2023-07-20:
I have checked here, definitely seeing `Content-Type: text/html;charset=utf-8` so icons should work.

Also definitely not seeing AP mode start after restart. You do have correct WiFi settings?

My current version is `2a856d5 2023-07-20T06:09:03` from `ota.revk.uk`

Aldoszx, 2023-07-20:
I will try to access the web interface from another computer.
Did that and the problem with the icons also appears on two different computers on my network.
Indeed, from my iPhone the icons are shown correctly.
Regarding the AP, I have the correct wifi settings, the ESP connects to the network, but I can see the SSID on the computer where the ESP32 is connected to and also on my phone. It appears after the restart of the ESP32 and stays there for about 4-5 minutes or so.

[image]
[image]

Indeed, on another laptop I have in the same network, I cannot see the AP entry for Faikin.
I is strange !
Also, sometimes after restart, some fan modes icons disappears from web interface.
It comes back after a few page refresh.

[image]

revk, 2023-07-20:
I suppose the icons could be your computer/browser lacking the current unicode symbols perhaps? The fact they appear as squares suggests maybe the utf-8 encoding is correct. In fact the ½ showing means it is utf-8. I can only assume a very old browser? The fact it shows from your iPhone probably means there is nothing really to "fix" here.

I still don't understand the AP appearing.

Aldoszx, 2023-07-20:
I have the latest versions of Firefox, Edge and Chrome browsers, and I've tested on 3 computers, the result being the same ! 
Anyways, no big deal, just wanted to mention the issue. :)

revk, 2023-07-20:
Does [Unicode power symbol](https://www.compart.com/en/unicode/U+23FB) work for you?

Aldoszx, 2023-07-20:
Unfortunately, no !
[image]

revk, 2023-07-20:
Ah, that is a clue. I mean that is unicode version 9.0 (June 2016), so only 7 years old, maybe your browser has not caught up.

Aldoszx, 2023-07-20:
I don't know !
Or maybe, the font family used by my browsers are not compatible with this character.

revk, 2023-07-20:
General unicode characters like these are not font family specific, i.e. fall back to generic emoji, etc.

Aldoszx, 2023-07-20:
Yeah, but it it is strange that the character is not displayed correctly even on a fresh installed Win 11 computer.

revk, 2023-07-20:
I'll consider changing to something different, but I really thought a standard character from 7 year old unicode would be fine, honest.

Aldoszx, 2023-07-20:
I think you are right, but there are a lot of settings "around the world" that can not be compatible with your specific coding.
I would suggest using a picture instead of a symbol, maybe will be more compatible.

revk, 2023-07-20:
Unicode is pretty universal - and 7 years is a long time - I am shocked that something so basic is not working. I think going back to the days of 1963 US-ASCII only and having GIF images for everything else would probably be a bad move.

Aldoszx, 2023-07-20:
I don't know. Maybe you should set the text encoding as UTF-8 not unicode.

revk, 2023-07-20:
Err, I have set to utf-8 (which is a way to encode unicode characters). I said that... It is working for the ½ unicode symbol!
[image]

Aldoszx, 2023-07-20:
Indeed, it seems to be OK from your part !
One of my colleagues also tried from his phone, (an old Android 7.0) and has the same issue.
So...

JeanOlivier, 2023-07-20:
> I have checked here, definitely seeing `Content-Type: text/html;charset=utf-8` so icons should work.
> 
> Also definitely not seeing AP mode start after restart. You do have correct WiFi settings?
> 
> My current version is `2a856d5 2023-07-20T06:09:03` from `ota.revk.uk`

FYI, I really don't mind, but icons have never worked for me on either of my 2 Faikin units, got the first one a few months ago. 

I also can't see the "powerful" option even when plugged on my unit with everything else working perfectly. I attributed that to a model-specific variation of the setting and just ignored it so far.

revk, 2023-07-20:
Yes, `powerful` will only come up if we get the appropriate S21 responses.

If you think I have it wrong, please do get me some debug S21 for it, and how it changes with the remote control, and I'll change it. S21 is not something for which I can find any proper specification, which is a shame, it is all people like me reverse engineering. So it is quite possible there is a variant I don't know.

As for icons not working - I don't use windows or android and so did not know they had such a problem with basic unicode characters. I may see if I can find icons that work better. But I do things 7 years should be long enough for such icons to be deployed now, really.

revk, 2023-07-20:
If you got on amazon, please do leave a review.

JeanOlivier, 2023-07-20:
> If you got on amazon, please do leave a review.

I did, and it was glowing, but it was on your previous listing.
I'll check if I can leave a review on the new one even though I didn't "buy" this one specifically.

revk, 2023-07-20:
Ah, yeh, Amazon messed me about on "brand" and would not allow an edit, so new listing, sorry about that.

JeanOlivier, 2023-07-20:
No problem, I just left an honest review (aka glowing).
It might show as coming from an "unconfirmed buyer" though, because of the aforementioned.

Aldoszx, 2023-07-27:
Maybe this could help, regarding power symbol:

https://unicodepowersymbol.com/font/

I had to install this font in order to see the symbol on Firefox, but in Chrome does not work.
It seems Windows does not have those fonts installed by default.
The embedded web fonts could be a solution.

revk, 2023-07-27:
For a 7 year old unicode character, which should not even need a special font really, it seems crazy.

Aldoszx, 2023-07-27:
I think that's right, but this is the true about windows

Amtho, 2023-07-28:
To be fair the power symbol is also not working for me, but the rest :D. Version is compiled 20 days ago or so:
[image]

revk, 2023-07-29:
I'll change to 0/1 for now, silly though.

closed 2023-07-29 by revk