revk/xESP32-Faikout #82: Faikin reports system is offline

FrancescoFdd5, 2023-07-27:
Hi,

I have just installed the Faikin module, it connected successfully to the WiFi, and I can access the webpage without any issues.

However, when I access the page, it gives this error:

System is offline

I have two AC units, and both are behaving the same way:

FTXM25M2V1B
FTXM35M2V1B

Do you have any ideas on how to resolve this?

Thanks,
Francesco

[image]

revk, 2023-07-27:
Not listed on the "confirmed working" list. I wonder if they have the same 5V working issue.

Where are you located?

revk, 2023-07-27:
That may not be a problem, but may want 5V working on Tx

FrancescoFdd5, 2023-07-27:
How can I proceed? What tests can I perform to make it work?

revk, 2023-07-28:
I suspect you need a slightly different circuit. I should have some next week, and could post out to you.

NLNoID, 2023-07-28:
Hello, just installed my Faikin (received in NL through Amazon) and I have the same issue on a FTXM25R2V1B and FTXM35R2V1B. Could this be the same issue?

revk, 2023-07-28:
It could be, and is one reason we have a list of known working units!

Once I have new ones with 5V Tx working, I can send some out.

Assuming this is the problem.

FrancescoFdd5, 2023-07-28:
This could be great!

NLNoID, 2023-07-28:
Yes, noted but figured that the when the FTXM20R2V1B is working (which is
on the list) the 25/35 would also work but unfortunately not it seems...

revk, 2023-07-28:
Expecting new boards end of next week.

NLNoID, 2023-07-28:
Ok, thanks for the update.

Amtho, 2023-07-28:
Im using your Software on a FTXM20R2V1B without issues, however Im using my own board, schematics attached:
[image]

revk, 2023-07-28:
Which rather suggests it is indeed unhappy with 3.3V tx, so new board should just work.

revk, 2023-08-02:
I think a couple of people are waiting for replacements. I have some here now, just need address. DM me on https://toot.me.uk/@RevK

closed 2023-08-03 by revk