revk/xESP32-Faikout #83: Disable automatic OTA updates

pfak, 2023-07-28:
> Note, automatic upgrade from ota.revk.uk is enabled. See instructions to make changes.

But the instructions do not seem to cover disabling OTA. I do not see any MQTT command to disable OTA.

Since I have a number of these units, I prefer to apply the updates in a controlled manner so if anything breaks I can deal with it.

revk, 2023-07-28:
See [RevK library](https://github.com/revk/ESP32-RevK/blob/master/revk-user.md) Software updates...

closed 2023-07-28 by revk

pfak, 2023-07-28:
Thank you @revk. It's not clear to look at another markdown file in a different Github repo, I looked at all the files in ESP32-Faikin repo, searching for OTA :(

revk, 2023-07-28:
I'll add a note on it.