revk/xESP32-Faikout #84: Compressor frequency entity missing Home Assitant

pfak, 2023-07-28:
With the BRP module + Daikin component, I see the frequency for my compressor as an exposed entity.

I also see the JSON output under `status` in MQTT:

```
{"protocol":"S21¬Tx","online":true,"home":21.0,"heat":false,"fanrpm":730,"comp":300,"outside":26.5,"liquid":7.0,"power":true,"mode":"C","temp":18.5,"fan":"Q","swingh":false,"swingv":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
```

However I do not see a corresponding entity in HomeAssistant:

[image]

Is this expected? Is there a way to expose it to HomeAssistant?

revk, 2023-07-29:
Should be easy enough. I don't suppose you know the HomeAssistance config for a compressor fan speed?

pfak, 2023-07-29:
https://github.com/home-assistant/core/blob/dev/homeassistant/components/daikin/sensor.py#L121

?

pfak, 2023-07-29:
I think you need to divide by 10 on the integer you are reporting. I was at 30Hz according to my BRP adapters when I pasted the JSON.

revk, 2023-07-29:
Hmm, the multiply by 10 may be something I have to configure. I'll have a look.

revk, 2023-07-29:
OK, I am not set up to test HA right now, but have added.

Upgrade and see if that does what you need...

apos-t, 2023-07-29:
I've upgraded to the new version and see new entities in ha, but without a value after 15min
[image]

revk, 2023-07-29:
Ah, well spotted, I added to the config but did not send the values, oops. Try again.

apos-t, 2023-07-29:
Comp is now available, but fan is still missing

revk, 2023-07-29:
OK, we have a clash, as I have `fan` for the fan mode, and `fan` for speed. I'll change one. One moment.

apos-t, 2023-07-29:
fanrpm and comprpm are now there - comprpm still fine. fanrpm stays at 0 should be 350 in my case

revk, 2023-07-29:
Is the fan rpm reporting over MQTT correctly?

revk, 2023-07-29:
OK, this would have been easier if I had set up an HA here, sorry. So a tad iterative as I don't have a test system here. Try again.

apos-t, 2023-07-29:
no problem, looking good now 
[image]

revk, 2023-07-29:
Is 14rpm correct? I removed the *10 from the code, but 14rpm is very slow.

apos-t, 2023-07-29:
you right should be 140, on an other unit on the same outside unit I see 140 in the mqtt logs.  If I understand it right the compressor is a value of the outside unit

revk, 2023-07-29:
I'll reinstate the *10, same as the `fanrpm`. Making a release now.

closed 2023-07-29 by revk

pfak, 2023-07-29:
Compressor is *Frequency* (Hz), not RPM. It should be `$Integer/10` for the correct value, and exposed as `UnitOfFrequency.HERTZ`,.

For example on the 4MXS36RMVJUA 4-head heat pump, the frequency (Hz) 15hz to 110hz:

[image]

This output is from the BRP069B4x. Like the "Outdoor Temperature", it's an overall system value.

revk, 2023-07-30:
OK, that is easy enough to sort, thanks. I'll do another release later today.

revk, 2023-07-30:
OK new issue being made now.

pfak, 2023-07-30:
Thank you ! The value is correct, however. Some changes to make it look proper in HA:

`comp`:

* `unit_of_meas` should be `Hz`
* `icon` should be `mdi:sine-wave`

`compfan`:

* `device_cla` does not need to be set (default in HA), and should NOT be `frequency` (so please unset)
* `icon` should be `mdi:fan`

Side note: I am slightly confused why `unit_of_meas`, and `device_cla` works even though it should be `unit_of_measurement` and `device_class`.

revk, 2023-07-30:
I'll make that change, and yes all of the config names have a shorter form version as well.

pfak, 2023-07-30:
Almost there.. Fan RPM is now a sine wave instead of a fan.

[image]

revk, 2023-07-30:
Hmm, I am not setting icons for fan or temp, so maybe just telling it is Hz will get a sine wave. They are the same function at present, but I can change. I'll try with Hz and no icon first though - see if that is any good.

pfak, 2023-07-30:
'fanrpm' is RPM, not Hz.

revk, 2023-07-30:
Yes, I mean does it do sign wave for comp now or am I going to have to set icons for each?

pfak, 2023-07-30:
Will check with your change in https://github.com/revk/ESP32-Faikin/commit/7433db7bfdc412c809cea0102b2bd49974806cad

pfak, 2023-07-30:
[image]

Looks like you need to set icons for each. I think the sensors inherits the icon from the device? above it? maybe?? I noticed that the `climate` entry matches all the sensors, and IMO it is wrong, it should be `mdi:thermostat`.

So probably to stay within HA standards:

main device: `mdi:thermostat`
comp `mdi:sine-wave`
compfan `mdi:fan`
liquid `mdi:coolant-temperature`
outside `mdi:thermometer`.

pfak, 2023-07-30:
Here's BRP device in HA:

[image]

Here's ecobee in HA:

[image]

pfak, 2023-07-30:
(Side note: Interesting that BRP can report 1/2 degree increments for outdoor temperature, but Faikin cannot?)

revk, 2023-07-30:
Ok fine I'll work out icons.

revk, 2023-07-30:
I am logging outside temp in ½ degrees, and reporting as such - is S21 only sending whole degrees I wonder?

pfak, 2023-07-30:
All good now! Half degree shows too (I dont think you made any changes to this, I just see it now)

[image]

XMaarten, 2023-11-06:
> 'fanrpm' is RPM, not Hz.

This was mentioned a couple of times here, but I think it was missed :).

I am getting errors on startup about this:
`Entity sensor.fanrpm (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'rpm' which is not a valid unit for the device class ('frequency') it is using; expected one of ['MHz', 'GHz', 'Hz', 'kHz'];`

Can you please change unit_of_meas to Hz instead of rpm?

revk, 2023-11-07:
Ok I'll have to change the name and about to be Hz then.

revk, 2023-11-07:
OK new code issues, let me know if that makes sense.

XMaarten, 2023-11-07:
> OK new code issues, let me know if that makes sense.

Yeah, that seems to work. I can see the new fanfreq value (11.5 Hz in low fan speed) in HomeAssistant. Thanks!

You could maybe consider to put all the entries below /homeassistant/sensor/ in a subcategory /homeassistant/sensor/faikin/ and name them using the device name instead of the mac address. So /homeassistant/sensor/faikin/aircolivingroom-blehum for example.  It works without any issues at the moment, but maybe more readable in MQTT itself.