revk/xESP32-Faikout #87: Home Assistant 2023.8.0 changes

patrickli, 2023-08-04:
Upgraded to `2023.8.0` and there are a few warnings appearing in the log.

### not a valid preset mode

```
'{"online":true,"target":18.00,"temp":17.50,"outside":15.00,"liquid":19.50,"comp":0,"fanrpm":0,"mode":"off","fan":"auto","swing":"off","preset":"home"}' received on topic E89F6D31A200. 'home' is not a valid preset mode
```

This is because in the autodiscovery payload we have `'preset_modes': ['boost', 'normal']`, and `home` is not one of `boost` or `normal`. Either we need to add all modes into the autodiscovery or report a valid `preset`.

### fan speed unit

```
Entity sensor.fanrpm (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'rpm' which is not a valid unit for the device class ('frequency') it is using; expected one of ['MHz', 'kHz', 'GHz', 'Hz'];
```

Not sure how to fix this as there is no way to specify unit in autodiscovery payload. I will try to do a customize for the sensor and see if that fixes it.

### MQTT device name is equal to entity name in your config

This comes up as big repair warnings in the settings page.

[image]

In the discovery payload we have something like:

```
{
  "unique_id":"0C8B9596F510",
  "device":{
    "identifiers":[
      "0C8B9596F510"
    ],
    "name":"0C8B9596F510",
    "sw_version":"e084988",
    "manufacturer":"RevK",
    "configuration_url":"http://0C8B9596F510.local/"
  },
  "icon":"mdi:thermostat",
  "name":"0C8B9596F510",
  "min_temp":16.0,
  "max_temp":32.0,
  "temperature_command_topic":"command/0C8B9596F510/temp",
  "temperature_state_topic":"0C8B9596F510"
}
```

I think what it is complaining about is that the top level `name` value is the same as `device.name`. According to the documentation we should set the top level `name` to `null`.

revk, 2023-08-04:
Will look now.

"Home" to "normal" was a silly mistake. Did not realise HA was so fussy.

patrickli, 2023-08-04:
It certainly wasn't this fussy before.

patrickli, 2023-08-04:
> Will look now.
> 
> "Home" to "normal" was a silly mistake. Did not realise HA was so fussy.
> 
> The other warnings I'm happy to take any advice on.

I don't have a good idea for fan RPM for now. For the last issue you can just set the top level `name` to `null` in the discovery payload.

revk, 2023-08-04:
OK changed back to "home" now, do update.

If you have specific advice for changes for other warnings, please let me know and I will make changes.

As for `name`, just to clarify, this needs to be `"name":null` yes?

[image]

patrickli, 2023-08-04:
Yes. not the one in `device`.

[image]

revk, 2023-08-04:
And they definitely mean `null`, not simply omitting `"name":` ?

patrickli, 2023-08-04:
That's what the doc says. So give that a try first.

revk, 2023-08-04:
OK, issuing now with `"name":null`, let me know - I'm going to get breakfast.

patrickli, 2023-08-04:
Ok the two issues has been resolved now. I'll have a think about the fan
rpm.

Also I noticed there are new "comp" sensor appeared. What does "comp"
stand for?

revk, 2023-08-04:
Compressor.

patrickli, 2023-08-04:
Some support readings regarding the change of name to null: https://www.hackster.io/news/home-assistant-2023-8-brings-new-mqtt-related-error-messages-but-don-t-worry-say-maintainers-bec84d16710e

revk, 2023-08-05:
May be neater to omit it then.

patrickli, 2023-08-06:
Closing the issue as can't do anything about the warning for unit of the fan. It is just a warning anyway.

closed 2023-08-06 by patrickli