revk/xESP32-Faikout #90: cant save settings on 8faada5

pfak, 2023-08-05:
I entered wrong MQTT password, and no settings will save on the Web UI. It says "Settings saved", but they don't actually update the settings ..

pfak, 2023-08-05:
if I remove the MQTT information all together, then I get Rebooted and settings save

pfak, 2023-08-05:
nevermind that only worked once

pfak, 2023-08-05:
Settings in general do not want to save. Tried in Firefox, Edge and Chrome.

It always just says "Settings stored." but never reboots.

revk, 2023-08-06:
That is weird? Settings are saving in tests here.

Can you check if any browser debug shows any clue. Though if it says settings stored that suggests it is posting Ok.

Does clearing everything out, even WiFi, save? Yes you will have to connect to it as AP to fix that. If settings do save.

I'll recheck the code anyway.

revk, 2023-08-06:
Just to double check - you are actually changing a setting in these cases, yes? It won't reboot if no changes.

I would upgrade to latest code as well if you can.

pfak, 2023-08-06:
I am now on e72ea1c. Every time I make a change to settings it now says "Rebooting" and the settings work. Before it would just say "MQTT Connecting" and not work. **Is the issue if the MQTT username is wrong then settings cant be saved??**

After rebooting for the OTA, the MQTT username is correct from what was supposedly 'saved'. I can confirm it was not working before because I had to create the wrong username for MQTT to be able to control via Home Assistant.

Previously I had tried to change Timezone, MQTT username, MQTT password.

There is one error occasionally, but this does not affect saving/rebooting:

[image]

revk, 2023-08-06:
OK, that really is weird - short of some special characters that are not being handled properly, or some weird caching, I am at a loss.

So working now? Apart from referencing the send button? That is odd.

pfak, 2023-08-06:
I can reproduce the issue.

If MQTT is stuck in "Connecting", i.e. username is wrong, then it will not reboot the device. How do I reboot it? :)

pfak, 2023-08-06:
[image]

pfak, 2023-08-06:
It also does not appear to ever exit the "MQTT Connecting" loop, even when I create the user that I set the settings to.

pfak, 2023-08-06:
It also says "No changes" if I try to change the username.

pfak, 2023-08-06:
Blanking MQTT username, password, host
"Settings stored", stays at that page
No "Rebooting"
Go back to Settings page, MQTT username, password and IP are still there.

revk, 2023-08-06:
"MQTT Connecting" is just the state of the MQTT task, it should probably say MQTT failed if the credentials are wrong, and I can look in to that. It may depend on how your MQTT server responds.

It should, however, not have any impact on either storing settings or rebooting, or allowing you to go back to the settings again and making further changes.

But I'll see if I can reproduce with incorrect MQTT settings - is it the hostname wrong, or just username wrong on a correct hostname?

revk, 2023-08-06:
Ooh, I wonder if that is crashing it somehow, I'll check that now.

pfak, 2023-08-06:
Username is just wrong to repro this problem. 

I changed it from "first-floor-faikin" to "first-floor-faikin2". Now I am in the same situation as when I submitted this ticket, where I can't fix the MQTT settings to make the unit work. I also can't change WiFi, Timezone, etc.

pfak, 2023-08-06:
Even creating an MQTT username "first-floor-faikin2" and it seems stuck :D

pfak, 2023-08-06:
Is there a way to force a reboot of the unit via a GET request?

revk, 2023-08-06:
No there is not - it should not need a reboot. Let me see if I can reproduce this at all.

revk, 2023-08-06:
Every change I make is working, storing settings, and rebooting...

Ideally I could do with a concrete example of specific SSID, password, MQTT hostname, username, password that definitely fails for you, assuming timezone is default. It could be some complicated matter with length of the post or something.

revk, 2023-08-06:
Also, when you say "stuck", I assume you can click on "Home", and then "Settings" again, and get back to the form, regardless?

pfak, 2023-08-06:
> Also, when you say "stuck", I assume you can click on "Home", and then "Settings" again, and get back to the form, regardless?

Correct.

Hostname: `First-Floor-DaikinAC`
SSID: `KIESER-IOT`
Password: `15 character string A-Za-z0-9`
MQTT host: `192.168.3.10`
MQTT username (failing on): `first-floor-daikinac2`
MQTT username (actual): `first-floor-daikinac`
MQTT password: `OyrasHejbutBuv1`
Timezone `PST8PDT,M3.2.0,M11.1.0` 

Originally I could not get the timezone off of BST.

pfak, 2023-08-06:
~5-10 minutes after I create the MQTT username `first-floor-daikinac2` it was able to connect, and then I was able to change the username back to `first-floor-daikinac` and now settings save and reboot again.

pfak, 2023-08-06:
But now ...
[image]

pfak, 2023-08-06:
Sorry I accidentally typed `first-floor-daikinac` when I meant to put `first-floor-faikin`. This was juts me typing it wrong it is faikin in the settings.

revk, 2023-08-06:
The only clue is the total post length is limited to 200 bytes, but I don't think you could be getting close unless something is % encoding a lot.

I'll just test a change to that in case it is the cause.

pfak, 2023-08-06:
The clue to me is it only fails when MQTT is stuck in connecting state.
As soon as MQTT is not stuck in connecting state, settings save fine.

revk, 2023-08-06:
The MQTT connecting is not a state for the whole system - it is one task in that state. The issue sounds like somehow it is crashing perhaps which would mean the web page does not reload, and settings may not be committed to flash, and so on. I'll see what I can recreate here.

revk, 2023-08-06:
I am testing a failed MQTT connection scenario now.

revk, 2023-08-06:
I may have reproduced it - and not a crash, which is weird in itself!

revk, 2023-08-06:
OK I have found the underlying issue. I'll work on a fix now, and a work around to get you back running. Bear with me.

pfak, 2023-08-06:
im back up and running, but fixing the underlying bug would be fantastic! thank you for your response

revk, 2023-08-06:
OK found it - was MQTT error handling in a subtle way - issuing fix, please do test.

If you are having trouble changing settings of upgrading, stop your MQTT server. It is the connection failing that is constantly aborting the normal reset. Hence you managing to occasionally save, etc.

closed 2023-08-06 by revk