revk/xESP32-Faikout #92: Availability

staugaard, 2023-08-15:
Hey, This looks like an awesome module. I desperately need one. When are they available again?

revk, 2023-08-15:
It is available now, on the UK Amazon store. Amazon do appear to ship to some other countries. But if all else fails there are forwarding companies that apparently work like https://www.forward2me.com

I used to have a note saying that on the listing but looks like Amazon actually removed it!

closed 2023-08-15 by revk

staugaard, 2023-08-15:
Amazon tells me that it is unavailable

[image]

staugaard, 2023-08-15:
I figured it out and placed an order to ship to New Zealand via forward2me.

revk, 2023-08-15:
So for some reason they are not shipping to New Zealand. It is a mystery why they will ship some places some times and not others. Maybe ask Amazon. They have shipped to NZ before. Or use a forwarding company perhaps.

[image]