revk/xESP32-Faikout #93: HomeKit (matter) work

revk, 2023-08-15:
Try and get matter working with Faikin

[image]

revk, 2023-08-15:
I have managed to build a test matter app and load on a test board. This is progress. I have yet to get as far as any integration with Faikin, yet.

revk, 2023-08-15:
This may be complicated, it looks like some specific flash partitions may be needed, and possibly even some per device "factory set up", which would be a bugger.

revk, 2023-08-16:
This is a nightmare, it seems that the `esp-matter` IDF itself has a load of things like WS1812 drivers, and when trying to change the example LED driver it links in those drivers regardless rather than letting me use the later managed component. This makes no sense - why would such a driver be ing the matter IDF anyway?

I may have to wait for matter to be more mature, a simple `idf.py add-dependency asp-matter` would be lovely, but we do not seem close to that at the moment.

I'll carry on working on trying to integrate matter in to my build systems anyway.

njh, 2023-08-17:
It does sound like a maturity problem. Are there any alternatives?
Thanks for looking into it!

revk, 2023-08-17:
Agreed, I have been looking for a while, so not really sure yet. Still working on it.

revk, 2023-08-17:
Driving me mad, but getting closer every day.

njh, 2023-08-20:
I am wondering how long it will be until you decide it would be easier to just write your own Matter library 😀

revk, 2023-08-20:
I have made a lot of progress testing this.

Key issues remain

- Ensuring that it can work without matter in the same way as now, with WiFI AP set up, and controls for WiFi via the web interface.
- Ensuring MDNS works for simple `name.local` which currently clashes with Matter system.

If I can make it all work I can upgrade current code. For now I am going to be working on a different build for this, which will be available for people to test with soon.

njh, 2023-08-20:
> If I can make it all work I can upgrade current code. For now I am going to be working on a different build for this, which will be available for people to test with soon.

Good work! I am happy to give it a go, once you have a build ready for testing 👍

revk, 2023-09-04:
OK coming back to this - I now have a nice reference for someone doing things the way I want to do them (i.e. I control wifi, etc, but allow matter as well). This should help a lot.

Cptmeme, 2023-11-14:
hi, I just bought one of your boards from amazon and was wondering if there were any updates regarding matter support? (not trying to be annoying, just interested) Thanks for this entire project!!

revk, 2023-11-15:
Slightly stalled for now, but still planned.

Cptmeme, 2023-11-15:
Okay great, thank you!

revk, 2024-05-29:
It is going on hold, sorry. One day, but to be honest, right now, HA seems a better choice for a lot of people.

benwa, 2025-02-21:
Is there any code for this? Hoping things can pick back up after May of 2024.

revk, 2025-02-21:
Somewhat stalled, sorry. I managed a basic proof of concept for matter, but we could never sort a certified matter install, and people using HA so much has kind of taken over (and HA can then work with matter).

benwa, 2025-02-21:
Are you able to share the POC? I'm interested to see if I can get it up and running.

revk, 2025-02-21:
The PoC was not Faikin code, just using my library on my LED project, sorry. Probably not helpful.

benwa, 2025-02-21:
Ah, I understand. Thank you though!