revk/xESP32-Faikout #96: FTX35KNV1B Reports System is Offline

jesusvallejo, 2023-08-18:
Hi, i just bought from amazon the module, i have set it up, but reports system is Offline. I manage to find #77 and #66 issues. Is it possible that i'm affected in the same way? I can solder, so in case there is a fix i could try to perform it. I'm also in possession of an  oscilloscope in case its of any help. Cheers

revk, 2023-08-18:
Can you send picture of your module, so I can check which one it is.

jesusvallejo, 2023-08-18:
[image]
[image]
Those are the best pictures i could take.

revk, 2023-08-18:
OK, yes, they are the older model that does not do 5V working. Where in the world are you?

jesusvallejo, 2023-08-18:
Spain, Madrid.

revk, 2023-08-18:
Makes posting you a replacement a tad more expensive. Are you able to DM me on https://toot.me.uk/@RevK with your address.

revk, 2023-08-20:
A new one is in the post so let me know if it does not work.

closed 2023-08-20 by revk

jesusvallejo, 2023-09-09:
Received it and working. Thank you very much.

revk, 2023-09-10:
Glad to hear it.