revk/xESP32-Faikout #100: Add Temperature_Unit to MQTT discovery for climate entity

philh30, 2023-08-27:
Great device - arrived yesterday and installation/setup were a breeze. I've run into on problem which is that Home Assistant is treating all temperatures from the climate entity as degrees Fahrenheit with no conversion (e.g. 23 C is reported as 23 F). The entity for outdoor temperature is properly converting from Celsius to Fahrenheit.

This is happening because the MQTT discovery topic doesn't specify a temperature unit for the climate entity. Adding `"temp_unit": "C",` to discovery causes HA to properly convert from Celsius to Fahrenheit.

revk, 2023-08-27:
Oooh, surprised I missed that - I'll have a look now.

revk, 2023-08-27:
OK I have `"unit_of_meas": "°C"` - I thought that is what it needed. I can add `"temp_unit":"C"` as well perhaps.

revk, 2023-08-27:
Ah, to the `climate` entry, Ok I'll add there... New code issue in a few minutes.

philh30, 2023-08-27:
I currently have my discovery topic jammed with `"temp_unit":"C"` and the conversion is working properly. Separate but related issue is that sending setpoints to the Faikin with HA in Fahrenheit works but shows errors in MQTT. For example:

- Set cooling setpoint to 71 in HA
- HA sends `temp=21.666666666666668` to Faikin
- Faikin throws `failed-set={"temp":21.7}` in MQTT
- Faikin reports setpoint is 21.5 C, which converts to the desired 71 F in HA

Setting to 68 F where the math works out perfectly to 20 C results in no error. Seems like this is stemming from something (I'm assuming my Daikin unit) only reporting in increments of 0.5 C.

revk, 2023-08-27:
Ok tricky.

I may be able to fix the error. The S21 interface works in ½°C steps, so it should, if S21, be trying to set to 21.5°C not 21.7°C as closes to 21.6666°C and then not error. But the X50 allow 0.1°C steps. So most of the code allows 0.1°C hence trying 21.7°C as closest to 21.66666°C.

The "fix" would be to make sure the settings know that S21 is ½°C steps only. I'll take a look.

revk, 2023-08-27:
Ok I have another fix for that - making temperature setting for S21 round to nearest ½°C so it will not error when it comes back as that. Code in a few minutes.

revk, 2023-08-27:
Do close issue if happy, and rate on amazon if you got from there :-)

philh30, 2023-08-29:
Thanks! Looks good - reporting properly in F and that error is gone. Will be happy to rate on amazon!

closed 2023-08-29 by philh30

revk, 2024-01-30:
OK set `otahost` to `test.revk.uk` and upgrade, now has Fahrenheit option for web interface. Let me know.