revk/xESP32-Faikout #101: the S21 Protocol has Power Meter ?

metarutaiga, 2023-09-04:
Is the FM Meter?
Is the FP Pulse?

I checked it on my FTXM-V.
The value is likely my Power Meter closely.

kWh = FM * FP / 1000.

revk, 2023-09-04:
Do you have examples of the exact messages. If we can work out what they are, I am happy to add that.

metarutaiga, 2023-09-04:
FM : 02.47.4D.37.33.31.32.61.03
FP : 02.47.50.45.32.30.30.6E.03

FM : hex(2137) = 8503
FP : hex(2E) = 46
391 kWh = 8503(Meter) * 46(Pulse) / 1000(kilo)

revk, 2023-09-04:
I am not sure what you mean by "meter" and "pulse" to be honest - what are their units?
Also seems very inconsistent to be using ASCII HEX, but possible.

MassiPi, 2023-09-15:
what request are you using to get these values?
and what is this measuring? seems total energy used, since when?

metarutaiga, 2023-09-16:
[image]
This is my Daikin of Home Assistant.
I don't know what is FP / FM, I just guess what it means.
I compared it with my power wires Amp.
So I guess kWh = FP * FM / 1000.

revk, 2023-09-16:
I am not sure how useful a cumulative kWh is? It would help to know what FP and FM are. How/when are they ever reset I wonder.

metarutaiga, 2023-09-16:
https://github.com/home-assistant/core/blob/57337b5cee5a5ab076ab89ded23219b40497e6b8/homeassistant/components/daikin/sensor.py#L132

I know it in the Home Assistant first time.
But now the WiFi module was updated by MobileApp and disabled local HTTP API, So the Home Assistant can't read it anymore.
So I check what the meter command is.

JeanOlivier, 2023-10-30:
> But now the WiFi module was updated by MobileApp and disabled local HTTP API

This is why we need a project like ES3P32-Faikin, oh my.

44616E, 2024-02-04:
This would be very useful for my use case, I was thinking of adding Current transformers on the AC feed, but if the air con can tell me that would be very useful. I'm trying to prioritize solar consumption

revk, 2024-02-06:
OK do we know the units of the component parts yet?

sjaakie09, 2024-03-12:
hello revk i mad some nice images you can also use for others 
[image]
and I like the tool a question did you also did things on using esp32 for metering like connecting it to puls gas or modbus rs485 kwh meters?

sjaakie09, 2024-03-12:
> That pin numbering is backwards, but nice pictures.

Beste, Dear,
is it wrong?

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>

revk, 2024-03-13:
Just the numbering, pin 1 is the 5V.

sm-Fifteen, 2024-03-17:
> I am not sure how useful a cumulative kWh is? It would help to know what FP and FM are. How/when are they ever reset I wonder.

It's pretty standard actually to have some kind of lifetime summation as the metric for energy usage, instead of having the accuracy of that metric depend on how often the server polls the device. Most ZigBee devices that measure electricity will report lifetime energy use over the smart power metering cluster as `CurrentSummationDelivered`, sent as a uint48 with a device-supplied multiplier, divider and unit of measure. 48 bits unsigned is big enough that (even in Wh) it wouldn't wrap around before 248 billion kWh, which seems to be roughly the amount of electricity that a country like Norway consumes over a year. 

Likewise, [Home Assistant doesn't use power sensors directly for its energy monitoring feature](https://www.home-assistant.io/docs/energy/faq/#creating-an-energy-sensor-out-of-a-power-sensor), and you need a virtual sensor that calculates the integration of that power reading if cumulative energy isn't reported.

revk, 2024-03-17:
OK so are they cumulative somehow, what are they and what units?

sm-Fifteen, 2024-03-17:
> OK so are they cumulative somehow, what are they and what units?

OP said they matched the readings on their power meter (in watt-hours) pretty closely when multiplied together, though I guess further testing would be needed to know what either factor represents and what that power consumption covers and excludes (like whether the values always match on multi-zone systems).

I have yet to recieve and install my own Faikin dongles, though, so I can't help with that right now.

revk, 2024-03-17:
Ok "when multiplied together". So what are the two values exactly?

44616E, 2024-03-18:
Can I query these values through MQTT as a custom command? I should be able to compare the values to a smart meter in the board to confirm if it holds true for my units as well.

revk, 2024-03-18:
There are debug messages to allow you to send any S21 command and see replies. The manual should cover it all.

sm-Fifteen, 2024-03-18:
> FM : 02.47.4D.37.33.31.32.61.03
> FP : 02.47.50.45.32.30.30.6E.03
> 
> FM : hex(2137) = 8503
> FP : hex(2E) = 46
> 391 kWh = 8503(Meter) * 46(Pulse) / 1000(kilo)

Do these look like valid custom commands/payloads? `GM7312a` and `GPE200n`? OP called them FM amd FP, but they seem to be GM and GP from the hex codes, which would correspond with most other polling messages being G.

revk, 2024-03-18:
Ok back to what is asked before.

What are "meter" and "pulse" and what units are they?

sm-Fifteen, 2024-03-18:
> Ok back to what is asked before.
> 
> What are "meter" and "pulse" and what units are they?

@revk : I believe that was OP trying to figure out what the values mean based on the ASCII letters, but I don't think they mean anything.

sm-Fifteen, 2024-03-19:
> Do these look like valid custom commands/payloads? `GM7312a` and `GPE200n`? OP called them FM amd FP, but they seem to be GM and GP from the hex codes, which would correspond with most other polling [...]

Oh, I only just found [this page](https://github.com/revk/ESP32-Faikin/wiki/S21-Protocol), I now understand that G is for reply payloads and that the ASCII transcriptions I wrote down are not valid requests, but actually responses. Assuming OP is right (would need more testing to find out whether or not he is) and these are integer values transmitted as hexadecimal ASCII strings (which I guess could make sense for unbounded multi-byte values?), the return value `GPE200n` is to be interpreted as `F/G field "P": 0 0 2 E` and decoded ax `0x002E = 46`.

Should I understand that all one needs to run to get these values is `mosquitto_pub -h 'my.mqtt.server.local' -t 'command/MyDaikinHostname/send' -m 'FM'`?

revk, 2024-03-19:
Maybe

sm-Fifteen, 2024-03-19:
> Can I query these values through MQTT as a custom command? I should be able to compare the values to a smart meter in the board to confirm if it holds true for my units as well.

@44616E: If you have a chance to test this on your end, you can use these commands and see if the readings match with your own measurements.

```
mosquitto_pub -h 'my.mqtt.server.local' -t 'command/MyDaikinHostname/send' -m 'FM'
mosquitto_pub -h 'my.mqtt.server.local' -t 'command/MyDaikinHostname/send' -m 'FP'
```

serimi89, 2024-03-21:
Played around with this today, the FP message (GP result) did not change. It was: {"protocol":"S21","dump":"024750443730307203","GP":"D700"}

Regarding the FM (GM result) messages I could see the below values:
14:18 / AC OFF {"protocol":"S21","dump":"02474D323233335E03","GM":"2233"}
14:50 / AC OFF {"protocol":"S21","dump":"02474D323233335E03","GM":"2233"}
14:51 / AC TURNED ON
14:54 / AC ON {"protocol":"S21","dump":"02474D333233335F03","GM":"3233"}
14:57 / AC ON {"protocol":"S21","dump":"02474D343233336003","GM":"4233"}
15:05 / AC ON {"protocol":"S21","dump":"02474D353233336103","GM":"5233"}
15:11 / AC ON {"protocol":"S21","dump":"02474D363233336203","GM":"6233"}
15:18 / AC ON {"protocol":"S21","dump":"02474D373233336303","GM":"7233"}

So obviously, something is counting. If we compare this to the the kWh on my power meter, it seems like it is incrementing by 0.1kwh every time the GM value increases.

[image]

Ok just like me you were wondering what comes after 9233, so here we go:
{"protocol":"S21","dump":"02474D413233336D03","GM":"A233"}

Hope this helps, If you need anything else, please let me know!

Seb

revk, 2024-03-21:
OK, now makes more sense.

I suspect GP is telling you that it meters pulses every 125mWh and is a fixed value.

I suspect GM is a meter telling you how many GP sized pulses.

OK, now to work out how best to report it, and handle meter wrap around.

Anyone know the HA config that would make sense for a "meter" like this?

revk, 2024-03-21:
I have made a beta with a `Wh` report, see if that makes sense.

No HA config yet, let me know what HA config it would need?

serimi89, 2024-03-21:
Great - just installed it and getting values.
How do you know its 125Wh per pulse? 

I have had some sensors that report Wh and imported it through the Utility Meter integration: https://www.home-assistant.io/integrations/utility_meter/ This can then be added to the energy dashboard to see time based stats.

revk, 2024-03-21:
That is what you said in the GP response. It uses whatever comes in on the GP from your unit.

serimi89, 2024-03-21:
well, I cant see the exact number since the internal meter of the AC and my energy meter are not in sync, both having different starting points, so I assumed it was 100Wh since that ballparks the increase of the energy meter (also its the smallest unit of the ONECTA App).

The values that are reported on the Wh value in the latest beta however are counting down:
159939 
159861

revk, 2024-03-21:
LOL, I'll investigate. It was just the first attempt.

revk, 2024-03-21:
Your quoted D700 with is 125

serimi89, 2024-03-21:
> Your quoted D700 with is 125

Ah! :) 

> LOL, I'll investigate. It was just the first attempt.

Haha.. well now I am at 2333, earlier it was 2233 - so obviously it increments the second digit, then the third, etc.

I dont know though if it counts until 1-Z or A-C or whatever.. 🤣

revk, 2024-03-21:
It's hex, and looks like I had an error, new beta in a moment.

revk, 2024-03-21:
I think this looks better now, try that.

(BTW it seems my unit uses 165Wh pulses).

serimi89, 2024-03-21:
works! got a 125Wh increase :) 

That total number that I am getting must be the total lifetime Wh then?

revk, 2024-03-21:
I expect so, and will wrap around at some point I guess.

I'll have to look in to HA, but glad we got to the bottom of what the fields mean.

sm-Fifteen, 2024-03-21:
> I have made a beta with a `Wh` report, see if that makes sense.
> 
> No HA config yet, let me know what HA config it would need?

Not sure how it works with MQTT-based devices, but the built-in Daikin integration [just registers a few sensors](https://github.com/home-assistant/core/blob/b4c36d46768dd45827847625ed5f067067197c7f/homeassistant/components/daikin/sensor.py#L106-L113) so Home Assistant creates entities for them, and these entities (only the energy ones, in kWh) can then be added to the energy dashboard.

sm-Fifteen, 2024-03-21:
> in order to get a daily consumption value in HA from any sensor value, I used the below code. This will reset itself to 0 every day at midnight and can be added to the energy dashboard as a device, monitoring its power consumption over time
> 
> ```
> utility_meter:
>   energy:
>     source: sensor.energy_in_kwh
>     cycle: daily
> ```

You don't need to make them cycle for them to work in the energy dashboard. My Zigbee devices just keep going up forever, and the dashboard just considers the difference between the value at the start of a time period compared to the end.

44616E, 2024-03-22:
Just tried the beta firmware and the Wh is about 2.8 times higher than it should be. Which seems to correlate with my GP value of 280 (at least that the increment in Wh). Perhaps GM is actually at a fixed multiplier and doesn't need to be multiplied by GP.Will collect some more data over the weekend.I'm also interested if we can get the realtime power value in Watts it exists in that home automation configuration linked to earlier. Not sure how you go from there to the commands.

MassiPi, 2024-03-22:
how is this working in a multi-split solution?
i assume all splits are reporting the same value, that is the "system" consumption, right?
my splits are still in winter mode so i can't test anything atm :)

revk, 2024-03-22:
No clue, sorry.

MassiPi, 2024-03-22:
thanks anyway :)
quite curious to know if these FP and FM are working also on old units (my FTXSxxG splits aren't even officially supported by daikin wifi cards..)

44616E, 2024-03-22:
I don't think these power measurements have anything to do with wifi adapters. I suspect its for building management systems like used in Hotels and offices etc. The daikin app doesn't show power usage.

sm-Fifteen, 2024-03-22:
> Just tried the beta firmware and the Wh is about 2.8 times higher than it should be. Which seems to correlate with my GP value of 280 (at least that the increment in Wh). Perhaps GM is actually at a [...]

@44616E: There are also [other unknown fields](https://github.com/revk/ESP32-Faikin/wiki/S21-Protocol) under the F command that could be involved in power readings:

> The basic polling loop seems to consist of F/G commands: 2, 1, 3, 4, 5, 8, 9, 6, 7, B, G, K, M, N, P, Q, S, T, U, U (again), X, and less frequently R/S commands: H, N, I, a, X, d, L

There might be other values involved in the energy calculations among these, and possibly a real time estimation of power consumption. I'm also guessing state indicators like "Stand by for defrost/hot start" and "Changeover controlled by the master indoor unit" like we see on [the more advanced wired thermostats](http://4ae617e62d.nxcli.net/content/assets/DOC/OperationManuals/BRC1E73-Operations-Manual.pdf) might show up.

serimi89, 2024-03-22:
I am happy to help testing various commands, can someone explain to me how to read / convert the results? as far as I understood they are hex, but also reversed?

serimi89, 2024-03-22:
> I am happy to help testing various commands, can someone explain to me how to read / convert the results? as far as I understood they are hex, but also reversed?

Well, correct me if I am wrong, but some values need to be converted to decimal, others are fine as hex, but its always inverted.

I wonder if FM / GM is not simply counting up the kWh, in my case the value is 13123 (hex: 3343), which could be 1312,3 kWh and which is incrementing by 0.1 kWh as seen [here]( https://github.com/revk/ESP32-Faikin/issues/101#issuecomment-2012334148). This would match the ONECTA App, where 0.1 kWh is the smallest unit. 
I dont have consumption data before last year (2023: ~500kWh), but if its a lifetime value, It could make sense since I have my AC almost 3 years.

44616E, 2024-03-22:
> Well, correct me if I am wrong, but some values need to be converted to decimal, others are fine as hex, but its always inverted.
>
> I wonder if FM / GM is not simply counting up the kWh, in my case [...]

I am thinking the same, my values look very close if we use a 0.1 multiplier. Will test some more tonight.

44616E, 2024-03-23:
"The ‘Outdoor compressor Energy consumption’ and ‘Outdoor compressor Estimated power consumption’ sensors are updated every time 100 Wh are consumed by all different operating modes summed together."

https://www.home-assistant.io/integrations/daikin#sensor

So seems the multiplier is fixed. Do I think GP is not required.

The estimated power consumption would be handy.

metarutaiga, 2024-03-23:
The Daikin integration in HA has wrong power consumption value with my aircon.
It seems that wifi module is read wrong data.

revk, 2024-03-23:
OK so what is GP?

This is why is asked the units in the first place!

I can change to 100Wh multiplier, no problem.

But is GP useful? Or shall we ignore it?

serimi89, 2024-03-23:
> OK so what is GP?
> 
> This is why is asked the units in the first place!
> 
> I can change to 100Wh multiplier, no problem.
> 
> But is GP useful? Or shall we ignore it?

I think the original assumption was just letter based, P = pulse, M = Meter - doesn't seem to be necessary a proof that both are necessary?
Anyway, I have been reading the new Daikin Cloud API I came across this re power consumption, which is exactly the way data is displayed on the ONECTA App, too and I think it it was displayed in the local api, too.

I just disconnected my Daikin WiFi Adapter last week and replaced it with the Faikin, but might connect it again in the next days to look for consumption data clues.

> Characteristic name: consumptionData
>
>Description: You can consult the energy consumption of your devices. Please note that not all Daikin devices support consumption data through the API, as such it's possible that the consumptionData object is not always present.
`"consumptionData": {`
`  "settable": false,`
`  "value": {`
`    "electrical": {`
`      "unit": "kWh",`
`      "heating": {`
`        "d": [ 0, 0, 0, 1, 2, 0, 2, 3, 0, 1, 1, 1, 1, 2, 1, 1, 2, 3, 1, 5, 7, null, null, null ],`
`        "w": [ 0, 0, 0, 0, 100, 30, 130, 0, 0, 0, 0, 100, null, null ],`
`        "m": [ 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 264 ]`
`      },`
`      "cooling": {`
`        "d": [ 0, 0, 0, 1, 2, 0, 2, 3, 0, 1, 1, 1, 1, 2, 1, 1, 2, 3, 1, 5, 7, null, null,  null ],`
`        "w": [ 0, 0, 0, 0, 100, 30, 130, 0, 0, 0, 0, 100, null, null],`
`        "m": [ 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 264 ]`
`      }`
`    }`
`  }`
`}`
>
> There can be electrical and gas data. The data is split up between heating and cooling (if available). There are three arrays of data available. "d" contains 24 fields spread over the current day and the day before.
`d[0] => 00:00-02:00 yesterday`
`d[1] => 02:00-04:00 yesterday`
`...`
`d[22] => 20:00-22:00 today`
`d[23] => 22:00-24:00 today`
>
> "w" contains 14 fields spread over this week and last week.
`w[0] => Monday last week`
`w[1] => Tuesday last week`
`...`
`w[12] => Saturday this week`
`w[13] => Sunday this week`
>
> "m" contains 24 fields spread over this year and last year.
`m[0] => January last year`
`m[1] => February last year`
`...`
`m[22] => November this year`
`m[23] => December this year`
>
> Due to the structure of the data, some fields can be empty (null), which means data is not (yet) available for this timeslot.
>
>Consumption data is expected to be updated every 2 hours, randomized from the startup time of the Daikin device. This means that, for example, data d[0] will be available between 02:00 & 04:00.

revk, 2024-03-23:
OK changing to 100wH units - beta issue shortly.

sm-Fifteen, 2024-03-23:
> Well, correct me if I am wrong, but some values need to be converted to decimal, others are fine as hex, but its always inverted.
>
> I wonder if FM / GM is not simply counting up the kWh, in my case [...]

Ah! It did seem weird to me that the energy reading value would depend on a device-dependent and possibly dynamic multiplier. FM/GM being hectowatt-hours makes a lot more sense.

44616E, 2024-03-24:
> OK changing to 100wH units - beta issue shortly.

Downloaded latest Beta on to my units, but now don't seem to have Wh in the State or Faikin MQTT messages. So can't compare the new values (I can only do this at night when the other loads are off as the power meter in for the whole building).

revk, 2024-03-24:
Grr, I set to poll `P` not `M`, sorry. Try again in moment - making beta now.

44616E, 2024-03-24:
Thanks, now getting WH. Looks good so far, will check it over night to see how it compares.

44616E, 2024-03-24:
[image]
Seems pretty good if averaged over a reasonable period of time. Over 20 minutes I got 3.3kw (converting from Wh to W) from Wh compared to around 3.28kW according to the meter.

sm-Fifteen, 2024-03-30:
Ok, so I finally plugged in my Faikin boards, I've been trying out a few things.

So far, I can't get either of my units (two FTXS12LVJU connected to a single 2MXS18NMVJUA outdoor unit) to reply to an `FM` command. I have in fact rested the others [listed here](https://github.com/revk/ESP32-Faikin/wiki/S21-Protocol), and I'm getting this:

* My unit replies to F1, F2, F3, F4, F5, F8
* It does not reply to F6, F7, F9, FB, FG, FK, __FM__, FN, FP, FQ, FP, FT, FU, FX
* It also replies to RH, RN, RI, Ra, RX, Rd, RL

So energy monitoring could be something that's not supported on all models, possibly?

[10 more replies]

closed 2024-04-04 by revk