revk/xESP32-Faikout #102: Fan mode "auto" jumps back to "night mode"

andreas-bulling, 2023-09-08:
Hi,

The Faiking modules overall work like a charm, including the HA integration, but one thing that doesn't seem to work robustly is the selection of the fan mode.

When I select "auto" for one of the indoor units it always jumps back to "night". For the other sometimes it works, sometimes it also jumps back. It also seems that the mode set through Faikin is not reflected on the remote control (but maybe it is also not supposed to, not sure).

revk, 2023-09-08:
Anyway, leave it with me for now. I'll see if I can do an update when I am settled on the boat.

andreas-bulling, 2023-09-08:
I am sure you are aware of this: https://github.com/joshbenner/esphome-daikin-s21

revk, 2023-09-08:
I have seen lots - are you saying they have cracked the auto/quiet mode stuff? If so I'll take a more detailed look.

joshbenner, 2023-09-08:
I don't think we've been able to figure out quiet mode yet.

revk, 2023-09-08:
So my plan is, as I had before, if you set quiet and it comes back auto we assume it is quiet. But if if are quiet mode and see a high fan speed we assume it is wrong and is auto, but not (as now) the other way. Ie auto will do low and high fan, but quiet will only do low fan. I hope.

andreas-bulling, 2023-09-10:
I am afraid I can't contribute much to this discussion/decision - I only added the above link because I believe in knowledge and linking people. And I wasn't sure whether the two of you know each other's projects. Likely, but not sure ;)

revk, 2023-09-12:
We are far enough North on this boat to have extra bad ship's internet. I may be able to get on to this change tomorrow. Sorry for the delay.

revk, 2023-09-14:
OK, new code issues, please upgrade and let me know if it is OK for you now.

closed 2023-09-14 by revk

andreas-bulling, 2023-09-14:
thanks, will give it a try tonight when I am back at home